emsesp/EMS-ESP32 #242: Unrecognized EMS device (GC2300W 23/30 C 23)

aibott, 2021-12-12:
Hello,

I am having a gas burner - Bosch Condense 2300 (GC2300W 23/30 C 23), which I wanted to connect to with EMS-ESP32 (version 3.3.1b0) via EMS bus. According to the documentation the burner should be supported, however, it seems that I am not able to parce the telegrams coming out from the burner (the LED is blinking rapidly and getting the following error): 
```
000+00:18:29.496 E 171: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B 88 02 00 20
000+00:18:29.496 W 172: [emsesp] Unrecognized EMS device (device ID 0x08, no product ID). Please report on GitHub.  
```
For start, I wanted only to read out the burner parameters, so tx – off, no wifi, no MQTT. I am using, a custom HW designed board (prototype), so first thought that the problem might be in the schematics. Checked the tx / rx waveforms on the ESP32 pins, both seems to me - OK. Also "watched" the received telegrams in raw, they are coming out and look OK too:  
```
Watching incoming telegrams, displayed as raw bytes
000+00:06:38.817 N 8: [telegram] Rx: 88 00 E5 00 21 00 20 00 00 78 00 00 40 00 00 27 8E 01 07 26 00 00 00 00 EA 21 00 18 28 64 00 EE
000+00:06:39.012 N 9: [telegram] Rx: 88 00 E5 1D 00 F0 50 B6
000+00:06:39.226 N 10: [telegram] Rx: 88 00 E9 00 2B 01 7E 00 00 00 00 01 7E 46 00 00 00 00 00 1D 05 00 0F 66 00 00 00 00 01 70
000+00:06:47.783 N 11: [telegram] Rx: 88 00 E3 00 04 00 00 00 04 00 00 00 00 00 00 01 A8 00 64 2B 01 00 00 4C
000+00:06:47.991 N 12: [telegram] Rx: 88 00 E4 00 00 2D 2D 00 00 CC 2B 01 A8 64 00 02 02 01 AC 00 00 00 00 00 00 14 00 01 A8 00 00 B8
```
Did not have enough time to investigate where the problem might come from this weekend, however, will continue during the next week. 

Just decided to give you a note, so if someone have time and experience with this particular burner may be can easily point out where the problem comes from or what I am doing wrong (attached are some tests which I was able to do and check).  

Regards,
Kamen 

[esp32-ems.zip](https://github.com/emsesp/EMS-ESP32/files/7699105/esp32-ems.zip)

MichaelDvP, 2021-12-12:
With tx_mode off you can only watch telegrams. To register a device the version information is needed. It works if another devices (thermostat) requests this info from boiler, but you only have the boiler and it does not send out the version telegram. It has to be requested by emsesp.
Your circuit allows tx, enable tx-mode.

aibott, 2021-12-12:
Michael, thank you for your prompt reply. Well noted, that explains a lot – I do not have any other devices on the bus, than the board and the boiler. Will test it and update.

aibott, 2021-12-13:
Fixed, thanks again Michael for the hint. Helped me just to find the problem. The emsesp is detecting  the burner as Logamax Plus GB122, seems that they are sharing same product IDs.

closed 2023-02-19 by proddy