emsesp/EMS-ESP32 #243: Setting mode via MQTT in HA doesn't work

Imaginous, 2021-12-13:
**Bug description**
Latest Home Assistant, using MQTT and 3.3.0 and beta 3.3.1b0.
When setting the thermostat mode to heat in HA it will stay on auto. 

(it looks like an earlier mode bug where data was not accepted, but value had to ve used.) 
I need MQTT for other connected devices. 

**Steps to reproduce**
Select a different heating mode in HA. 

**Expected behavior**
The thermostat changing mode.

**Device information**
`
{
  "System": {
    "version": "3.3.1b0",
    "uptime": "000+00:12:23.327",
    "uptime (seconds)": 743,
    "freemem": 168,
    "reset reason": "Software reset CPU / Software reset CPU",
    "Dallas sensors": 0
  },
  "Network": {
    "connection": "WiFi",
    "hostname": "ems-esp-nefit",
    "SSID": "Imaginous",
    "BSSID": "7C:10:C9:CD:B6:30",
    "RSSI": -48,
    "MAC": "78:E3:6D:17:23:38",
    "IPv4 address": "192.168.0.24/255.255.255.0",
    "IPv4 gateway": "192.168.0.1",
    "IPv4 nameserver": "192.168.0.1"
  },
  "Status": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "telegrams received": 502,
    "read requests sent": 105,
    "write requests sent": 1,
    "incomplete telegrams": 8,
    "tx fails": 0,
    "rx line quality": 100,
    "tx line quality": 100,
    "MQTT": "connected",
    "MQTT publishes": 321,
    "MQTT publish fails": 0,
    "Dallas reads": 0,
    "Dallas fails": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "BK13/BK15/Smartline/GB1x2 (DeviceID:0x08, ProductID:64, Version:03.05)",
      "handlers": "0x10 0x11 0xC2 0x14 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "Thermostat",
      "name": "Moduline 400 (DeviceID:0x10, ProductID:78, Version:03.01)",
      "handlers": "0xA3 0x06 0xA2 0x12 0x41 0xA7"
    },
    {
      "type": "Controller",
      "name": "BC10/RFM20 (DeviceID:0x09, ProductID:68, Version:02.03)"
    }
  ]
}`

proddy, 2021-12-13:
please look at the [troubleshooting guide](https://emsesp.github.io/docs/#/Troubleshooting) and provide more details. Does it work from the web? does it work from the console? does it work via the API? what are you using in HA to change the value? What does the log say? etc

Imaginous, 2021-12-13:
I will do some exta testing tomorrow. I will keep you posted.

Imaginous, 2021-12-14:
I did some more testing. Sending following commands trough MQTT:

`{
  "cmd": "mode",
  "data": "auto"
 }`

or

`{
  "cmd": "mode",
  "data": "heat"
 }`

to: `ems-esp-nefit/thermostat`

does not work.

However sending

`{
  "cmd": "temp",
  "data": 18.5
}`

does work normally.

So it seems somthing goes wrong with the "mode" command.
I noticed it is now displayed as "hamode", but sending it like that does not help.

This is my MQTT readout of the thermostat:
`{"id":78,"errorcode":"(0)","datetime":"17:48:48 14/12/2021","hc1":{"seltemp":18.5,"temp":18.5,"currtemp":19.9,"hatemp":19.9,"hamode":"heat"}}`

I never used the API directly. I just switched from an ESP8266 to the ESP32 with the newest firmware. On the ESP8266 with the latest firmware it worked flawlessly.

proddy, 2021-12-14:
What do you see in the log after sending those commands?

Imaginous, 2021-12-14:
I see the following:

`2021-12-14 17:57:14.536 D 3929: [command] Command failed: invalid command 'mode'
2021-12-14 17:57:14.536 E 3930: [mqtt] Call failed with error: invalid command (Not Found)
2021-12-14 17:57:15.057 D 3931: [mqtt] Publishing topic ems-esp-nefit/boiler_data (#13668, retain=0, retry=1, size=717, pid=1)
2021-12-14 17:57:15.158 D 3932: [mqtt] Publishing topic ems-esp-nefit/boiler_data_ww (#13669, retain=0, retry=1, size=417, pid=1)
2021-12-14 17:57:15.259 D 3933: [mqtt] Publishing topic ems-esp-nefit/thermostat_data (#13670, retain=0, retry=1, size=141, pid=1)`

Imaginous, 2021-12-14:
It seems that he *flowtemp* command does the same:

`2021-12-14 17:58:05.082 D 3965: [command] Command failed: invalid command 'flowtemp'
2021-12-14 17:58:05.082 E 3966: [mqtt] Call failed with error: invalid command (Not Found)`

the temperature setting does work:
`2021-12-14 17:59:38.965 I 4019: [command] Calling thermostat command 'temp', value 18.0, id is default
2021-12-14 17:59:38.965 I 4020: [thermostat] Setting thermostat temperature to 18.0 for heating circuit 1, mode auto`

proddy, 2021-12-14:
Ok I'll check tonight. Can you also see which commands are available? Either via the console or the api call. Instructions are in the doc

Imaginous, 2021-12-14:
With the old ESP8266 and version 2.2.1 all the commands like *mode*, *flowtemp* and so on worked fine.
I only had some WiFi issues with my new router so I upgraded to an ESP32... didn't think of loosing functionality. ;)

The commands listed for *thermostat*:
`{
  "commands": "lists all commands",
  "entities": "lists all entities",
  "info": "lists all values",
  "seltemp": "selected room temperature"
}`

The commands listed for *boiler*:
`{
  "boilhystoff": "hysteresis off temperature",
  "boilhyston": "hysteresis on temperature",
  "burnmaxpower": "burner max power",
  "burnminperiod": "burner min period",
  "burnminpower": "burner min power",
  "commands": "lists all commands",
  "entities": "lists all entities",
  "heatingactivated": "heating activated",
  "heatingtemp": "heating temperature",
  "info": "lists all values",
  "maintenance": "maintenance scheduled",
  "maintenancedate": "maintenance set date",
  "maintenancetime": "maintenance set time",
  "pumpdelay": "pump delay",
  "pumpmodmax": "burner pump max power",
  "pumpmodmin": "burner pump min power",
  "reset": "reset messages",
  "selburnpow": "burner selected max power",
  "selflowtemp": "selected flow temperature",
  "wwactivated": "activated",
  "wwcirc": "circulation active",
  "wwcircmode": "circulation pump frequency",
  "wwcircpump": "circulation pump available",
  "wwcomfort": "comfort",
  "wwdisinfect": "disinfection",
  "wwdisinfectiontemp": "disinfection temperature",
  "wwflowtempoffset": "flow temperature offset",
  "wwhystoff": "hysteresis off temperature",
  "wwhyston": "hysteresis on temperature",
  "wwmaxpower": "max power",
  "wwonetime": "one time charging",
  "wwseltemp": "selected temperature",
  "wwseltemplow": "selected lower temperature",
  "wwseltempsingle": "single charge temperature",
  "wwtapactivated": "turn on/off DHW by going into maintenance mode"
}`

proddy, 2021-12-14:
I'm sorry that this free software you're using which I spent the last 3 years building is not to your satisfaction 🤪

But seriously, you don't have the commands which is why it's not working. That's obvious from the output you just posted. Looks like the moduline 400 is not configured correctly in the code. Should be an easy fix.

Imaginous, 2021-12-14:
I'm not disappointed, and I love your work.

I have my own project for unlocking the speed limit of Cowboy e-bikes. It's also voluntary. I know it costs loads of time.

So I really appreciate it. I just didn't anticipate on losing functionality. So I hope it can be fixed easily.
It worked on the last old 2.2 version, so good luck in solving it.

Again, I respect your work. :)

Imaginous, 2021-12-14:
I think the moduline 400 (Nefit) should have the same options (commands) as the RC35 (Buderus).

Just my 2 cents.

proddy, 2021-12-14:
I noticed the RC30 logic which is also used for the Moduline400 is missing. It must have been accidentally left off during all the changes and we never noticed. I'll add it back

Imaginous, 2021-12-14:
Thanks for the quick response :)

proddy, 2021-12-14:
Check the latest build 3.3.1b1 - I may have the modes switched (on/off/auto) so let me know if something looks strange

Imaginous, 2021-12-14:
It all seems to work fine.
This seems resolved. :)

For some reason I'm also not able to set the boiler flow temperature since the new version.
... But that's for another time ;)

closed 2021-12-14 by Imaginous

proddy, 2021-12-14:
http://ems-esp.local/api/boiler/commands should show you the comands and `selflowtemp` should be there ? Otherwise make a new issue!