emsesp/EMS-ESP32 #244: EMS-ESP32 restarting

maciejka104, 2021-12-14:
My device running v3.3.1b0 firmware is restarting/rebooting by itself. 
EMS connected to Junkers cerapur. 
Approximately every 4-6 hours is restarting. Max uptime is 10 hrs. 
Anything I could check in the settings?
I did reset to factory settings and configured minimum – basically MQTT (connected to HA) and EMS Bus (HT3). 
What could be a problem?

proddy, 2021-12-14:
wow, that could be anything. Did you build the circuit yourself? Does it crash if you disable the MQTT? Try and pinpoint which service is crashing. Use the syslog to capture the logs which may hint to something weird happening. Also let us know what your config is

bbqkees, 2021-12-15:
@maciejka104 had a Premium II which we upgraded with an ESP32.
Both the old and the new board have the same rebooting issue, even when powered via USB without EMS connected so I basically ruled out a hardware issue at this point. Perhaps a fringe case with his setup that freaks out EMS-ESP.

proddy, 2021-12-15:
Thanks. I think it is software related, most likely a null pointer/buffer overflow problem. The arduino core is very sensitive to those.

proddy, 2021-12-21:
any update @maciejka104 ?

maciejka104, 2021-12-22:
I have disabled MQTT for a couple of days. 
Basically still the same, up time - up to 20 h
Tested on v3.3.0  and v3.3.1b1

proddy, 2021-12-23:
These are the worst type of bug. It's like looking for a needle in a haystack. All I can recommend at this point is 1) check the System Log for the restart reason (it'll be the first line) and 2) setup a Syslog server and capture the logs just before it crashes.

glitter-ball, 2021-12-24:
Do you have a Dallas sensor connected? If so, is the data coming back from this always OK? I found weird Dallas values and restarts went together on my system. That was down to power issues in my case but it might help narrow things down for you too.

maciejka104, 2021-12-24:
No Dallas sensor. 
Just junkers boiler.

proddy, 2021-12-25:
then my only recommendation is https://emsesp.github.io/docs/#/Troubleshooting?id=ems-esp-sometimes-crashes-and-restarts

maciejka104, 2021-12-26:
I have checked syslog server today and it was empty (qnap). 
Checked EMS-ESP configuration and then I noticed that I have set up EMS-ESP IP as syslog server IP :/
Silly me 😊 
Corrected it and let’s wait for the result.

proddy, 2021-12-29:
fingers crossed!

maciejka104, 2021-12-30:
So … correcting syslog's IP didn't help :/
Few min ago it has restarted 
Here is part of the logs https://pastebin.com/DrBGUFw1 
Take a look at 2021-12-30T10:47:16 and on 
Last log was:
ems-esp telegram - - - telegram- 000+05:14:01.597 D 55449: Sending read Tx [#124], telegram: 8B 90 FF 00 20 00 65
Any ideas? 
If you need I have a logs for a few days. 
For the records
Below uptime taken from HA
 
[image]

proddy, 2021-12-30:
so restarts every 3 hrs with MQTT enabled, and with MQTT disabled it crashes after 20hrs? Wifi has a strong signal and the ESP32 is powered by a steady 5V supply?

maciejka104, 2021-12-30:
Power: 5V power supply - yes, I have tested few PS. 
WIFI:  based on HA info: WIFI strength 50-70%, WIFI RSSI: 71 dBm. In general I have good wifi coverage (cisco).
MQTT: looks like you said - disabling mqtt extends the time between reboots but it still reboots.

proddy, 2022-01-08:
and if you disconnect from the EMS bus and run it standalone (with MQTT) does it still reboot every few hours?

maciejka104, 2022-01-12:
I think problem is solved.
1. I have uploaded firmware to v3.3.0 (same as old one) 
2. Reboot/Restart
3. Factory reset
4. Configured Boiler settings, MQTT, syslog 
5. Reboot/Restart

Device is up for 3 days now :)

proddy, 2022-01-12:
great! but still weird what it was failing before

maciejka104, 2022-01-12:
I really think key action was factory reset  and / or  re flash same firmware.
Before doing that I have tested connection to boiler, checked power supply and it did not help.

proddy, 2022-01-12:
ok, I'll make a note of this in the doc and close this issue. thanks for getting back to us.

closed 2022-01-12 by proddy