emsesp/EMS-ESP32 #246: heating curve heat pump

rickroetenberg, 2021-12-15:
Anybody idea if it would be possible to adjust the heating curve from the heat pump through EMS-ESP32? So the foot point and the end point?

bbqkees, 2021-12-16:
Can this be done from the app/web interface of the heat pump?

If not then we have no way of finding out how it's implemented.

rickroetenberg, 2021-12-21:
No, but the base point and max. flow temperature can be changed through EMS-ESP32.

Base point = hc1 min flow temperature
Max flow temperature = hc1 max flow temperature

I think that the end point of the heating point should be the hc1 max flow temperature in EMS-ESP32.

[image]
[image]

closed 2023-02-19 by proddy