emsesp/EMS-ESP32 #247: Solar cylinder priority

PhillyGilly, 2021-12-16:
Just uploaded 3.3 and working though my Solar installation
{
  "id": 164,
  "collectortemp": 11.6,
  "tankbottomtemp": 50.7,
  "solarpump": "off",
  "pumpworktime": 394114,
  "tankmaxtemp": 65,
  "solarpumpmodulation": 0,
  "pumpminmod": 1,
  "turnondiff": 8,
  "turnoffdiff": 5,
  "tank2bottomtemp": 18.5,
  "cylinderpumpmodulation": 0,
  "valvestatus": "off",
  "tankheated": "off",
  "collectorshutdown": "off",
  "collectormaxtemp": 120,
  "collectormintemp": 20,
  "energylasthour": 0,
  "energytoday": 0,
  "energytotal": 5561,
  "heattransfersystem": "off",
  "externaltank": "off",
  "thermaldisinfect": "off",
  "heatmetering": "off",
  "activated": "on",
  "solarpumpmode": "pwm",
  "pumpkick": "off",
  "plainwatermode": "off",
  "doublematchflow": "off",
  "climatezone": 85,
  "collector1area": 4,
  "collector1type": "flat"
}

I'm pretty 'egal' about the cylinders becoming tanks although I noticed that we still have a cylinderpumpmodulation and solarpumpmodulation. 
But the "solar cylinder priority" mod that @MichaelDvP had in EMS-ESP_solar-3_2_2b14-ESP32.bin after #137 is missing.

MichaelDvP, 2021-12-16:
It is not merged to dev. The latest change was from #212 and waiting for data from second collector field of #164. 
Actual status is the branch referenced in #212, which contains the solar development and the latest dev-changes.

proddy, 2021-12-16:
let's first get the new web code in (from ft_reacthooks branch) into dev and then bring in the solar changes. It's dev so it's ok if its a bit shaky and experimental.

PhillyGilly, 2021-12-16:
Sorry guys. I assumed it would be without reading the long list of changes.
If you have noted the duplicate pumps, I will live without the priority cylinder until it arrives.
Thanks.

PhillyGilly, 2022-01-19:
Hi @proddy is that closing as "select solar cylinder will be in 3.4“ or closing because @MichaelDvP did not get the info he needed. Hopefully the first one?

MichaelDvP, 2022-01-19:
All the solar changes and renamings are in v3.4. There are some values for second collector missing, but this can be added later.

proddy, 2022-01-19:
this is all Michael's great work, I just merged it into v3.4. I'm putting the v3.4 builds into Discord for now

PhillyGilly, 2022-01-21:
Are there 3.4 bin files available?  I’m on 3.3.1 and happy to roll forward/back. I can see the commits on Discord under #ems-esp32. Previously you used to put the “assts” on GitHub.

 

 

best regards

 

Phil Gilbert

 

7 Eastgate,

Hallaton,

Leicestershire

LE16 8UB

 

email:  ***@***.***> ***@***.***

MichaelDvP, 2022-01-21:
See "announcements" on discord. I've also posted some changes before merging in #294.

PhillyGilly, 2022-01-21:
Thank you. Too obvious for me to see it!

 

 

best regards

 

Phil Gilbert

 

7 Eastgate,

Hallaton,

Leicestershire

LE16 8UB

 

email:  ***@***.***> ***@***.***

proddy, 2022-01-26:
Hi @PhillyGilly , I'm updated some of my test data. When you have time free and settled into v3.4 would you mind sending over a copy of 
http://ems-esp.local/api/solar/info
http://ems-esp.local/api/solar

so I can re-use some of that data in the test suite? thanks

PhillyGilly, 2022-01-26:
@proddy. Here they are:
{
  "collector temperature (TS1)": 63.2,
  "cylinder bottom temperature (TS2)": 45.8,
  "pump (PS1)": "on",
  "pump working time": "276 days 17 hours 2 minutes",
  "maximum cylinder temperature": 65,
  "collector shutdown": "off",
  "cyl heated": "off",
  "pump modulation (PS1)": 15,
  "minimum pump modulation": 5,
  "pump turn on difference": 8,
  "pump turn off difference": 5,
  "heat assistance valve (M1)": "off",
  "heat assistance valve power (M1)": 0,
  "pump 2 (PS4)": "off",
  "pump 2 modulation (PS4)": 0,
  "second cylinder bottom temperature (TS5)": 52.7,
  "cylinder pump modulation (PS5)": 0,
  "valve status": "on",
  "maximum collector temperature": 120,
  "minimum collector temperature": 20,
  "energy last hour": 581.8,
  "energy today": 6323,
  "energy total": 5605.2,
  "pump 2 working time": "0 days 0 hours 0 minutes",
  "differential control working time": "0 days 0 hours 0 minutes",
  "heattransfer system": "off",
  "external cylinder": "off",
  "thermal disinfection": "off",
  "heatmetering": "off",
  "activated": "on",
  "pump mode": "pwm",
  "pumpkick": "off",
  "plain water mode": "off",
  "doublematchflow": "off",
  "pump 2 turn on difference": 10,
  "pumpkick 2": "off",
  "climate zone": 85,
  "collector 1 area": 4,
  "cylinder priority": "cyl_2",
  "heat counter impulses": 0
}
{
  "collectortemp": 63.2,
  "cylbottomtemp": 45.8,
  "solarpump": "on",
  "pumpworktime": 398462,
  "cylmaxtemp": 65,
  "collectorshutdown": "off",
  "cylheated": "off",
  "solarpumpmod": 15,
  "pumpminmod": 5,
  "turnondiff": 8,
  "turnoffdiff": 5,
  "heatassistvalve": "off",
  "heatassistpower": 0,
  "solarpump2": "off",
  "solarpump2mod": 0,
  "cyl2bottomtemp": 52.7,
  "cylpumpmod": 0,
  "valvestatus": "on",
  "collectormaxtemp": 120,
  "collectormintemp": 20,
  "energylasthour": 581.8,
  "energytoday": 6323,
  "energytotal": 5605.2,
  "pump2worktime": 0,
  "m1worktime": 0,
  "heattransfersystem": "off",
  "externalcyl": "off",
  "thermaldisinfect": "off",
  "heatmetering": "off",
  "activated": "on",
  "solarpumpmode": "pwm",
  "pumpkick": "off",
  "plainwatermode": "off",
  "doublematchflow": "off",
  "turnondiff2": 10,
  "pump2kick": "off",
  "climatezone": 85,
  "collector1area": 4,
  "cylpriority": "cyl_2",
  "heatcnt": 0
}
Hope that helps.
BTW I had to use the URL http://192.168.2.101/api/solar/ as my Unifi set up doesn't recognise names as URLs. I don't suppose if you know if that is caused by setting static IP of for another reason?.

proddy, 2022-01-26:
Thanks Phil. I've had issues with DNS not resolving hostnames on my Unifi setup too with Ethernet, never with WiFi. Your EMS-ESP is on WiFi correct and also showing up correctly on the USG ?

PhillyGilly, 2022-01-26:
All good on USG and my cloud key. However I had named it "ems-esp v3". I thought this was just a "friendly name" but it seems it seems to have affected the DNS too. When I deleted the the name in the console details it respods to ems-esp.local. I thought it was some thing profound. Doh!

closed 2023-02-19 by proddy