emsesp/EMS-ESP32 #248: Trying to fully emulate a thermostat.

guillaume042, 2021-12-16:
Hello,

Here is the "test lab" :

```
ems-esp:/# show devices
These EMS devices are currently active:

(1) Boiler: Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3 (DeviceID:0x08, ProductID:95, Version:10.12)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0x14 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 

(3) Thermostat: RC100/Moduline 1000/1010 (DeviceID:0x18, ProductID:165, Version:40.04) ** master device **
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 0xA2 0x12 0x2A5 0x2B9 0x2AF 0x29B 0x2A6 0x2BA 0x2B0 0x29C 0x2A7 0x2BB 0x2B1 0x29D 0x2A8 0x2BC 0x2B2 0x29E 0x2F5 0x31B 0x31D 0x31E 0x23A 0x267 0x240 

(2) Controller: HT3 (DeviceID:0x09, ProductID:95, Version:10.12)
```

A boiler, a thermostat (in the ceiling so the T° is really low) and an ems esp gateway.

- initial situation : 
  Heating active: on
  Selected flow temperature: 47 °C

-  Setting manual temp on thermostat to low value. the thermostat shut all.
  Heating active: off
  Selected flow temperature: 0 °C

- Setting manual temp on thermostat to high value. the thermostat put boiler on. 
  Heating active: on
  Selected flow temperature: 47 °C

I've watch all the packets during these operations (attached file).
[logems.txt](https://github.com/emsesp/EMS-ESP32/files/7727126/logems.txt)

Can someone help me to find how to fully emulate the thermostat ?
I would like to know what it does excatly, and how it manage to make the values retains by the boiler.

Thanks.

glitter-ball, 2021-12-18:
This question gets asked regularly. My question back would be : "Why emulate a thermostat when you can buy one?" There's a lot going on inside the thermostat:

- Monitoring internal and external temperature sensors
- Managing the heating and hot water time programmes
- Calculating boiler flow temperatures using weather compensation
- Protecting the boiler and heating installation with frost protection
- Controlling the pumps inside and outside of the boiler
- Controlling the 3-way valve, if there is one
- Optimum start of heating periods, if enabled
 
to name just a few.

Back in my younger years, a mate and I tried to write a heating control system from scratch. It worked OK, but building a system that caters for every eventuality and works reliably is very hard! Users (especially family...) aren't sympathetic when it stops working and they are cold or it goes mad and wastes lots of energy...

So I'd say buy a good thermostat and use EMS-ESP to monitor it and fine-tune its settings. Then, most of the time, you can relax and just let it get on with the job! Others might disagree...

guillaume042, 2021-12-18:
I understand your point of view. 

Mine is that i like to understand how things work.
If i can do it why not give a try that might be fun.

Thanks for your tips about what goes inside the thermostat. 

Guillaume

glitter-ball, 2021-12-18:
OK. I'd start with the core stuff first : timed programmes for heating and hot water and basic temperature control. You'd have to guess some sensible flow temperatures. For hot water, you need the boiler running at least 5ºC than the stored hot water, possibly 10ºC. For the heating you could start with 55ºC (if you have radiators, or lower if you have under-floor) which would almost guarantee the boiler will condense, for improved efficiency. That might be higher than you need or, in cold weather, it might have to be hotter. Also depends on your heating system - under-floor, radiators etc.

SzymonZy, 2021-12-24:
As I have just ordered PCB and going to mount the ems board (https://github.com/rocksolidsr/esp-ems-board) I have similar question. 
Would it be possible to setup third temp zone (beside day, night). I'm working alone for 6h daily at home and I don't need the same temp as during family life or night. Can it be automated to daily change Day temp from 8 till 2PM  to 20, and at 2PM bumped up to 22 and go down at 10 PM to 18. 
In general third temp zone is necessary. As rc35 does not provide it I want to update it's config on the fly with HA and NodeRed.

closed 2023-02-19 by proddy