emsesp/EMS-ESP32 #250: Bosh FF20 fresh water station not recognized correctly

Torsten-, 2021-12-19:
**Bug description**
My whole system consist of:
- Cerapur ZSB as boiler
- a big water tank as buffer
- CW400 to control the system
- Solar station (MS 200) with one solar circuit
- One Mixer for floor heating (MM 100)
- FF20 fresh water station for warm water (MS 100)
- bbqkees E32 with FW v3.3.0

The following devices were detected:

``` 
ems-esp:# show devices

These EMS devices are currently active:

(1) Boiler: GBx72/Trendline/Cerapur/Greenstar Si/27i (DeviceID:0x08, ProductID:123, Version:06.03)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0xC2 0x14 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 

(3) Thermostat: RC300/RC310/Moduline 3000/1010H/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:74.02)
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 0xA2 0x12 0x2A5 0x2B9 0x2AF 0x29B 0x471 0x2A6 0x2BA 0x2B0 0x29C 0x472 0x2A7 0x2BB 0x2B1 0x29D 0x473 0x2A8 0x2BC 0x2B2 0x29E 0x474 0x2F5 0x31B 0x31D 0x31E 0x23A 0x267 0x240 

(5) Mixer: MM100 (DeviceID:0x21, ProductID:160, Version:24.03)
 This Mixer will respond to telegram type IDs: 0x2D8 

(4) Solar: SM200/MS200 (DeviceID:0x30, ProductID:164, Version:25.05)
 This Solar will respond to telegram type IDs: 0xF9 0x358 0x35A 0x362 0x363 0x366 0x364 0x36A 0x380 0x38E 0x391 

(6) Solar: SM100/MS100 (DeviceID:0x2A, ProductID:163, Version:27.05)
 This Solar will respond to telegram type IDs: 0x7D6 0x7AA 0x7AB 

(2) Controller: Controller (DeviceID:0x09, ProductID:152, Version:03.01)

ems-esp:#
```

There are two "problems" - the rest is working fine. Really nice project, thanks a lot!! This is a much better solution than the original Bosh Gateway.
1. "Controller" has no data - is that correct?! ("no data available for this device")
2. more important: my fresh water station FF20 is displayed as "solar" and the values are nonsense

```
ems-esp:# show values
[...]
Solar: SM100/MS100 (DeviceID:0x2A, ProductID:163, Version:27.05)
  type: warm water circuit
  ww temperature 1: 1.0 °C
  ww temperature 3: 1.0 °C
  ww temperature 4: 128.0 °C
  ww temperature 5: 128.0 °C
  ww temperature 7: 1.0 °C
  ww pump: 0.0 °C

Controller: Controller (DeviceID:0x09, ProductID:152, Version:03.01)

ems-esp:#
```

Is it possible to get the thing working?
Not just the temperatures but also the configuration parameters.
Du you need more information?

**Device information**
```
{
  "System": {
    "version": "3.3.0",
    "uptime": "000+23:00:02.727",
    "uptime (seconds)": 82802,
    "freemem": 153,
    "reset reason": "Software reset CPU / Software reset CPU",
    "Dallas sensors": 0
  },
  "Network": {
    "connection": "Ethernet",
    "hostname": "ems-esp",
    "MAC": "A8:03:2A:A1:58:2F",
    "IPv4 address": "192.168.42.6/255.255.255.0",
    "IPv4 gateway": "192.168.42.2",
    "IPv4 nameserver": "192.168.42.2"
  },
  "Status": {
    "bus status": "connected",
    "bus protocol": "HT3",
    "telegrams received": 171426,
    "read requests sent": 42806,
    "write requests sent": 0,
    "incomplete telegrams": 1,
    "tx fails": 0,
    "rx line quality": 100,
    "tx line quality": 100,
    "MQTT": "connected",
    "MQTT publishes": 42779,
    "MQTT publish fails": 0,
    "Dallas reads": 0,
    "Dallas fails": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "GBx72/Trendline/Cerapur/Greenstar Si/27i (DeviceID:0x08, ProductID:123, Version:06.03)",
      "handlers": "0x10 0x11 0xC2 0x14 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "Thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:74.02)",
      "handlers": "0xA3 0x06 0xA2 0x12 0x2A5 0x2B9 0x2AF 0x29B 0x471 0x2A6 0x2BA 0x2B0 0x29C 0x472 0x2A7 0x2BB 0x2B1 0x29D 0x473 0x2A8 0x2BC 0x2B2 0x29E 0x474 0x2F5 0x31B 0x31D 0x31E 0x23A 0x267 0x240"
    },
    {
      "type": "Mixer",
      "name": "MM100 (DeviceID:0x21, ProductID:160, Version:24.03)",
      "handlers": "0x2D8"
    },
    {
      "type": "Solar",
      "name": "SM200/MS200 (DeviceID:0x30, ProductID:164, Version:25.05)",
      "handlers": "0xF9 0x358 0x35A 0x362 0x363 0x366 0x364 0x36A 0x380 0x38E 0x391"
    },
    {
      "type": "Solar",
      "name": "SM100/MS100 (DeviceID:0x2A, ProductID:163, Version:27.05)",
      "handlers": "0x7D6 0x7AA 0x7AB"
    },
    {
      "type": "Controller",
      "name": "Controller (DeviceID:0x09, ProductID:152, Version:03.01)"
    }
  ]
}
```

**Additional context**
```
ems-esp:# watch 0x2A
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a device ID or telegram type of 0x2A
000+23:03:09.394 N 384: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 00 F1 00 F1 80 00 00 80 00 80 00 80 00 00
000+23:03:16.067 N 385: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7A9), data: 4B
000+23:03:17.520 N 386: [emsesp] Thermostat(0x10) -> Solar(0x2A), SM100wwCommand(0x7AB), data: 00 00 02
000+23:03:19.544 N 387: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 00 F1 00 F1 80 00 00 80 00 80 00 80 00 00
000+23:03:29.119 N 388: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 00 F1 00 F1 80 00 00 80 00 80 00 80 00 00
000+23:03:39.369 N 389: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 00 F1 00 F1 80 00 00 80 00 80 00 80 00 00
000+23:03:45.842 N 390: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7A9), data: 4B
000+23:03:46.047 N 391: [emsesp] Solar(0x2A) -> All(0x00), SM100wwStatus(0x7AA), data: 00 00 03 00 03 00 32 02 0F
000+23:03:46.622 N 392: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7AD), data: 64 00 00 03
000+23:03:46.822 N 393: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7B1), data: 01 80 00 01 00 00 00 00
000+23:03:47.421 N 394: [emsesp] Solar(0x2A) -> All(0x00), SM100wwTemperature(0x7D6), data: 01 5E 00 00 01 F6 80 00 80 00 80 00 01 5E
000+23:03:47.652 N 395: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 00 00 00 00 00 00 00 03 00 00 00 00 00 00 03 03 03 03
000+23:03:50.419 N 396: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 00 F1 00 F1 80 00 00 80 00 80 00 80 00 00
000+23:03:59.344 N 397: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 00 F1 00 F1 80 00 00 80 00 80 00 80 00 00
000+23:04:09.469 N 398: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 00 F1 00 F1 80 00 00 80 00 80 00 80 00 00
000+23:04:16.842 N 399: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7A9), data: 4B
000+23:04:17.346 N 400: [emsesp] Thermostat(0x10) -> Solar(0x2A), SM100wwCommand(0x7AB), data: 00 00 02
000+23:04:19.644 N 401: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 00 F1 00 F1 80 00 00 80 00 80 00 80 00 00
000+23:04:29.094 N 402: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 00 F1 00 F1 80 00 00 80 00 80 00 80 00 00
000+23:04:39.419 N 403: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 00 F1 00 F2 80 00 00 80 00 80 00 80 00 00
000+23:04:46.692 N 404: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7A9), data: 4B
000+23:04:46.898 N 405: [emsesp] Solar(0x2A) -> All(0x00), SM100wwStatus(0x7AA), data: 00 00 03 00 03 00 32 02 0F
000+23:04:47.847 N 406: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7AD), data: 64 00 00 03
000+23:04:48.047 N 407: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7B1), data: 01 80 00 01 00 00 00 00
000+23:04:48.846 N 408: [emsesp] Solar(0x2A) -> All(0x00), SM100wwTemperature(0x7D6), data: 01 5E 00 00 01 F2 80 00 80 00 80 00 01 5E
000+23:04:49.077 N 409: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 00 00 00 00 00 00 00 03 00 00 00 00 00 00 03 03 03 03
000+23:04:50.094 N 410: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 00 F1 00 F1 80 00 00 80 00 80 00 80 00 00
000+23:04:59.394 N 411: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 00 F1 00 F1 80 00 00 80 00 80 00 80 00 00
000+23:05:09.420 N 412: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 00 F1 00 F1 80 00 00 80 00 80 00 80 00 00
000+23:05:17.346 N 413: [emsesp] Thermostat(0x10) -> Solar(0x2A), SM100wwCommand(0x7AB), data: 00 00 02
000+23:05:17.993 N 414: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7A9), data: 4B
000+23:05:19.620 N 415: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 00 F1 00 F1 80 00 00 80 00 80 00 80 00 00
ems-esp:# 
```

proddy, 2021-12-19:
Looks like the MS100 is sending out on new telegrams unknown to EMS-ESP (0x7a9-0x7ad, 0x7b1, 0x7e0). We''ll need to work together to reverse engineer and decipher them

proddy, 2021-12-19:
It's also normal not to see any data from the controller. It's been removed in v3.4 from the UI to avoid confusion.

Torsten-, 2021-12-19:
> Looks like the MS100 is sending out on new telegrams unknown to EMS-ESP (0x7a9-0x7ad, 0x7b1, 0x7e0). We''ll need to work together to reverse engineer and decipher them

Ok, just tell me if and how I can help and I'll do my very best ;-)

MichaelDvP, 2021-12-19:
This is all in #212 and #164 and should work with the ft_solar branch.

Torsten-, 2021-12-19:
@MichaelDvP thanks a lot - I've used you binary and now I see the values.

[image]

Something new popped up:
In my solar-MS200 now the value for "collector 1 area" is "8,4 undefined".

MichaelDvP, 2021-12-19:
Hm, the unit should be there, try to reload the webpage to update cache.

Torsten-, 2021-12-19:
Ok, refresh did the job - now it's 8,4 sqm.

I just tested with one hand on the CW400.
- "ww temperature 1" is the same as "ww temperature 7" - it is the "warm water temperature"
- "ww temperature 3" is the temperature of the feeding water (in my case out of the buffer tank)
- "pump" keeps displaying "off", even if the pump is running
- "ww circulation active" is shown as "on", but I don't have a circulation pump

Missing values from "diagnose" display in CW400
- Runtime of the pump (in my case 20 h)
- Flow of the warm water (in my case 0 l/min, later 2,2 l/min)
- Pump speed (in my case 0 %, later 5-8 %)
- Then there is a status as text-value?!

Also there are several settings missing. I've took some photos. Also I did record the "watch" output of the MS100 Module. There are some more unrecognized packets.

[image]
[image]
[image]
[image]
[image]
[image]

```
ems-esp:# watch 0x2A
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a device ID or telegram type of 0x2A
000+04:45:59.538 N 1: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 00 F6 00 F6 80 00 00 80 00 80 00 80 00 00
000+04:46:10.163 N 2: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 00 F6 00 F6 80 00 00 80 00 80 00 80 00 00
000+04:46:14.186 N 3: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7A9), data: 4B
000+04:46:14.391 N 4: [emsesp] Solar(0x2A) -> All(0x00), SM100wwStatus(0x7AA), data: 00 00 03 00 03 00 32 02 0F
000+04:46:15.017 N 5: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7AD), data: 64 00 00 03
000+04:46:15.216 N 6: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7B1), data: 01 80 00 01 00 00 00 00
000+04:46:16.040 N 7: [emsesp] Solar(0x2A) -> All(0x00), SM100wwTemperature(0x7D6), data: 01 2B 00 00 01 E7 80 00 80 00 80 00 01 2B
000+04:46:16.272 N 8: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 00 00 00 00 00 00 00 03 00 00 00 00 00 00 03 03 03 03
000+04:46:17.630 N 9: [emsesp] Me(0x0B) <- Solar(0x2A), SM100wwCirc(0x7A5), data: 20
000+04:46:17.668 N 10: [emsesp] Solar(0x2A) -> Me(0x0B), SM100wwCirc(0x7A5), data: FF 03 0A 04 00 1E 32
000+04:46:18.530 N 11: [emsesp] Me(0x0B) <- Solar(0x2A), SM100wwParam(0x7A6), data: 20
000+04:46:18.612 N 12: [emsesp] Solar(0x2A) -> Me(0x0B), SM100wwParam(0x7A6), data: FF 05 0F 5F 00 01 3C 3C 3C 32 2D 12 46 00 3C 1E 03 07 3C 00 0F 00 05
000+04:46:19.789 N 13: [emsesp] Thermostat(0x10) -> Solar(0x2A), SM100wwCommand(0x7AB), data: 00 00 02
000+04:46:20.389 N 14: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 00 F6 00 F6 80 00 00 80 00 80 00 80 00 00
000+04:46:29.563 N 15: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 00 F6 00 F6 80 00 00 80 00 80 00 80 00 00
000+04:46:39.813 N 16: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 00 F6 00 F6 80 00 00 80 00 80 00 80 00 00
000+04:46:45.084 N 17: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7A9), data: 4B
000+04:46:51.088 N 18: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 00 F6 00 F6 80 00 00 80 00 80 00 80 00 00
000+04:46:59.513 N 19: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 00 F6 00 F6 80 00 00 80 00 80 00 80 00 00
000+04:47:09.888 N 20: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 00 F6 00 F6 80 00 00 80 00 80 00 80 00 00
000+04:47:15.086 N 21: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7A9), data: 4B
000+04:47:15.291 N 22: [emsesp] Solar(0x2A) -> All(0x00), SM100wwStatus(0x7AA), data: 00 00 03 00 03 00 32 02 0F
000+04:47:16.341 N 23: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7AD), data: 64 00 00 03
000+04:47:16.541 N 24: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7B1), data: 01 80 00 01 00 00 00 00
000+04:47:16.906 N 25: [emsesp] Me(0x0B) <- Solar(0x2A), SM100wwCirc(0x7A5), data: 20
000+04:47:16.947 N 26: [emsesp] Solar(0x2A) -> Me(0x0B), SM100wwCirc(0x7A5), data: FF 03 0A 04 00 1E 32
000+04:47:17.665 N 27: [emsesp] Solar(0x2A) -> All(0x00), SM100wwTemperature(0x7D6), data: 01 2A 00 00 01 E3 80 00 80 00 80 00 01 2A
000+04:47:17.897 N 28: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 00 00 00 00 00 00 00 03 00 00 00 00 00 00 03 03 03 03
000+04:47:18.281 N 29: [emsesp] Me(0x0B) <- Solar(0x2A), SM100wwParam(0x7A6), data: 20
000+04:47:18.363 N 30: [emsesp] Solar(0x2A) -> Me(0x0B), SM100wwParam(0x7A6), data: FF 05 0F 5F 00 01 3C 3C 3C 32 2D 12 46 00 3C 1E 03 07 3C 00 0F 00 05
000+04:47:20.438 N 31: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 00 F6 00 F6 80 00 00 80 00 80 00 80 00 00
000+04:47:20.791 N 32: [emsesp] Thermostat(0x10) -> Solar(0x2A), SM100wwCommand(0x7AB), data: 00 00 02
000+04:47:21.206 N 33: [emsesp] Solar(0x2A) -> All(0x00), ?(0xBF), data: 2A A3 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
000+04:47:29.563 N 34: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 00 F6 00 F6 80 00 00 80 00 80 00 80 00 00
000+04:47:40.088 N 35: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 00 F6 00 F6 80 00 00 80 00 80 00 80 00 00
000+04:47:46.437 N 36: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7A9), data: 4B
000+04:47:51.088 N 37: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 00 F6 00 F6 80 00 00 80 00 80 00 80 00 00
[... cut ...]
000+05:09:10.344 N 463: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 00 F9 00 F9 80 00 00 80 00 80 00 80 00 00
000+05:09:16.462 N 464: [emsesp] Me(0x0B) <- Solar(0x2A), SM100wwCirc(0x7A5), data: 20
000+05:09:16.499 N 465: [emsesp] Solar(0x2A) -> Me(0x0B), SM100wwCirc(0x7A5), data: FF 03 0A 04 00 1E 32
000+05:09:17.537 N 466: [emsesp] Me(0x0B) <- Solar(0x2A), SM100wwParam(0x7A6), data: 20
000+05:09:17.619 N 467: [emsesp] Solar(0x2A) -> Me(0x0B), SM100wwParam(0x7A6), data: FF 05 0F 5F 00 01 3C 3C 3C 32 2D 12 46 00 3C 1E 03 07 3C 00 0F 00 05
ems-esp:# watch off
```

Torsten-, 2021-12-19:
I found the packet for "Warmhaltung" - in english "keep warm" - it is a setting with just on/off
ON: [emsesp] Thermostat(0x10) -> Solar(0x2A), ?(0x7AE), data: FF
OFF: [emsesp] Thermostat(0x10) -> Solar(0x2A), ?(0x7AE), data: 00

The setting "Tägl. Aufheizung Temp" (daily heating temperature):
[emsesp] Solar(0x2A) -> All(0x00), SM100wwParam(0x7A6), data: 3E (offset 6)
where "3E" is the set temperature (in this case i set it to 62°C)
=> It seems to be recognized, but I can't find the value

"Tägl. Aufheizung Zeit" (daily heating time):
[emsesp] Thermostat(0x10) -> All(0x00), RC300WWmode(0x2F5), data: 1D (offset 9)
where 1D = 7:15 and 1C = 7:00

"Tägl. Aufheizung" (daily heating) on/off
000+05:21:31.697 N 635: [emsesp] Thermostat(0x10) -> All(0x00), ?(0xF7), data: FF 01 F5 0D (offset 1)
000+05:21:33.420 N 636: [emsesp] Thermostat(0x10) -> All(0x00), RC300WWmode(0x2F5), data: 00 (offset 8)
000+05:21:39.597 N 637: [emsesp] Thermostat(0x10) -> All(0x00), ?(0xF7), data: FF 01 F5 0F (offset 1)
000+05:21:41.145 N 638: [emsesp] Thermostat(0x10) -> All(0x00), RC300WWmode(0x2F5), data: FF (offset 8)

I set it to "off" and then to "on" again.

"Automat. Therm. Desinfekt." on/off value seems to be in ems-esp under thermostat->ww disinfaction
"Therm. Desinfektion Temp" (disinfaction temperature):
[emsesp] Thermostat(0x10) -> Solar(0x2A), SM100wwParam(0x7A6), data: 45 (offset 12)

Torsten-, 2021-12-19:
And the last thing: I searched my whole terminal-log for all lines with "?", so here is a conclusion of all unrecognized packets.

The Mixer seems to have also an unrecognized packet?! If there is something I can look further in, just tell me.

```
torsten@torsten-laptop:~$ grep '?' tmp | cut -d " " -f 4- | sort | uniq
[emsesp] Mixer(0x21) -> All(0x00), ?(0x255), data: 00
[emsesp] Mixer(0x21) -> All(0x00), ?(0xBF), data: 21 A0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[emsesp] Mixer(0x21) <- Thermostat(0x10), ?(0x292), data: 01 (offset 5)
[emsesp] Solar(0x2A) -> All(0x00), ?(0x7A9), data: 4B
[emsesp] Solar(0x2A) -> All(0x00), ?(0x7AD), data: 64 00 00 03
[emsesp] Solar(0x2A) -> All(0x00), ?(0x7AE), data: 00
[emsesp] Solar(0x2A) -> All(0x00), ?(0x7AE), data: FF
[emsesp] Solar(0x2A) -> All(0x00), ?(0x7B1), data: 01 80 00 01 00 00 00 00
[emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 00 00 00 00 00 00 00 03 00 00 00 00 00 00 03 03 03 03
[emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 00 00 00 00 00 02 05 06 0A 00 00 00 00 00 03 03 03 03
[emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 00 00 00 00 00 03 03 06 03 00 00 00 00 00 03 03 03 03
[emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 00 00 00 00 00 03 05 06 05 00 00 00 00 00 03 03 03 03
[emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 00 00 00 00 00 06 06 06 06 00 00 00 00 00 03 03 03 03
[emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 00 00 00 00 01 05 06 06 06 00 00 00 00 00 03 03 03 03
[emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 00 03 (offset 6)
[emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 00 14 00 00 00 06 08 06 08 00 00 00 00 00 03 03 03 03
[emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 00 (offset 1)
[emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 03 (offset 7)
[emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 0A (offset 1)
[emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 14 (offset 1)
[emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 (offset 7)
[emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 (offset 9)
[emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 03 00 (offset 8)
[emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 03 (offset 8)
[emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 05 05 (offset 6)
[emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 05 06 (offset 7)
[emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 05 (offset 6)
[emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 05 (offset 7)
[emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 05 (offset 9)
[emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 06 0A (offset 8)
[emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 06 (offset 8)
[emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 09 09 (offset 6)
[emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 09 (offset 8)
[emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 0A (offset 9)
[emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 0B 0C (offset 6)
[emsesp] Solar(0x2A) -> All(0x00), ?(0xBF), data: 2A A3 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[emsesp] Solar(0x2A) -> All(0x00), ?(0xF7), data: FF 06 AE 01
[emsesp] Solar(0x2A) -> All(0x00), ?(0xF7), data: FF 06 AE 03
[emsesp] Solar(0x2A) -> Thermostat(0x10), ?(0x7A8), data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[emsesp] Solar(0x2A) -> Thermostat(0x10), ?(0x7A8), data: 00 00 00 00 (offset 24)
[emsesp] Solar(0x2A) -> Thermostat(0x10), ?(0x7CC), data: 00 00 04 8F
[emsesp] Solar(0x2A) -> Thermostat(0x10), ?(0xF9), data: FF 06 A6 06 07 00 00 00 3C 00 00 00 3C 00 00 00 50 00 00 00 3C
[emsesp] Solar(0x2A) -> Thermostat(0x10), ?(0xF9), data: FF 06 A6 06 07 00 00 00 3C 00 00 00 3C 00 00 00 50 00 00 00 3E
[emsesp] Solar(0x2A) -> Thermostat(0x10), ?(0xF9), data: FF 06 A6 0C 07 00 00 00 41 00 00 00 46 00 00 00 50 00 00 00 45
[emsesp] Solar(0x2A) -> Thermostat(0x10), ?(0xF9), data: FF 06 A6 0C 07 00 00 00 41 00 00 00 46 00 00 00 50 00 00 00 46
[emsesp] Solar(0x2A) -> Thermostat(0x10), ?(0xF9), data: FF 06 A6 13 47 00 00 00 00 00 00 00 00 00 07 00 02 00 07 00 00
[emsesp] Solar(0x2A) -> Thermostat(0x10), ?(0xF9), data: FF 06 A6 13 47 00 00 00 00 00 00 00 00 00 07 00 02 00 07 00 01
[emsesp] Solar(0x2A) -> Thermostat(0x10), ?(0xF9), data: FF 06 A6 13 47 00 00 00 00 00 00 00 00 00 07 00 02 00 07 00 02
[emsesp] Solar(0x2A) -> Thermostat(0x10), ?(0xF9), data: FF 06 AE 00 CF 00 00 00 FF 00 00 00 00 00 03 00 00 00 03 00 00
[emsesp] Solar(0x2A) -> Thermostat(0x10), ?(0xF9), data: FF 06 AE 00 CF 00 00 00 FF 00 00 00 00 00 03 00 00 00 03 00 FF
[emsesp] Solar(0x30) -> All(0x00), ?(0x368), data: 0C 00
[emsesp] Solar(0x30) -> All(0x00), ?(0x369), data: 00
[emsesp] Solar(0x30) -> All(0x00), ?(0x3A9), data: 80 00
[emsesp] Solar(0x30) -> All(0x00), ?(0x3AA), data: 00 00 03 64 00
[emsesp] Thermostat(0x10) -> All(0x00), ?(0xBF), data: 10 9E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[emsesp] Thermostat(0x10) -> All(0x00), ?(0xF7), data: FF 01 F5 0D (offset 1)
[emsesp] Thermostat(0x10) -> All(0x00), ?(0xF7), data: FF 01 F5 0F (offset 1)
[emsesp] Thermostat(0x10) -> Mixer(0x21), ?(0x292), data: 00 (offset 5)
[emsesp] Thermostat(0x10) -> Mixer(0x21), ?(0x2E2), data: 01 00 00 00 01
[emsesp] Thermostat(0x10) -> Mixer(0x21), ?(0x2E2), data: 01 20 64 00 01
[emsesp] Thermostat(0x10) -> Mixer(0x21), ?(0x2EC), data: 00
[emsesp] Thermostat(0x10) <- Solar(0x2A), ?(0x7A8), data: 03 (offset 24)
[emsesp] Thermostat(0x10) <- Solar(0x2A), ?(0x7A8), data: 18
[emsesp] Thermostat(0x10) -> Solar(0x2A), ?(0x7AE), data: 00
[emsesp] Thermostat(0x10) -> Solar(0x2A), ?(0x7AE), data: FF
[emsesp] Thermostat(0x10) <- Solar(0x2A), ?(0x7CC), data: 04
[emsesp] Thermostat(0x10) <- Solar(0x2A), ?(0xF9), data: 11 FF 06 A6 06
[emsesp] Thermostat(0x10) <- Solar(0x2A), ?(0xF9), data: 11 FF 06 A6 0C
[emsesp] Thermostat(0x10) <- Solar(0x2A), ?(0xF9), data: 11 FF 06 A6 13
[emsesp] Thermostat(0x10) <- Solar(0x2A), ?(0xF9), data: 11 FF 06 AE 00
```

Torsten-, 2021-12-19:
As I said there is a status-text in Diagnose->warm water system

I found the possible "values" as text:
[image]
In my log above there was first the status "Aus" and later when the warm water was flowing (pump to x% an flow to x l/min) the status was "Wär.Anf"

MichaelDvP, 2021-12-20:
I've added the values that you have determined. keepWarm, daily heating temp, disinfection temp and daily heating + time to thermostat. Nice catches.

The other missing values for pump, modulation, time, flow, etc. i can not find in this log. Try to check the diagnostic while logging and not the time  and values of a change, than we can search the telegram and value in the log. 

You have seen that there are a lot of unknown telegrams from water station and in the known telegrams there are only a few values identified. 
Some telegrams with ? are known but igored by emsesp: 
0xBF is for error reporting, these errors are logged by boiler and thermostat and republished with date, thats what we read for 'last error'. 
The 0xF7 and 0xF9 are configuration telegrams for the control display, what values has to be shown with limits (but i don't know how they work exactly). The configured telegram type is in offset 1/2 of these telegrams. 
Mixer 0x255 is always publishing a single zero byte,  Often seen, but no idea what's for.

Torsten-, 2021-12-22:
Hi,
thanks a lot, works great.

I did some more capture:

1) watch filtered on 0x2A (warm water station)
```
001+17:39:01.999 N 62: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 01 61 01 61 80 00 00 80 00 80 00 80 00 00
001+17:39:03.250 N 63: [emsesp] Thermostat(0x10) -> Solar(0x2A), SM100wwCommand(0x7AB), data: 00 00 02
001+17:39:09.769 N 64: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 05 (offset 7)
001+17:39:09.952 N 65: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 0A (offset 9)
001+17:39:10.569 N 66: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 06 (offset 8)
001+17:39:11.049 N 67: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 01 61 01 61 80 00 00 80 00 80 00 80 00 00
001+17:39:12.996 N 68: [emsesp] Solar(0x2A) -> All(0x00), SM100wwTemperature(0x7D6), data: 01 5F
001+17:39:13.180 N 69: [emsesp] Solar(0x2A) -> All(0x00), SM100wwTemperature(0x7D6), data: 01 5F (offset 12)
001+17:39:13.819 N 70: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 05 (offset 9)
001+17:39:19.897 N 71: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7A9), data: 4B
001+17:39:20.266 N 72: [emsesp] Me(0x0B) <- Solar(0x2A), SM100wwCirc(0x7A5), data: 20
001+17:39:20.304 N 73: [emsesp] Solar(0x2A) -> Me(0x0B), SM100wwCirc(0x7A5), data: FF 03 0A 04 00 1E 32
001+17:39:20.716 N 74: [emsesp] Me(0x0B) <- Solar(0x2A), SM100wwParam(0x7A6), data: 20
001+17:39:20.798 N 75: [emsesp] Solar(0x2A) -> Me(0x0B), SM100wwParam(0x7A6), data: FF 05 0F 5F 00 01 3C 3C 3C 32 2D 12 46 00 3C 1E 03 07 3C 00 0F 00 05
001+17:39:21.324 N 76: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 01 61 01 61 80 00 00 80 00 80 00 80 00 00
001+17:39:21.791 N 77: [emsesp] Me(0x0B) <- Solar(0x2A), SM100wwKeepWarm(0x7AE), data: 20
001+17:39:21.815 N 78: [emsesp] Solar(0x2A) -> Me(0x0B), SM100wwKeepWarm(0x7AE), data: FF 0F
001+17:39:28.871 N 79: [emsesp] Solar(0x2A) -> All(0x00), SM100wwTemperature(0x7D6), data: 01 8D
001+17:39:29.056 N 80: [emsesp] Solar(0x2A) -> All(0x00), SM100wwTemperature(0x7D6), data: 01 8D (offset 12)
001+17:39:31.274 N 81: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 01 61 01 61 80 00 00 80 00 80 00 80 00 00
001+17:39:40.796 N 82: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 0A (offset 1)
001+17:39:40.982 N 83: [emsesp] Solar(0x2A) -> All(0x00), SM100wwTemperature(0x7D6), data: 02 0A (offset 4)
001+17:39:42.074 N 84: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 01 61 01 61 80 00 00 80 00 80 00 80 00 00
001+17:39:49.271 N 85: [emsesp] Solar(0x2A) -> All(0x00), SM100wwTemperature(0x7D6), data: 01 F4 (offset 12)
001+17:39:49.847 N 86: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7A9), data: 4B
001+17:39:50.052 N 87: [emsesp] Solar(0x2A) -> All(0x00), SM100wwStatus(0x7AA), data: 00 00 03 00 03 00 32 02 0F
001+17:39:50.627 N 88: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7AD), data: 64 00 00 03
001+17:39:50.826 N 89: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7B1), data: 01 80 00 01 00 00 00 00
001+17:39:51.024 N 90: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 01 61 01 61 80 00 00 80 00 80 00 80 00 00
001+17:39:52.226 N 91: [emsesp] Solar(0x2A) -> All(0x00), SM100wwTemperature(0x7D6), data: 01 B1 00 10 02 11 80 00 80 00 80 00 01 F4
001+17:39:52.457 N 92: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 00 0A 00 00 00 04 05 06 05 00 00 00 00 00 03 03 03 03
001+17:39:53.846 N 93: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 14 (offset 1)
001+17:40:01.249 N 94: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 01 62 01 62 80 00 00 80 00 80 00 80 00 00
001+17:40:03.475 N 95: [emsesp] Thermostat(0x10) -> Solar(0x2A), SM100wwCommand(0x7AB), data: 00 00 02
001+17:40:12.496 N 96: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 1E (offset 1)
001+17:40:12.974 N 97: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 01 62 01 62 80 00 00 80 00 80 00 80 00 00
001+17:40:14.496 N 98: [emsesp] Solar(0x2A) -> All(0x00), SM100wwTemperature(0x7D6), data: 01 BF
001+17:40:19.391 N 99: [emsesp] Me(0x0B) <- Solar(0x2A), SM100wwCirc(0x7A5), data: 20
001+17:40:19.432 N 100: [emsesp] Solar(0x2A) -> Me(0x0B), SM100wwCirc(0x7A5), data: FF 03 0A 04 00 1E 32
001+17:40:19.841 N 101: [emsesp] Me(0x0B) <- Solar(0x2A), SM100wwParam(0x7A6), data: 20
001+17:40:19.923 N 102: [emsesp] Solar(0x2A) -> Me(0x0B), SM100wwParam(0x7A6), data: FF 05 0F 5F 00 01 3C 3C 3C 32 2D 12 46 00 3C 1E 03 07 3C 00 0F 00 05
001+17:40:20.341 N 103: [emsesp] Me(0x0B) <- Solar(0x2A), SM100wwKeepWarm(0x7AE), data: 20
001+17:40:20.365 N 104: [emsesp] Solar(0x2A) -> Me(0x0B), SM100wwKeepWarm(0x7AE), data: FF 0F
001+17:40:20.572 N 105: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7A9), data: 4B
001+17:40:21.024 N 106: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 01 62 01 62 80 00 00 80 00 80 00 80 00 00
001+17:40:31.274 N 107: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 01 62 01 62 80 00 00 80 00 80 00 80 00 00
001+17:40:33.371 N 108: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 28 (offset 1)
001+17:40:41.924 N 109: [emsesp] Boiler(0x08) -> All(0x00), MC110Status(0x2A), data: 00 00 00 00 03 00 00 01 62 01 62 80 00 00 80 00 80 00 80 00 00
001+17:40:50.698 N 110: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7A9), data: 4B
[... cut ...]
- Pump 0 %
- flow 1.6 l/min
- Status "keine Anf"

around 001+17:59:12.683
- Pump 0 %
- flow 0.0 l/min
- Status "keine Anf"

The runtime of the pump was 21 h and didn't change - I'll try to caputer a change later if it is not in te log.

One last thing: "ww daily temperature" is for the "daily heating" function - the other parameters seems to be at the thermostat, this temperature at the module - maybe it's better to keep them together and to name them similar? (like ww daily heating temperature)

MichaelDvP, 2021-12-23:
I think the pump is in 0x7E0 offset 9 and in offset 8 bit2 for on/off. 
If pump is running an speed changes there are single publishes of offset 9, if pump starts/stops it is published with offset 8, 2 bytes (including offset 9). Maybe you can verify tht with `watch 7E0` 
For the other value i have no clear idea. 7E0 offset 1 seems to be a changing 16bit value, but i do not see the scaling for flow.

Torsten-, 2022-01-07:
7E0 offset 9 seems to be "pump in %". I can confirm that:

```
=> no flow
017+19:12:49.783 N 50: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 00 00 00 00 00 00 00 03 00 00 00 00 00 00 03 03 03 03
017+19:13:50.533 N 51: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 00 00 00 00 00 00 00 03 00 00 00 00 00 00 03 03 03 03
017+19:14:51.233 N 52: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 00 00 00 00 00 00 00 03 00 00 00 00 00 00 03 03 03 03
017+19:15:52.609 N 53: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 00 00 00 00 00 00 00 03 00 00 00 00 00 00 03 03 03 03
=> flow started - pump 10%, then 5%, flow-rate 0,0 to 1,1 to 2,4
017+19:16:10.920 N 54: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 05 (offset 7)
017+19:16:11.103 N 55: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 0A (offset 9)
017+19:16:11.670 N 56: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 06 (offset 8)
017+19:16:13.970 N 57: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 05 (offset 9)
017+19:16:45.047 N 58: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 0A (offset 1)
017+19:16:53.458 N 59: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 00 0A 00 00 00 03 05 06 05 00 00 00 00 00 03 03 03 03
017+19:17:12.323 N 60: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 14 (offset 1)
017+19:17:54.509 N 61: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 00 14 00 00 00 03 05 06 05 00 00 00 00 00 03 03 03 03
017+19:18:10.996 N 62: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 06 (offset 6)
017+19:18:29.523 N 63: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 0A (offset 1)
017+19:18:55.285 N 64: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 00 0A 00 00 00 06 07 06 07 00 00 00 00 00 03 03 03 03
017+19:18:57.824 N 65: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 00 (offset 6)
017+19:18:58.006 N 66: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 (offset 9)
017+19:18:58.596 N 67: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 03 (offset 8)
017+19:18:58.780 N 68: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 00 (offset 1)
017+19:19:25.824 N 69: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 09 09 (offset 6)
017+19:19:26.006 N 70: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 0A (offset 9)
017+19:19:27.196 N 71: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 06 (offset 8)
017+19:19:29.121 N 72: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 05 (offset 9)
=> flow stopped
017+19:19:56.334 N 73: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 00 00 00 00 00 05 05 06 05 00 00 00 00 00 03 03 03 03
017+19:20:22.124 N 74: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 04 04 (offset 6)
017+19:20:22.309 N 75: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 03 00 (offset 8)
017+19:20:58.711 N 76: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 00 00 00 00 00 00 00 03 00 00 00 00 00 00 03 03 03 03
017+19:21:59.435 N 77: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 00 00 00 00 00 00 00 03 00 00 00 00 00 00 03 03 03 03
017+19:22:59.810 N 78: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 00 00 00 00 00 00 00 03 00 00 00 00 00 00 03 03 03 03
```

Some other idea for offset 8: maybe it's the status?

I started a "manual thermal desifection" and I got:
```
017+19:31:28.060 N 87: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 08 (offset 8)
```

In the log above I just saw changing from "keine Anf" (no demand / off) to "Wär Anf" (demand / system running, flow active) and back to "keine Anf".
maybe 00 == keine Anf and 06 == Wär Anf and 08 == Therm.D.
but what about 03 in the log above?! I didn't see another status text.

[image]

Torsten-, 2022-01-07:
Ah, offset 8 is/was never 00 ..
03 == "keine Anf"
06 == "Wär.Anf"
08 == "Therm.D."

Now I'm pretty sure.

Torsten-, 2022-01-07:
Ok, new finding .. there are two "Status" fields in diagnose screen :-/

1. Status - is not captured with 7E0
2. Status is just like mentioned above (03, 06, 08)

Another nice caputer - this time it is about offset 1
Just an idea: maybe it is a counter of water released water?
It starts with 00 00 and just counts up. After turning the tap off it gets reset to 00 00.
In Diagnose the flow in l/min takes a while until a number shows up -> maybe because the calculate it from offset 01 divided by time?

Example 1 (flow in Display was around 2,2 l/min):
```
017+20:14:36.763 N 138: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 0A (offset 1)
017+20:14:47.313 N 139: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 14 (offset 1)
017+20:14:50.675 N 140: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 00 14 00 00 00 05 07 08 05 01 00 00 00 00 03 03 03 03
017+20:14:58.214 N 141: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 1E (offset 1)
017+20:14:58.396 N 142: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 09 (offset 7)
017+20:15:06.188 N 143: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 28 (offset 1)
017+20:15:15.489 N 144: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 32 (offset 1)
017+20:15:24.188 N 145: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 3C (offset 1)
017+20:15:31.613 N 146: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 46 (offset 1)
017+20:15:38.864 N 147: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 50 (offset 1)
017+20:15:39.986 N 148: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 0E (offset 7)
017+20:15:46.489 N 149: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 5A (offset 1)
017+20:15:51.850 N 150: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 00 5A 00 00 01 05 0F 08 05 01 00 00 00 00 03 03 03 03
017+20:15:53.839 N 151: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 64 (offset 1)
017+20:16:01.213 N 152: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 6E (offset 1)
017+20:16:08.539 N 153: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 78 (offset 1)
017+20:16:14.864 N 154: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 82 (offset 1)
017+20:16:15.437 N 155: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 13 (offset 7)
017+20:16:21.714 N 156: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 8C (offset 1)
017+20:16:27.789 N 157: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 96 (offset 1)
017+20:16:34.914 N 158: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 A0 (offset 1)
017+20:16:40.714 N 159: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 AA (offset 1)
017+20:16:46.914 N 160: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 B4 (offset 1)
017+20:16:47.487 N 161: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 18 (offset 7)
017+20:16:52.851 N 162: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 00 B4 00 00 01 05 18 08 05 01 00 00 00 00 03 03 03 03
017+20:16:54.439 N 163: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 BE (offset 1)
017+20:17:00.389 N 164: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 C8 (offset 1)
017+20:17:06.164 N 165: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 D2 (offset 1)
017+20:17:11.714 N 166: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 DC (offset 1)
017+20:17:19.012 N 167: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 1D (offset 7)
017+20:17:19.198 N 168: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 E6 (offset 1)
017+20:17:25.164 N 169: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 F0 (offset 1)
017+20:17:31.440 N 170: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 FA (offset 1)
017+20:17:36.764 N 171: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 01 04 (offset 1)
017+20:17:42.964 N 172: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 01 0E (offset 1)
017+20:17:48.939 N 173: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 01 18 (offset 1)
017+20:17:49.512 N 174: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 22 (offset 7)
017+20:17:53.900 N 175: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 01 18 00 00 01 05 22 08 05 01 00 00 00 00 03 03 03 03
017+20:17:54.689 N 176: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 01 22 (offset 1)
017+20:18:00.689 N 177: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 01 2C (offset 1)
017+20:18:07.039 N 178: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 01 36 (offset 1)
017+20:18:08.589 N 179: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 00 (offset 1)
017+20:18:09.362 N 180: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 (offset 7)
```

Example 2 (flow in Display was around 3,6 - 3,8 l/min):
```
017+20:24:48.789 N 187: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 09 (offset 7)
017+20:24:50.764 N 188: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 09 (offset 6)
017+20:25:02.202 N 189: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 00 00 00 00 00 09 09 08 05 01 00 00 00 00 03 03 03 03
017+20:25:17.741 N 190: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 0A (offset 1)
017+20:25:25.466 N 191: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 14 (offset 1)
017+20:25:31.291 N 192: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 1E (offset 1)
017+20:25:37.541 N 193: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 28 (offset 1)
017+20:25:37.724 N 194: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 0E (offset 7)
[... cut ...]
017+20:26:21.439 N 205: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 18 (offset 7)
017+20:26:24.816 N 206: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 8C (offset 1)
017+20:26:29.441 N 207: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 96 (offset 1)
017+20:26:30.292 N 208: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 00 (offset 6)
017+20:26:30.965 N 209: [emsesp] Solar(0x2A) -> All(0x00), ?(0x7E0), data: 00 00 (offset 1)
```

No time left now - I'll try to get more status-codes later.

Torsten-, 2022-01-08:
> Ah, offset 8 is/was never 00 .. 03 == "keine Anf" 06 == "Wär.Anf" 08 == "Therm.D."
> 
> Now I'm pretty sure.
> 
> I'll try to get "Tägl. Aufh" and "Warmh." For the others I don't know how to force them manually.

For the Status of 7E0 I got:
03 == "keine Anf" 
06 == "Wär.Anf" 
08 == "Therm.D."
09 == "Warmh"

I think that's all I'm able to reproduce. If you need more info for each status (what it does) just say a word.

For the second status I found I'll have to do more investigation - no clue where to find it in the Hey-Dumps. Maybe it is from Thermostat?! I'll try to find out later.

proddy, 2022-01-19:
Do you think you could summerize all the new values you found into one table, so we can implement them?

pschaumburg, 2022-01-22:
This is great. I'll need this as well :D 
@proddy What do you need specifically? When @Torsten- does not have the time to create such a table, I might be able to when I know what exactly is needed? 
(I have another device which can not be detected as well (0x7EA), but this will be then another issue in this repo later on.)

MichaelDvP, 2022-01-22:
My suggestion:
- load the latest v3.4 software to emsesp, as it contains most of the identified values.
- go through the settings of the thermostat and write down the parameter and value names to a table, for settings also note min/max limits.
- if you know the telegram/offset list it in a extra column
- now check in emsesp and note if the value is wrong/missing/wrong named/not setable/etc.

maybe looks like this.
```
name              |  value   | min/max (if setable) | telegram/offset | emsesp v3.4.0a17
-----------------------------------------------------------------------------
flow              |  2.2 l/m | -                    |  0x7E0 / 7      | value displayed  22, wrong uints.
disinfection temp | 70C      | 60 / 90              |  0x7A6 / 12     | read and write ok
pump2 modulation  | 10%      |                      | unknown         | missing
 ```

closed 2024-09-15 by proddy