emsesp/EMS-ESP32 #251: Boiler Bosch Condens 8700i - error code H11/1021

Nunak, 2021-12-20:
*Before creating a new issue please check that you have:*

* *searched the existing [issues](https://github.com/emsesp/EMS-ESP32/issues) (both open and closed)*
* *searched the [documentation help section](https://emsesp.github.io/docs)*

*Completing this template will help developers and contributors help you. Try to be as specific and extensive as possible. If the information provided is not enough the issue will likely be closed.*

*You can now remove this line and the above ones. Text in italic is meant to be replaced by your own words. If any of the sections below are not relevant to the issue (for instance, the screenshots) then you can delete them.*

**Question**
*A clear and concise description of what the problem/doubt is.*
I have question. My boiler random getting error "H11/1021 start: 2021.12.19 13:18, end: none" by Bosch service I got information that temperature sensor is fine.
My question is if this could be related to emsesp somehow ? I mean if is possible that emsesp could somehow inject this error code to boiler ?
The boiler is brand new and getting this error random but often and only solution is restart boiler. Reset errors from boiler will not reset this error.

**Screenshots**
*If applicable, add screenshots to help explain your problem.*
[image]

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information from http://ems-esp.local/api?device=system&cmd=info*
EPS32, version v3.3.1b1

**Additional context**
*Add any other context about the problem here.*

proddy, 2021-12-20:
EMS-ESP is just reading the values. There are no write operations so don't expect EMS-ESP is interfering with the Boiler. Has this suddenly happened?

bbqkees, 2021-12-20:
This sounds like an older issue with the outdoor sensor and the RC310 thermostat.
Is it the outdoor sensor that gives the error?

Nunak, 2021-12-20:
> EMS-ESP is just reading the values. There are no write operations so don't expect EMS-ESP is interfering with the Boiler. Has this suddenly happened?

Yep it's happening suddenly I didn't found any interaction which could cause this .

Nunak, 2021-12-20:
> This sounds like an older issue with the outdoor sensor and the RC310 thermostat. Is it the outdoor sensor that gives the error?

I have termostat Bosch CW200 with sensor outside. But what I found is a input sensor of heating water in boiler.

bbqkees, 2021-12-20:
How often do you get the error?
If you remove EMS-ESP from your system, do you still get the error?
How long have you tested it?

Nunak, 2021-12-20:
> How often do you get the error? If you remove EMS-ESP from your system, do you still get the error? How long have you tested it?

It's happening very randomly sometime twice per day sometime one per 3 days.
I didn't tested yet to remove EMS-ESP from the system.
I have EMS-ESP for years and this happenning only with the new boiler Bosch conedens 8700 with previous boiler Junkers ZBS 14 was without errors with the same termostat CW200 from bosch

Due to I am asking if this could happen due to EMS-ESP.

bbqkees, 2021-12-20:
Try out by running your boiler for a few days without EMS-ESP connected.
If the error goes away, we should have a better look at it. If the error persists it's the boiler which is faulty.

MichaelDvP, 2021-12-20:
I've found H11/1021 is a faulty dhw sensor, the error is priority 16, low, reset only by reparing the fault.

What does emsesp shows as `ww current intern temperature`  and ww storage temperature in fault state?

Last error shows the the first entry of error log from the boiler, restarting the boiler does not help, it's only that emsesp takes a while to read the information, but last error will reappear. Can you check the complete log by telnet command `read 8 c2`, i think we can see a list of the faults with starting times.

Nunak, 2021-12-25:
> I've found H11/1021 is a faulty dhw sensor, the error is priority 16, low, reset only by reparing the fault.
>
> What does emsesp shows as `ww current intern temperature` and ww storage temperature in [...]

Sorry for late. There are answers

esmesp is showing temperature from "not working" sensor and temperature is changing
[image]

There is output from  `read 8 c2`
```
ems-esp:$ read 8 c2
000+00:15:17.437 N 0: [emsesp] Boiler(0x08) -> Me(0x0B), UBAErrorMessage3(0xC2), data: 08 C3 00 10 11 48 31 31 03 FD 95 0C 0F 18 32 00 00 00 00 00 08 C3 00 10 11 48 31
000+00:15:17.653 N 1: [emsesp] Boiler(0x08) -> Me(0x0B), UBAErrorMessage3(0xC2), data: 31 03 FD 95 0C 0D 13 12 95 0C 10 14 21 08 C3 00 10 11 48 31 31 03 FD 95 0C 14 12 (offset 27)
000+00:15:17.823 N 2: [emsesp] Boiler(0x08) -> Me(0x0B), UBAErrorMessage3(0xC2), data: 15 95 0C 15 12 27 08 C3 00 10 11 48 31 31 03 FD 95 0C 12 11 26 95 0C 15 11 09 08 (offset 54)
000+00:15:18.030 N 3: [emsesp] Boiler(0x08) -> Me(0x0B), UBAErrorMessage3(0xC2), data: C3 00 10 11 48 31 31 03 FD 95 0C 06 10 38 95 0C 09 10 21 08 C3 00 10 11 48 31 31 (offset 81)
000+00:15:18.245 N 4: [emsesp] Boiler(0x08) -> Me(0x0B), UBAErrorMessage3(0xC2), data: 03 FD 95 0C 0A 0F 15 95 0C 11 0F 0D 08 C3 00 10 11 48 31 31 03 FD 95 0C 13 0D 12 (offset 108)
000+00:15:18.417 N 5: [emsesp] Boiler(0x08) -> Me(0x0B), UBAErrorMessage3(0xC2), data: 95 0C 07 0E 2C 08 C3 00 10 11 48 31 31 03 FD 95 0C 06 0D 2E 95 0C 09 0D 22 08 C3 (offset 135)
000+00:15:18.619 N 6: [emsesp] Boiler(0x08) -> Me(0x0B), UBAErrorMessage3(0xC2), data: 00 10 11 48 31 31 03 FD 95 0C 0A 0B 18 95 0C 11 0B 19 08 C3 00 10 11 48 31 31 (offset 162)
ems-esp:$
```

Now I will try reset this new error and disconnect emsesp.

Thank you for your support!

MichaelDvP, 2021-12-25:
Here is the decoded error-log, always the same error, nearly daily:
```
boiler 08, ID C3 00, class 10(16dez), ?,  Code: H11/1021
08 C3 00 10 11 48 31 31 03 FD 95 0C 0F 18 32 00 00 00 00 00  2021/12/24 15:50 - now
08 C3 00 10 11 48 31 31 03 FD 95 0C 0D 13 12 95 0C 10 14 21  2021/12/19 13:18 - 2021/12/20 16:33
08 C3 00 10 11 48 31 31 03 FD 95 0C 14 12 15 95 0C 15 12 27  2021/12/18 20:21 - 2021/12/18 21:39
08 C3 00 10 11 48 31 31 03 FD 95 0C 12 11 26 95 0C 15 11 09  2021/12/17 18:38 - 2021/21/17 21:09
08 C3 00 10 11 48 31 31 03 FD 95 0C 06 10 38 95 0C 09 10 21  2021/12/16 06:56 - 2021/12/16 09:33
08 C3 00 10 11 48 31 31 03 FD 95 0C 0A 0F 15 95 0C 11 0F 0D  2021/12/15 10:21 - 2021/12/15 17:13
08 C3 00 10 11 48 31 31 03 FD 95 0C 13 0D 12 95 0C 07 0E 2C  2021/12/13 19:18 - 2021/12/14 07:44
08 C3 00 10 11 48 31 31 03 FD 95 0C 06 0D 2E 95 0C 09 0D 22  2021/12/13 06:46 - 2021/12/13 09:34
08 C3 00 10 11 48 31 31 03 FD 95 0C 0A 0B 18 95 0C 11 0B 19  2021/12/11 10:24 - 2021/12/11 17:25
```

Afaik the boiler also checks if the sensor data points to a bad thermal contact or wrong sensor position. If ww heating is active and the sensor-measurement does not rise instantly, the error is set. 
Check the data if the start-times are correlated with ww-heating. Also check if the end-times are your manual resets.

Nunak, 2022-01-14:
Many thanks for your support. You were right. Error was ocuring also with disconnected emsesp board.

Issue was on the boiler on this part
[image]

What's interesting bosch know about this wrong part on every new boiler but they do not have enough of this spare part to replace it :o

MichaelDvP, 2022-01-14:
Do you know what this part is doing? Or where it waas placed. I can't find part no. 200 79R04 in web.

Nunak, 2022-01-14:
> Do you know what this part is doing? Or where it waas placed. I can't find part no. 200 79R04 in web.

Yep, not sure about the correct name but it's a code plug (socket) which is connected to main board of boiler. Connected from the right side. If you want  I can take a photo.

MichaelDvP, 2022-01-15:
> If you want I can take a photo. 

If it's not much effort, please do. 
Also your first picture shows a unknown device (i think it's the controller), can post here the device information like product-ID, etc.  Then we can add it to the database.

Nunak, 2022-01-15:
> > If you want I can take a photo.
> 
> If it's not much effort, please do. Also your first picture shows a unknown device (i think it's the controller), can post here the device information like product-ID, etc. Then we can add it to the database.

Sure there is picture
[image]

About the unkown controller that would be great.
There is picture of boiler
[image]
I am using this boiler Condens 8700i W GC8700iW 30/35 C23. For controller I am using Junkers CW400
[image]
Or you mean BBQ board  ? I did [Order #923] (02/01/2020)

Thank you.

MichaelDvP, 2022-01-15:
Thanks, i've meant the device information shown on emsesp web page, the first sceenshot you've posted shows a unknown device: with mouse over you get the device information. Or download the system-info from help-page.

Nunak, 2022-01-15:
> Thanks, i've meant the device information shown on emsesp web page, the first sceenshot you've posted shows a unknown device: with mouse over you get the device information. Or download the system-info from help-page.

There is system-info
[emsesp_system_info.txt](https://github.com/emsesp/EMS-ESP32/files/7874781/emsesp_system_info.txt)

MichaelDvP, 2022-01-15:
Thanks, as expected it's the controller (device-ID 0x09). We'll add the product id to database. Controllers do not have values, there is nothing more to do.

closed 2023-02-19 by proddy