emsesp/EMS-ESP32 #252: Syslog has double entries (hex and non-hex) for read requests

tp1de, 2021-12-21:
When issuing a read request (console or api) the syslog contains always 2 entries.
This is Independent from ems-esp settings. (syslog as hex marked or not).  
ems-esp settings are ignored. The telegram "0B 90 FF 00 20 01 C3" returns in syslog:

[image]

BTW: 1C3 (2C3) is the switch-program A for hc1 which I tried to read.

tp1de, 2021-12-21:
When issuing the read requests by API V3 for 1C3,  syslog contains the 4 telegrams (85 byte length in 4 parts) and the read request 4 times as well even when send only once:
[image]

Any reason why? Shouldn't syslog contain just all the telegrams on the bus?

MichaelDvP, 2021-12-21:
Loglevels are hirarchical, the level trace includes all lower levels. If you have syslog trace (raw) you will see the telegrams raw and all messages from lower levels. A read command shows the result in decoded format in level notice, thats included in syslog additional to the normal telegram.

The read command normally shows the complete telegram. if it contains more than 25/27 bytes the higher offsets will be requested. If you only want to read a part of the telegram you have to set the length to a smaller value. Try with `0B 90 FF 00 14 01 C3` to to read 20 bytes and you will get only a single answer.

tp1de, 2021-12-21:
@MichaelDvP 
For ems-esp settings I can decide to get within syslog telegrams with text or hex.
It's either or (not both in parallel). 

When sending a read command I get both telegram types in parallel as shown above. This is confusing.
Don't you think that this should be changed and ems-esp settings should be applied for read requests as well?
(read request telegram itself and answers) 

Yes I understand to get multiple telegrams for longer data content.
By sending only one read request with length 0x20 are the other read-requests generated by ems-esp?

MichaelDvP, 2021-12-21:
There is only one logger (not log4j ;-)) getting the messages from emsesp and sending the selected levels to telnet/weblog/syslog. A message generated in emsesp will show up in all these logs. If you select the raw format all incoming telegrams are show raw in telnet, syslog and weblog. The read command generates an extra logmessage in readable format, independend of incoming messages. If telnet is in log info mode and syslog in trace and you enter a read command in telnet you want to see the response, but this causes double message in syslog.

proddy, 2021-12-21:
(love the log4j comment!! I've spent many hours these last days battling with that in my company)

tp1de, 2021-12-21:
@MichaelDvP Thanks for clarifying but I have difficulties in understanding what you mean with syslog trace.
Is this something which depends on ems-esp settings?  .. or an attribute of syslog server?

To explain what I am doing:
I imbedded a simple syslog server within my ioBroker adapter which reads all telegrams and searches for new data fields or unknown telegram types  like 1C3. The fields I was looking for so far where published regularly. No read requests needed.
But 1C3 needs regular read polling because it is send only once on system start and on value change.
Since km200 is integrated and starts with system/boiler it does not need read polling.

closed 2023-02-19 by proddy