emsesp/EMS-ESP32 #254: Question: Format of read request telegram for EMS+ telegrams

tp1de, 2021-12-21:
I took me some time to recognize that read-requests for ems+ telegrams differ from ems.
As far as I have understood they contain a 0x20 before  the telegram type. E.g type 1C3 from thermostat (0x10):

0B 90 FF 00 20 01 C3

Do all read request for ems+ have to have the 0x20 or does the value differ per telegram type?

MichaelDvP, 2021-12-21:
This is the requested length. ems+ telegram reads are `<src><dst>0xFF<offset><length><id_high><id_low>`
for ems its:  `<src><dst><id><offset><length>`

tp1de, 2021-12-21:
Thanks @MichaelDvP for clarifying. In my case of 1C3 the length 0x20 means the complete content (4 parts).
Is 0x20 the maximum data-length within a telegram?

MichaelDvP, 2021-12-21:
No, the 0x20 is historical and not really correct. A single telegram on the bus may have up to 32 bytes, so the idea was to request all bytes. But we request databytes and the telegram contains a header of 4(ems) or 6(ems+) bytes + crc, So the max datalength is 25 or 27 bytes.
Requesting 32 bytes will result in a complete telegram with max. 25/27 databytes. If we get all these bytes in a read we request the next part until the telegram is shorter.
But we have seen telegrams that responde to all offsets up to 255 (id 0xC2) making the logic to read the next part rolling over.

tp1de, 2021-12-21:
@MichaelDvP A very good explanation thanks.

closed 2023-02-19 by proddy