emsesp/EMS-ESP32 #256: Feature: upload customization settings from a file

proddy, 2021-12-21:
In v3.4 there is new customization service that currently contains
 - sensor aliases and temperature offsets
 - list of device entities that are excluded, from #206 

It'll be nice to be able to download and upload these settings.

proddy, 2021-12-28:
I was thinking
 - add a label to each of the files ditacting what type it is, `customizations`, `settings`
 - We the file is uploaded it's stored on the FileSystem prefixed with a special tag, like '.new'
 - When EMS-ESP starts it looks for any of these new files and renames them (if possible in LittleFS)

tp1de, 2021-12-30:
@proddy 
Good ideas. Especially being able to give some attributes / data structure to all the fields.
I believe It would be very convenient to filter and/or rename fields as necessary.

Would it be possible to define "own data-fields" by definition file to be read within ems-esp?
I have done it within my ioBroker Adapter by analyzing the hex syslog by imbedded syslog-server, filtering the data and api/send requests for updates.
Parameterset (under construction) looks like this:
[image]
 
I am using this to test new telegram-types and data-fields. What do you think, would it be possible to implement within ems-esp as well or is cpu-power and/or timing an issue?

proddy, 2021-12-30:
I would prefer to hardcode the values into EMS-ESP, if they are useful to others.

Renaming device entities (what you call fields) is on my to-do list for the new customization service, like what it does now in renaming dallas and analog sensors (there's an alpha build on Discord you can try out).

And I do want to extend each device entity with a real min & max value which would help validate the settings in the WebUI and also from Home Assistant.

tp1de, 2021-12-30:
> I would prefer to hardcode the values into EMS-ESP, if they are useful to others.

I agree, but for testing new device entities and / or telegram types it might be useful to have such a function for temporary use. (at least for me)

proddy, 2022-05-06:
I have a working version of uploading settings files - I'll make a PR. If it's good enough we'll add to v3.4

proddy, 2022-05-07:
@MichaelDvP did a PR (https://github.com/emsesp/EMS-ESP32/pull/494). I had to touch quite a few files so please take a look. Also tidied up a few things and fixed small bugs.

Key changes:
 - the export of the settings and customizations files are in JSON and is an exact representation of the internal class, so all passwords etc are visible. There is no translation so it's not very readable, but I think that's ok since that's why we have the WebUI for.
 - Renamed Firmware Upload to just Upload and renamed the classes & filenames
 - You can only drop a .bin or .json file into the Upload box, any other file type is ignored
 - Axios error messages weren't showing correctly, so fixed that

MichaelDvP, 2022-05-08:
A lot to review, this will take some time. First check:
- settings down/upload works good. 
- Settings command now should  have `CommandFlag::ADMIN_ONLY`, we can see all passwords in [ems-esp.local/api/system/settings](ems-esp.local/api/system/settings)
- asking for settings here is now dangerous, maybe add a safe version to info?
- customization upload does not work, the new file have an extra nest. Uploading and downloading again results in: 
 ```
{
  "type": "customizations",
  "Customizations": {
    "type": "customizations",
    "Customizations": {
 ```
instead of replacing the file, i use the same logic as in settings, this works:
```
                JsonObject section_json = input["Customizations"];
                if (section_json) {
                    LittleFS.remove(EMSESP_CUSTOMIZATION_FILE);
                    File section_file = LittleFS.open(EMSESP_CUSTOMIZATION_FILE, "w");
                    if (section_file) {
                        serializeJson(section_json, section_file);
                        section_file.close();
                    }
                }
```

proddy, 2022-05-08:
Thanks for checking so quickly. Good spot on adding the `CommandFlag::ADMIN_ONLY`. I was initially thinking of maybe encrypting the file but then you wouldn't be able to make custom edits, and that's a good benefit to have for tweaking or loading defaults (bbqkees would love this). Another option is to build a new command called 'debug' which outputs only the relevant settings which users can attach to GitHub issues when they need help? what do you think?

proddy, 2022-05-08:
Actually, why not use "System Settings" for when users to send debug info, and remove the necessary things like
 - Network (apart from maybe whether its Wifi or ETH), AP stuff, MQTT stuff
and add only the important things we need to troubleshoot like
 - board profile, txmode and most stuff from emsesp_settings

I think that's the better idea
 - which services are active (AP, NTP, OTA, MQTT)

proddy, 2022-05-08:
I'm going to add this and merge the PR - cos I'll be away again for the next 10 days so want to get it pushed. we can fine tune later

MichaelDvP, 2022-05-08:
One last thing: [remove this line](https://github.com/emsesp/EMS-ESP32/blob/85f66287187dfb3978ff03e8055c080a5535e05b/src/system.cpp#L919) to close and remove the temp file (reboot_required is false)

closed 2022-05-20 by proddy