emsesp/EMS-ESP32 #257: Setting selflowtemp not working on V3.3.1b1

Imaginous, 2021-12-22:
**Question/bug?**
Since I have upgraded to an ESP32 and send *selflowtemp* commands instead of the former *flowtemp* command it does not seem to work. 

There also seems to be a correlation between *wwseltemp* and *selflowtemp*. As soon as I set the *wwseltemp* the *selflowtemp* value jumpts to 65, this is the maximum temperature set on the boiler with the physical dail.

When I  the *selflowtemp* command to let's say 45 nothing happens, it stays at 0.0. But when I set a value of 7, it jumps to 65 and can't be changed anymore.

The *setflowtemp* value is 90 in those cases.

With the old V2.2.x firmware I was able to control the boiler without any problems.

**Device information**
`{
  "System": {
    "version": "3.3.1b1",
    "uptime": "000+00:04:20.935",
    "uptime (seconds)": 260,
    "freemem": 167,
    "reset reason": "RTC watch dog reset: CPU+RTC / APP CPU reseted by PRO CPU",
    "Dallas sensors": 0
  },
  "Network": {
    "connection": "WiFi",
    "hostname": "ems-esp-nefit",
    "SSID": <LEFT OUT FOR SAFETY>,
    "BSSID": <LEFT OUT FOR SAFETY>,
    "RSSI": -40,
    "MAC": <LEFT OUT FOR SAFETY>,
    "IPv4 address": "192.168.0.24/255.255.255.0",
    "IPv4 gateway": "192.168.0.1",
    "IPv4 nameserver": "192.168.0.1"
  },
  "Status": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "telegrams received": 319,
    "read requests sent": 59,
    "write requests sent": 2,
    "incomplete telegrams": 0,
    "tx fails": 0,
    "rx line quality": 100,
    "tx line quality": 100,
    "MQTT": "disconnected",
    "MQTT publishes": 10,
    "MQTT publish fails": 0,
    "Dallas reads": 0,
    "Dallas fails": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "BK13/BK15/Smartline/GB1x2 (DeviceID:0x08, ProductID:64, Version:03.05)",
      "handlers": "0x10 0x11 0xC2 0x14 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "Thermostat",
      "name": "Moduline 400 (DeviceID:0x10, ProductID:78, Version:03.01)",
      "handlers": "0xA3 0x06 0xA2 0x12 0x41 0xA7"
    },
    {
      "type": "Controller",
      "name": "BC10/RFM20 (DeviceID:0x09, ProductID:68, Version:02.03)"
    }
  ]
}`

**Additional context**
Might be related to some missing old bits in the newer firmware?

Log seems to respond fine:
`2021-12-22 11:28:21.435 I 9: [command] Calling boiler command 'selflowtemp', value 45, id is default
2021-12-22 11:28:21.435 I 10: [boiler] Setting boiler flow temperature to 45 C
2021-12-22 11:29:24.350 I 11: [command] Calling boiler command 'selflowtemp', value 7, id is default
2021-12-22 11:29:24.350 I 12: [boiler] Setting boiler flow temperature to 7 C`

proddy, 2021-12-22:
I'll need to check the code. So the selflowtemp command with value 45 worked in v2.2?

Imaginous, 2021-12-22:
Yes it worked in with the old EP8266 and firmware V2.2

I just checked again and the command does not work at all. I thought it did, but appearantly the boiler jumps to the max set flow temp with the boiler knob after a minute. Setting the value does not work.

MichaelDvP, 2021-12-22:
@Imaginous You have a thermostat that controls the heating and sets the boiler flowtemp. If you try to set flowtemp in the boiler it is instantly overwritten by thermostat. Also there is a bit confusion with setflowtemp and selflowtemp, the writeable setpoint is setflowtemp and that is written by the command, but the boiler publishes the actual internal setpoint as selflowpoint.

Imaginous, 2021-12-22:
@MichaelDvP it used to work.
I turn off the thermostat, then I can control it.

Once in a while the value is overwritten, but as soon as I detect it I will rewrite the value.

But since the new firmware (and ESP32) it does not work anymore.

MichaelDvP, 2021-12-23:
Sorry, i can not reproduce, the code in v2 and v3 seems to be consistent and setting seems to work for me, but my thermostat instantly overwrites:
```
ems-esp32:# watch 1a
ems-esp32:# call boiler selflowtemp 62
000+00:18:16.840 I 4: [boiler] Setting boiler flow temperature to 62 C
000+00:18:16.944 N 5: [emsesp] Me(0x0B) -> Boiler(0x08), UBASetPoints(0x1A), data: 3E
000+00:18:18.658 N 6: [emsesp] Thermostat(0x10) -> Boiler(0x08), UBASetPoints(0x1A), data: 42 64 00 00
000+00:18:20.131 I 7: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorFast(0x18), data: 42 03 07 64 00 01 01 00 00 01 D6 7D 00 80 00 00 00 FF 30 59 00 00 FF 00 00 83 00
000+00:18:20.412 I 8: [emsesp] Boiler(0x08) -> Me(0x0B), UBAMonitorFast(0x18), data: 00 00 00 00 00 00 00 4C 00 00 (offset 27)
```
Maybe your boiler is different?

Imaginous, 2021-12-23:
I will try to disconnect the thermostat and try again to see if it will work.

I will also check if the overwrite happens directly.
I will keep you informed.

proddy, 2021-12-28:
thanks. We do get this question a lot so I think at some point it needs writing up in the wiki documentation

proddy, 2022-01-08:
@Imaginous any update on this?

Imaginous, 2022-01-08:
It seems to kind of work.
I think it's something with my rather old boiler.
If I send a couple of selflowtemp commands the boiler does not accept any commands any more.

Can't quite figure out why and why it worked better with the old ESP. 
This also happens when the thermostat is disconnected.

I think this can be closed for now.

closed 2023-02-19 by proddy