emsesp/EMS-ESP32 #264: MQTT Discovery: Device class is not being provided for some temperature values

MizterB, 2021-12-23:
**Bug description**
When using Home Assistant MQTT discovery, the device class is not being provided for some temperature values.
*A clear and concise description of what the bug is. Mention which EMS-ESP version you're using.*

According to the Home Assistant logs, lemperature conversion for entities without correct device_class is deprecated and will be removed from Home Assistant Core 2022.3.

HA logs:
```
2021-12-23 14:43:05 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.boiler_selected_flow_temperature (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) with device_class None reports a temperature in °C which will be converted to °F. Temperature conversion for entities without correct device_class is deprecated and will be removed from Home Assistant Core 2022.3. Please update your configuration if device_class is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2021-12-23 14:43:05 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.boiler_heating_temperature (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) with device_class None reports a temperature in °C which will be converted to °F. Temperature conversion for entities without correct device_class is deprecated and will be removed from Home Assistant Core 2022.3. Please update your configuration if device_class is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2021-12-23 14:43:05 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.boiler_hysteresis_on_temperature (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) with device_class None reports a temperature in °C which will be converted to °F. Temperature conversion for entities without correct device_class is deprecated and will be removed from Home Assistant Core 2022.3. Please update your configuration if device_class is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2021-12-23 14:43:05 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.boiler_hysteresis_off_temperature (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) with device_class None reports a temperature in °C which will be converted to °F. Temperature conversion for entities without correct device_class is deprecated and will be removed from Home Assistant Core 2022.3. Please update your configuration if device_class is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2021-12-23 14:43:05 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.boiler_ww_selected_temperature (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) with device_class None reports a temperature in °C which will be converted to °F. Temperature conversion for entities without correct device_class is deprecated and will be removed from Home Assistant Core 2022.3. Please update your configuration if device_class is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2021-12-23 14:43:05 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.boiler_ww_hysteresis_on_temperature (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) with device_class None reports a temperature in °C which will be converted to °F. Temperature conversion for entities without correct device_class is deprecated and will be removed from Home Assistant Core 2022.3. Please update your configuration if device_class is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2021-12-23 14:43:05 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.boiler_ww_hysteresis_off_temperature (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) with device_class None reports a temperature in °C which will be converted to °F. Temperature conversion for entities without correct device_class is deprecated and will be removed from Home Assistant Core 2022.3. Please update your configuration if device_class is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2021-12-23 14:43:05 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.boiler_ww_disinfection_temperature (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) with device_class None reports a temperature in °C which will be converted to °F. Temperature conversion for entities without correct device_class is deprecated and will be removed from Home Assistant Core 2022.3. Please update your configuration if device_class is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2021-12-23 14:43:05 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.thermostat_hc1_selected_room_temperature (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) with device_class None reports a temperature in °C which will be converted to °F. Temperature conversion for entities without correct device_class is deprecated and will be removed from Home Assistant Core 2022.3. Please update your configuration if device_class is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
```

**Device information**
EMS-ESP Version
v3.3.0

Device (Platform / SDK)
ESP32 / v3.3.5-1-g85c43024c

proddy, 2021-12-24:
I'm not seeing those warnings on my latest HA (2021.12.5) which was released a few days ago, but read it's coming soon. The device class for temperatures is set to 'temperature'. Should this be set now to 'measurement'? Can you follow-up and test please.

MizterB, 2021-12-26:
Still getting these messages with 2021.12.5.  I don't believe the device_class needs to be changed to 'measurement'.  The issue is that for these temperature entities mentioned in the log, no there is no device class being assigned back in the MQTT entity config.

But what is interesting is that some temperature values from the boiler are correctly having the 'temperature' device class assigned.  So I feel like not all temperature sensors are being recognized as such my the EMS Gateway. Here is a screenshot showing all of my boiler's temperature sensors.  As you can see, only a handful have a device_class assigned to them.

[image]

proddy, 2021-12-26:
You're right. I'm seeing incorrect assignments in HA too now. I'll fix it. Thanks for reporting

proddy, 2021-12-27:
This should be fixed now. I used @MichaelDvP's code as it looks like he fixed that a while back. I'll add it the v3.4 branch

closed 2023-02-19 by proddy