emsesp/EMS-ESP32 #268: topic `ems-esp/thermostat_data` does not get updated

balk77, 2021-12-27:
**Bug description**
The topic `ems-esp/thermostat_data` does not get updated. Actually, new topics are being issued but with old values. When the EMS devices gets restarted new values are being read and published, but nothing new after that. I own a Nefit Easy thermostat; although I am aware of the limitations this reading used to work properly until I installed 3.3.0.

Is there anything I can do to track the issue down?

**Steps to reproduce**
Reboot of the EMS device results in fresh values. No updates after that. I have a long list of identical MQTT topics in MQTT Explorer. 

**Expected behavior**
Updated temperatures every now and then. It used to work with same hardware with earlier versions of EMS-ESP32.

**Screenshots**
The MQTT output on `ems-esp/thermostat_data` stayed like this all day.
```
{"id":202,
  "hc1":{
    "seltemp":20,"currtemp":18.9,"hatemp":18.9,"hamode":"heat"
  }
}
```
while at the same time:
[image]
Also the web frontend shows the old values:
[image]

**Device information**
```
{
  "System": {
    "version": "3.3.0",
    "uptime": "000+09:52:00.262",
    "uptime (seconds)": 35520,
    "freemem": 160,
    "reset reason": "Software reset CPU / Software reset CPU",
    "Dallas sensors": 0
  },
  "Network": {
    "connection": "WiFi",
    "hostname": "ems-esp",
    "SSID": "xyz",
    "BSSID": "xyz",
    "RSSI": -60,
    "MAC": "xyz",
    "IPv4 address": "192.168.4.28/255.255.255.0",
    "IPv4 gateway": "192.168.4.1",
    "IPv4 nameserver": "192.168.4.1"
  },
  "Status": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "telegrams received": 25448,
    "read requests sent": 43,
    "write requests sent": 0,
    "incomplete telegrams": 2,
    "tx fails": 0,
    "rx line quality": 100,
    "tx line quality": 100,
    "MQTT": "connected",
    "MQTT publishes": 11422,
    "MQTT publish fails": 0,
    "Dallas reads": 0,
    "Dallas fails": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Nefit GBx72/Trendline/Cerapur/Greenstar Si/27i (DeviceID:0x08 ProductID:123, Version:06.07)",
      "handlers": "0x10 0x11 0xC2 0x14 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "Thermostat",
      "name": "Logamatic TC100/Moduline Easy (DeviceID:0x18, ProductID:202, Version:02.22)",
      "handlers": "0xA3 0x06 0xA2 0x12 0x0A"
    },
    {
      "type": "Controller",
      "name": "BC10 (DeviceID:0x09, ProductID:190, Version:01.04)"
    }
  ]
}
```

**Additional context**
*Add any other context about the problem here.*

proddy, 2021-12-27:
can you post your settings?

proddy, 2021-12-28:
https://emsesp.github.io/docs/#/Troubleshooting?id=mqtt-is-not-always-working

balk77, 2021-12-28:
Is there a better way to post settings than just a screenshot?
[image]

I believe I had 3.2.1 installed.
I don't think it is an MQTT issue because the same (wrong) values are listed on the front end which (I believe) feeds the MQTT module. But I will take a look later today.

proddy, 2021-12-28:
You can download both the settings and info JSON files from the Help tab. Or call it directly via the API using a URL. I'll update the doc here to explain a little better. https://emsesp.github.io/docs/#/Support

balk77, 2021-12-28:
Here it is
```
{
  "System": {
    "version": "3.3.0"
  },
  "Network": {
    "hostname": "ems-esp",
    "static_ip_config": false,
    "enableIPv6": false
  },
  "AP": {
    "provision_mode": 1,
    "ssid": "ems-esp",
    "local_ip": "192.168.4.1",
    "gateway_ip": "192.168.4.1",
    "subnet_mask": "255.255.255.0"
  },
  "MQTT": {
    "enabled": true,
    "host": "192.168.0.7",
    "port": 1883,
    "username": "",
    "client_id": "ems-esp32",
    "keep_alive": 60,
    "clean_session": false,
    "publish_time_boiler": 10,
    "publish_time_thermostat": 10,
    "publish_time_solar": 10,
    "publish_time_mixer": 10,
    "publish_time_other": 10,
    "publish_time_sensor": 10,
    "ha_climate_format": 1,
    "ha_enabled": true,
    "mqtt_qos": 0,
    "mqtt_retain": false,
    "send_response": false
  },
  "NTP": {
    "enabled": true,
    "server": "192.168.0.7",
    "tz_label": "Europe/Amsterdam",
    "tz_format": "CET-1CEST,M3.5.0,M10.5.0/3"
  },
  "OTA": {
    "enabled": true,
    "port": 8266
  },
  "Settings": {
    "board_profile": "S32",
    "tx_mode": 2,
    "ems_bus_id": 11,
    "syslog_enabled": false,
    "syslog_level": 3,
    "syslog_mark_interval": 0,
    "syslog_host": "",
    "syslog_port": 514,
    "master_thermostat": 0,
    "shower_timer": false,
    "shower_alert": false,
    "rx_gpio": 23,
    "tx_gpio": 22,
    "dallas_gpio": 18,
    "pbutton_gpio": 0,
    "led_gpio": 2,
    "phy_type": 0,
    "hide_led": true,
    "notoken_api": false,
    "dallas_parasite": false,
    "dallas_format": 1,
    "bool_format": 1,
    "enum_format": 1,
    "analog_enabled": false
  }
}
```

proddy, 2021-12-29:
Everything looks fine. I Can't think of what can be causing it. Try the ideas in https://emsesp.github.io/docs/#/Troubleshooting?id=mqtt-is-not-always-working and also look at the Mosquitto logs

MichaelDvP, 2021-12-30:
I see some strange things:
```
"uptime": "000+09:52:00.262",
"telegrams received": 25448,
 "read requests sent": 43,
```
normaly emsesp send every minute, this much to low sendings, do you have set `tx-mode 0` for a while?

Also the log does not show any telegram send to emsesp, it's 2 minutes but no request. And the log does not contain any 0x0A telegram from thermostat, but this telegram holds the roomtemp and seltemp.

Can you switch to tx-mode 1 and read the telegrams in telnet: `read 18 0a`

BTW: the 0x0A telegram should be 26 databytes, but we know only 4 of them, there is some more information from thermstat.

proddy, 2022-01-19:
@balk77 any update on this? Do you still want us to look into it?

balk77, 2022-01-19:
> I see some strange things:
>
> ```
> "uptime": "000+09:52:00.262",
> "telegrams received": 25448,
>  "read requests sent": 43,
> ```
>
> normaly emsesp send every minute, this much to low sendings, do you have [...]

Sorry, I was distracted by the year-end activities.... Issue is still present but worked around (I am reading through the Nefit Easy custom component instead of ems-esp)
```
ems-esp:$ show ems
EMS Bus is connected.

EMS Bus info:
  Tx mode: 2
  Bus protocol: Buderus
  #telegrams received: 40715
  #read requests sent: 93
  #write requests sent: 0
  #incomplete telegrams: 22
  #tx fails (after 3 retries): 8
  Rx line quality: 100%
  Tx line quality: 92%

Rx Queue is empty

Tx Queue is empty
```
Result of `read 18 0a`
```
000+19:12:46.332 N 4: [emsesp] Thermostat(0x18) -> Me(0x0B), EasyMonitor(0x0A), data: 00 00 00 00 00 00 00 00 08 07 08 02 00 0D 37 A5 10 0E 00 00 00 0D 00 00 00 00

```

MichaelDvP, 2022-01-20:
I think this is the same as #282, the easy monitor needs to be fetched and this is stopped after a tx-error.
In v3.4 this is fixed in PR#287. @balk77 Would you like to try with the alpha v3.4, compiled builds are published on discord and in #294, or wait for a fix in dev?
@proddy Ill make a PR to dev to fix it before you release v3.3.1.

proddy, 2022-01-20:
Ok I'll wait before releasing 331

balk77, 2022-01-20:
@MichaelDvP that works! Topics get updated after installing the alpha version listed in #294 . Thanks (and also @proddy !!!!!) for your help 👍 and for creating this fine piece of firmware 🥇

closed 2022-01-20 by balk77