emsesp/EMS-ESP32 #269: Attach to strongest/best Access Point

PhillyGilly, 2021-12-28:
I think this is probably a @bbqkees request, but I thought I would air it here as there may be a  simple answer that I can't see.
My home network unifi system has four APs in the house to cover the volume/shape and overcome the tinfoil backed insulation.
Two of the APs are hard wired back to the switch and two are connected wirelessly via the wired APs.
Most of the time everything is rock solid but occasionally devices get attached to the "wrong" AP and the data flow can be compromised. This has happed twice on my EMS/ESP with both incidents requiring a power cycle to resolve it.
I can't work out what causes this but for my (many) tasmota devices it isn't a serious problem because of the available configuration options:
<html>
<body>

SetOption56 | Wi-Fi network scan to select strongest signal on restart (network has to be visible)0 = disable (default)1 = enable
-- | --
SetOption57 | Wi-Fi network re-scan every 44 minutes with alternate to +10dB stronger signal if detected (only visible networks)0 = disable1 = enable (default)

</body>
</html>

However the EMS/ESP doesn't appear to behave like this, so if the best/strongest AP becomes temporarily unavailable, it connects to another AP and then stays it there even when the best/stongest AP comes back again. 

Any thoughts/ideas welcome.

bbqkees, 2021-12-28:
I believe with Unifi you can force a particular client to a particular AP, and also set a lower dB limit when a client should be kicked off the AP. That should provide a way to mitigate some of these issues.

PhillyGilly, 2021-12-28:
@bbqkees thanks. I'll engage with Mr Google and see how I get on.

proddy, 2021-12-28:
yes, Unifi released a new version a few weeks ago with this support. I use it so my EMS-ESP test circuit always connects the AC Pro in my study. You can bind a particular device to an AP and change its routing based on rules. There is also a 'preferred channel' option in EMS-ESP v3.4 as another workaround if you want to mess around with channels.

PhillyGilly, 2021-12-28:
Thanks @bbqkees and @proddy. The new function was hidden from me as I stuck with the classic interface. I've just configured about 50 devices that are in fixed locations, so all good now. FWIW I did try to replicate the Tasmota functionality in one of my Arduino projects but it was beyond my skillset.

closed 2021-12-28 by PhillyGilly