emsesp/EMS-ESP32 #270: Enhancement - Boiler GC7000F

wt19, 2021-12-28:
Dears, I am using EMS-ESP with a CW400 thermostat and a GC7000F Boiler from Bosch. CW400 works really great!
GC7000F Boiler is not (yet) supported. I would like use this enhancement issue to share my console output that you already have some data where you could look in for future enhancements.

Just let me know if you you need additional data/ support from my end.

show:
`Unknown: Junkers unknown (DeviceID:0x08 ProductID:132, Version:01.44)`

watch on 0x08:
`000+00:18:21.352 N 0: [emsesp] Unknown(0x08) -> All(0x00), ?(0xE4), data: 10 20 2D 48 00 C8 2B 01 B7 64 25 03 00 00 00 00 00 01 95 00 95 10 00 01 B7 80 00
000+00:18:21.507 N 1: [emsesp] Unknown(0x08) -> All(0x00), ?(0xE4), data: 01 B1 80 00 7F FF 80 00 (offset 27)
000+00:18:21.695 N 2: [emsesp] Unknown(0x08) -> All(0x00), ?(0xE3), data: 01 00 01 00 00 00 00 00 00 00 00 01 B7 25 64 50 00 00 00 00
000+00:18:21.878 N 3: [emsesp] Unknown(0x08) -> All(0x00), ?(0xE9), data: 37 02 47 02 47 00 00 00 00 46 3C 00 00 01 00 89 53 00 04 60 00 00 00 37 00 37
000+00:18:22.036 N 4: [emsesp] Unknown(0x08) -> All(0x00), ?(0x8E4), data: 00 00 00 2D 48 00 00 2B 64 00
000+00:18:31.302 N 5: [emsesp] Unknown(0x08) -> All(0x00), ?(0xE4), data: 10 20 2D 48 00 C8 2B 01 B7 64 25 03 00 00 00 00 00 01 95 00 8A 10 00 01 B7 80 00
000+00:18:31.459 N 6: [emsesp] Unknown(0x08) -> All(0x00), ?(0xE4), data: 01 B1 80 00 7F FF 80 00 (offset 27)
000+00:18:31.647 N 7: [emsesp] Unknown(0x08) -> All(0x00), ?(0xE3), data: 01 00 01 00 00 00 00 00 00 00 00 01 B7 25 64 50 00 00 00 00
000+00:18:31.878 N 8: [emsesp] Unknown(0x08) -> All(0x00), ?(0xE9), data: 37 02 47 02 47 00 00 00 00 46 3C 00 00 01 00 89 53 00 04 60 00 00 00 37 00 37
000+00:18:32.037 N 9: [emsesp] Unknown(0x08) -> All(0x00), ?(0x8E4), data: 00 00 00 2D 48 00 00 2B 64 00
000+00:18:32.213 N 10: [emsesp] Gateway(0x48) <- Unknown(0x08), ?(0x18), data: 02 (offset 11)
000+00:18:32.224 N 11: [emsesp] Unknown(0x08) -> Gateway(0x48), ?(0x18), data: <empty> (offset 11)
000+00:18:41.093 N 12: [emsesp] Gateway(0x48) <- Unknown(0x08), ?(0x04), data: 02 (offset 25)
000+00:18:41.124 N 13: [emsesp] Unknown(0x08) -> Gateway(0x48), ?(0x04), data: <empty> (offset 25)
000+00:18:41.313 N 14: [emsesp] Unknown(0x08) -> All(0x00), UBADevices(0x07), data: 09 01 00 00 00 00 00 00 01 00 00 00 00
000+00:18:41.503 N 15: [emsesp] Unknown(0x08) -> All(0x00), ?(0xE4), data: 10 20 2D 48 00 C8 2B 01 B7 64 25 03 00 00 00 00 00 01 95 00 86 10 00 01 B7 80 00
000+00:18:41.658 N 16: [emsesp] Unknown(0x08) -> All(0x00), ?(0xE4), data: 01 B1 80 00 7F FF 80 00 (offset 27)
000+00:18:41.926 N 17: [emsesp] Unknown(0x08) -> All(0x00), ?(0xD1), data: 00 52
000+00:18:42.120 N 18: [emsesp] Unknown(0x08) -> All(0x00), ?(0xE3), data: 01 00 01 00 00 00 00 00 00 00 00 01 B7 25 64 50 00 00 00 00
000+00:18:42.303 N 19: [emsesp] Unknown(0x08) -> All(0x00), ?(0xE5), data: 00 00 27 00 00 D0 00 CF 00 00 00 13 40 04 6E D9 04 6E D9 03 E5 9F 00 0E E4 64 00
000+00:18:42.454 N 20: [emsesp] Unknown(0x08) -> All(0x00), ?(0xE5), data: 00 00 00 00 00 (offset 27)
000+00:18:42.653 N 21: [emsesp] Unknown(0x08) -> All(0x00), ?(0xE9), data: 37 02 47 02 47 00 00 00 00 46 3C 00 00 01 00 89 53 00 04 60 00 00 00 37 00 37
000+00:18:42.808 N 22: [emsesp] Unknown(0x08) -> All(0x00), ?(0x2D6), data: 01 00 00 00 00 2B
000+00:18:43.012 N 23: [emsesp] Unknown(0x08) -> All(0x00), ?(0x8E4), data: 00 00 00 2D 48 00 00 2B 64 00
000+00:18:43.217 N 24: [emsesp] Gateway(0x48) <- Unknown(0x08), ?(0x4A4), data: 01
000+00:18:43.226 N 25: [emsesp] Unknown(0x08) -> Gateway(0x48), ?(0x4A4), data: <empty>
000+00:18:43.252 N 26: [emsesp] Gateway(0x48) <- Unknown(0x08), ?(0x4A2), data: 01
000+00:18:43.276 N 27: [emsesp] Unknown(0x08) -> Gateway(0x48), ?(0x4A2), data: <empty>
000+00:18:43.297 N 28: [emsesp] Gateway(0x48) <- Unknown(0x08), ?(0x4A2), data: 01 (offset 4)
000+00:18:43.326 N 29: [emsesp] Unknown(0x08) -> Gateway(0x48), ?(0x4A2), data: <empty> (offset 4)
000+00:18:43.356 N 30: [emsesp] Gateway(0x48) <- Unknown(0x08), ?(0x48A), data: 02
000+00:18:43.376 N 31: [emsesp] Unknown(0x08) -> Gateway(0x48), ?(0x48A), data: <empty>
000+00:18:43.697 N 32: [emsesp] Gateway(0x48) <- Unknown(0x08), ?(0x4A2), data: 01 (offset 2)
000+00:18:43.726 N 33: [emsesp] Unknown(0x08) -> Gateway(0x48), ?(0x4A2), data: <empty> (offset 2)
000+00:18:43.969 N 34: [emsesp] Gateway(0x48) <- Unknown(0x08), ?(0x4A2), data: 01 (offset 1)
000+00:18:44.001 N 35: [emsesp] Unknown(0x08) -> Gateway(0x48), ?(0x4A2), data: <empty> (offset 1)
000+00:18:44.217 N 36: [emsesp] Gateway(0x48) <- Unknown(0x08), ?(0x4A4), data: 01 (offset 1)
000+00:18:44.226 N 37: [emsesp] Unknown(0x08) -> Gateway(0x48), ?(0x4A4), data: <empty> (offset 1)
000+00:18:44.441 N 38: [emsesp] Thermostat(0x10) -> Unknown(0x08), ?(0xE7), data: 01 02 00
000+00:18:44.946 N 39: [emsesp] Gateway(0x48) <- Unknown(0x08), ?(0x4A4), data: 02 (offset 4)
000+00:18:44.976 N 40: [emsesp] Unknown(0x08) -> Gateway(0x48), ?(0x4A4), data: <empty> (offset 4)
000+00:18:45.001 N 41: [emsesp] Gateway(0x48) <- Unknown(0x08), ?(0x48D), data: 02 (offset 10)
000+00:18:45.026 N 42: [emsesp] Unknown(0x08) -> Gateway(0x48), ?(0x48D), data: <empty> (offset 10)
000+00:18:45.418 N 43: [emsesp] Thermostat(0x10) -> Unknown(0x08), ?(0x2E0), data: 01 2B 64 00 01
000+00:18:45.466 N 44: [emsesp] Thermostat(0x10) -> Unknown(0x08), ?(0x2EA), data: 00
000+00:18:45.721 N 45: [emsesp] Gateway(0x48) <- Unknown(0x08), ?(0x48D), data: 01
000+00:18:45.751 N 46: [emsesp] Unknown(0x08) -> Gateway(0x48), ?(0x48D), data: <empty>
000+00:18:45.777 N 47: [emsesp] Gateway(0x48) <- Unknown(0x08), ?(0x48D), data: 01 (offset 8)
000+00:18:45.801 N 48: [emsesp] Unknown(0x08) -> Gateway(0x48), ?(0x48D), data: <empty> (offset 8)
000+00:18:45.825 N 49: [emsesp] Gateway(0x48) <- Unknown(0x08), ?(0x48F), data: 02 (offset 8)
000+00:18:45.851 N 50: [emsesp] Unknown(0x08) -> Gateway(0x48), ?(0x48F), data: <empty> (offset 8)
000+00:18:45.873 N 51: [emsesp] Gateway(0x48) <- Unknown(0x08), ?(0x48F), data: 02 (offset 10)
000+00:18:45.901 N 52: [emsesp] Unknown(0x08) -> Gateway(0x48), ?(0x48F), data: <empty> (offset 10)
000+00:18:46.169 N 53: [emsesp] Gateway(0x48) <- Unknown(0x08), ?(0x48F), data: 02 (offset 12)
000+00:18:46.201 N 54: [emsesp] Unknown(0x08) -> Gateway(0x48), ?(0x48F), data: <empty> (offset 12)
000+00:18:46.282 N 55: [emsesp] Unknown(0x08) -> All(0x00), ?(0xE4), data: 10 20 2D 48 00 C8 2B 01 B6 64 25 03 00 00 00 00 00 01 95 00 87 10 00 01 B6 80 00
000+00:18:46.433 N 56: [emsesp] Unknown(0x08) -> All(0x00), ?(0xE4), data: 01 B1 80 00 7F FF 80 00 (offset 27)
000+00:18:46.721 N 57: [emsesp] Gateway(0x48) <- Unknown(0x08), ?(0x48F), data: 02 (offset 38)
000+00:18:46.754 N 58: [emsesp] Unknown(0x08) -> Gateway(0x48), ?(0x48F), data: <empty> (offset 38)
000+00:18:46.777 N 59: [emsesp] Gateway(0x48) <- Unknown(0x08), ?(0x48F), data: 02 (offset 14)
000+00:18:46.801 N 60: [emsesp] Unknown(0x08) -> Gateway(0x48), ?(0x48F), data: <empty> (offset 14)
000+00:18:47.121 N 61: [emsesp] Gateway(0x48) <- Unknown(0x08), ?(0x4A4), data: 01 (offset 3)
000+00:18:47.151 N 62: [emsesp] Unknown(0x08) -> Gateway(0x48), ?(0x4A4), data: <empty> (offset 3)
000+00:18:47.177 N 63: [emsesp] Gateway(0x48) <- Unknown(0x08), ?(0x4AC), data: 01
000+00:18:47.201 N 64: [emsesp] Unknown(0x08) -> Gateway(0x48), ?(0x4AC), data: <empty>`

proddy, 2021-12-28:
thanks for sharing. I'll add it the latest build and see where that takes us...

proddy, 2021-12-29:
if you DM on Discord I'll send over a new firmware to try out

wt19, 2021-12-29:
Today I have received the first alpha version 3.4.0a3 which is supporting my junkers/bosch boiler GC7000F.
I did some testing and so far it looks very promising. I will come back to this issue tomorrow and update the status if I couldn't find any problem.

Thanks to @proddy for that outstanding support and development. Enjoy the next coffees ;-)

wt19, 2021-12-30:
As I found no issues with the integration I will close this Issue.

closed 2021-12-30 by wt19