emsesp/EMS-ESP32 #271: Extend analog sensor support

proddy, 2021-12-28:
I've migrated @MichaelDvP's code over to 3.4 and included in the customizations. There are 2 types ADC and I/O Counter although I haven't tested any of it yet. I'll check it in shortly to the v3.4 branch and then use his GH issue to track any changes/defects

MichaelDvP, 2021-12-30:
This looks very nice and configuring works... for a while.. and then the esp crashes. Seems to be a corrupted filesystem. 
When happend i got this:
```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8

┌──────────────────────────────────────┐
│ EMS-ESP version 3.4.0a5              │
│ https://github.com/emsesp/EMS-ESP32  │
│                                      │
│ type help to show available commands │
└──────────────────────────────────────┘

ems-esp:$ abort() was called at PC 0x401bb3f7 on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x400891f8:0x3ffb1d10 0x40089475:0x3ffb1d30 0x401bb3f7:0x3ffb1d50 0x401bb43e:0x3ffb1d70 0x401baa4b:0x3ffb1d90 0x401bc38b:0x3ffb1db0 0x400d3889:0x3ffb1dd0 0x400d3c49:0x3ffb1e00 0x400d3d79:0x3ffb1ee0 0x400d3e96:0x3ffb1f50 0x400fb056:0x3ffb1f70 0x400fef8f:0x3ffb1f90 0x4013c4cc:0x3ffb1fb0 0x4008aef6:0x3ffb1fd0

Rebooting...
```
Reflashing via USB does not help, I've made a version that formats littlefs on start and this boots. Then i can reflash the  normal version, but writing network-settings to FS does not work, i stuck in AP after every reboot. I have to flash v3.3.x to configure network.
Switch back to old littlefs-lib does not help, it's somewhere else. 
Is it only my compilation?

proddy, 2021-12-30:
Thanks for trying it out Michael. I really want to get this branch over to dev to get more feedback.

On the crash, I haven't seen this at all and been running this version for a few weeks now. What I did do is update the LittleFS to the latest version (in prep for IDF 4.2) as you noticed, but if you reverted that back to it's original then it must be something else I guess.

Maybe it's a continuous loop and the FS size is growing somehow?
Does it still crash without any Analog sensors added?

Maybe we can take that backtrace and debug it via pio with `monitor_filters = esp32_exception_decoder`?

To be honest I haven't tested any of the Analog stuff. I just implemented what you wanted (to make you happy :))  using your code. I was going to add a simple push button to test the I/O Counter and Reads on a bare ESP32.  For ADC I'm not even sure what a good hardware test would be

MichaelDvP, 2021-12-31:
Ok, i'll try again with fresh copy of v3.4. 
The crashes happend when configuring mqtt or settings, not in custom-menu. And mostly on a restart. 
I have not found the systematic yet, but i can try with a second esp, it happens also without bus connection.

proddy, 2021-12-31:
I'll also erase the flash and start from a fresh install. I remember in the old LittleFS there was a flag to make it behave like SPIFFS - but can't find that anymore in the latest. I think that was only to do with recursing through directories so probably not related

MichaelDvP, 2021-12-31:
My fault, seems to work now. I've changed analog.value to float (because factor is float) and this [crashes here](https://github.com/emsesp/EMS-ESP32/blob/a132b3136963d62da0726d7d4ce77fbb9bdda4e2/src/analogsensor.cpp#L214). With format 2 no crashes now. 

Btw: Can't add analog sensor if there is no dallas. [This](https://github.com/emsesp/EMS-ESP32/blob/a132b3136963d62da0726d7d4ce77fbb9bdda4e2/interface/src/project/DashboardData.tsx#L387) shold be `return (data.active_sensors > 0 || data.analog_enabled);`

proddy, 2021-12-31:
do you want to make those changes or shall I?

I also noticed the new LittleFS libs is a bit problematic. It doesn't create the FS after an 'erase'. I'll rollback those changes

proddy, 2021-12-31:
I've added your changes and will checkin shortly.

MichaelDvP, 2021-12-31:
Thanks, i need some time to read all your changes and keep up to date. Your'e changing very fast.
There are some small things, like factor should be initialized as 1 on new sensor, and 500ms poll for adc is good, but gpio/counter needs fast poll. I'll collect, test and make a PR or inform. The counter via polling  is only for slow changing things, maybe better  to use the [pulsecounter of esp32](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/pcnt.html)? I've never used it, but have a tasmota counter on water and current meter not far from heating.

proddy, 2022-01-01:
thanks, just push a PR when you're ready. I agree the polling should be different per type (read, counter, adc). And yes the factor must be 1 otherwise all the values will be zero! didn't spot that.

MichaelDvP, 2022-01-03:
Why do you have removed the uom from analog? The fixed `mV` is wrong for factors != 1.
(I use a pressure sensor on analog and calculate by factor to `bar`).

proddy, 2022-01-03:
because I thought the ADC only read in mV ?

MichaelDvP, 2022-01-03:
> because I thought the ADC only read in mV ?

The raw ADC: yes, `sensor.analog_` is the `mV` reading, But if you apply a factor 0,001 the result will be Volts, if you attach a pressure sensor with scale 2bar/V, the factor is 0,002 and `sensor.value()` is in `bar`. For the offset `mV` is always correct.

proddy, 2022-01-03:
Oh ok. Shall I add it back or does it mess with your changes?

MichaelDvP, 2022-01-03:
You can add it back.

closed 2022-01-19 by proddy