emsesp/EMS-ESP32 #273: Buderus RC25 - "hc1 mode type" incorrect value

emcekk, 2021-12-29:
**Bug description**
First of all, I don't think that's really important, but maybe can be addressed when fixing other issues

When the RC25 is in Auto mode a "hc1 mode type" parameter appears, but it's value is always set to "day", regardless of the current state. See pictures attached below.

**Steps to reproduce**
*Steps to reproduce the behavior.*

**Expected behavior**
When in auto+day mode a value 'day' should be set to "hc1 mode type"

When in auto+night mode, a value 'night' should be set to "hc1 mode type"

**Screenshots**
[image]

[image]

**Device information**
[image]

**Additional context**
*Add any other context about the problem here.*

MichaelDvP, 2021-12-30:
Please check the in terminal the telegram 0xAE in these modes. (`read 17 AE`)
It was a long discussion about the[ RC20/RC25 settings here](https://github.com/emsesp/EMS-ESP/issues/361#issuecomment-609473501).

proddy, 2022-01-08:
@emcekk can you look into this so we know whether to fix in the upcoming version

emcekk, 2022-01-08:
Sure, will give it a deeper look in the evening. I'm currently using 3.4a9 and this field is just blank [image]

proddy, 2022-01-08:
ok, we need to see the data from those telegrams when in night and day so we can figure out how to translate it. or you send me over an RC25 ;)

MichaelDvP, 2022-01-08:
The empty field is due to a wrong modytype reference [here](https://github.com/emsesp/EMS-ESP32/blob/a59a7fe44f2d2a4ccf90bbb4f0fe9cc7068c0fcd/src/devices/thermostat.cpp#L3243) and [here](https://github.com/emsesp/EMS-ESP32/blob/a59a7fe44f2d2a4ccf90bbb4f0fe9cc7068c0fcd/src/devices/thermostat.cpp#L3257), it should be `enum_modetype3`.

proddy, 2022-01-09:
thanks Michael. I've added it to 3.4a11

emcekk, 2022-01-09:
After flashing 3.4a11 it seems that the mode is recognized correctly

[image]

[image]

And for auto mode: 
[image]

"night" is correct at the moment the screenshot was taken

Sorry for not providing those telegrams, looks like the newborn is eating up most of my time.

Should I still sniff them for you?

proddy, 2022-01-10:
if all works as expected no need to grab those telegrams. You can just close this issue.

closed 2022-01-10 by emcekk