emsesp/EMS-ESP32 #276: support for fahrenheit

proddy, 2022-01-02:
@MichaelDvP I merged in your changes with the support for Fahrenheit, but don't fully understand the logic. You created a new Unit of Measure called DEGREES_R which I'm guessing is a relative temperature and then use that to set `fahrenheit` to 1 or 2 in Helpers::round2(). But if it's 1 won't the value always be zero?

If I understand the logic we can think of a nicer way to handle the conversions, without creating a fake UoM

MichaelDvP, 2022-01-02:
The conversion to Fahrenheit is different for absolute temperatures and relative temperatures like hysteresis. 
(absolute 1.8*t + 32, relative: 1.8*t). I use a logic with fahrenheit 0-off, 1-relative, 2-absolut and calculate if fahrenheit>0: 1.8*t + 32*(fahrenheit-1).
Maybe it's better not to use the fahrenheit-flag, but use the uom in helpers to calculate, or mark relative values by something different.

MichaelDvP, 2022-01-02:
If you don't like the DEGREES_R with `°C` unit, we can rename it to KELVIN and use `K` (without degree). 
BTW: [Here](https://github.com/emsesp/EMS-ESP32/issues/264#issuecomment-1001186490) you can see that HA calculates the hysteresis wrong. Hysteresis off should be 0, not 32. I don't know what HA calculates for a Kelvin value.

proddy, 2022-01-02:
Ok, I think I understand now. The` fahrenheit` flag in Helpers::round2() is also used to calculate the relative temperatures (DEGREES_R) like offsets, differences and hysteresis. Then they should stay as DEGREES and be output as degrees C or F. 

I agree we should probably find a better way to mark these. I like to try and keep things tidy so with UoM having only unique unit of measurements (so no duplicates). And the helpers() functions being independent and not having to know about the device value type. At the moment though I can't think of a better way and I don't want to introduce yet another parameter to the register_device_value() class if I can prevent it. So suggest we just keep it as it is for now.

proddy, 2022-01-05:
one thing I noticed that needs fixing is the min & max values in
- EMSdevice::get_value_info() used an API call
- the HA limits of the Number component

proddy, 2022-01-05:
Also another bug with the relative temps in DEGREES_R.

for example on degrees C it shows:

  dhw hysteresis on temperature: -5.0 °C
  dhw hysteresis off temperature: 0.0 °C

and in Fahrenheit it's

  dhw hysteresis on temperature: 22.9 °F
  dhw hysteresis off temperature: 32.0 °F

MichaelDvP, 2022-01-05:
Yes, these 2 values has to be set to DEGREES_R ([here](https://github.com/emsesp/EMS-ESP32/blob/72def5a2bd496d5feac48d19493747fe6d8524a1/src/devices/boiler.cpp#L324))

bbqkees, 2022-01-06:
Just wondering whats the main reason to add Fahrenheit as it's only used in the US where there is no actual demand for EMS-ESP?
Mainly for sake of completeness?

PhillyGilly, 2022-01-06:
A long time ago in school and university I was taught that it was C is for Celsius (superceding °C for Centigrade), following the dropping of the degree in Kelvin, however more recently it has been confirmed that Celsius should include the degree symbol too (probably to humour the fine people in the the US) . Although I am pretty sure nobody would be confused by a temperature of 55C.
When I started reading the conversation/explanation above I was initially confused as I thought DEGREES_R would refer to temperatures in Rankine which are degrees of Fahrenheit above 0 K. And I was impressed that temperatures in °C were translated to K then °R and then °F !! Obviously it's relative measurement. There is a school of thought that says that temperature differences should be expressed as Celsius degrees C° but that really is a bit OCD and definitely would confuse a lot of people!
I just thought I'd share this with you guys, but as I will never be using °F - and as the US folk say - I don't have a dog in this fight.

proddy, 2022-01-08:
> Just wondering whats the main reason to add Fahrenheit as it's only used in the US where there is no actual demand for EMS-ESP? Mainly for sake of completeness?

@MichaelDvP had added it to his fork and copied it over. Thought it was a nice addition and Michael did all the hard work.

SPCulhane, 2022-02-16:
Thank you for adding this!! (US folk here). I did notice when you select Fahrenheit and have MQTT enabled. The unit of measure in MQTT is still C but with an F temperature. HA converts it back to F but since it thought 180F was C, HA is displaying the temperature as 356F instead.

proddy, 2022-02-16:
that's not good. We'll fix it.

MichaelDvP, 2022-02-16:
I think best is to add it here:
```
const std::string EMSdevice::uom_to_string(uint8_t uom) {
    if (EMSESP::system_.fahrenheit() && (uom == DeviceValueUOM::DEGREES || uom == DeviceValueUOM::DEGREES_R)) {
        return read_flash_string(DeviceValue::DeviceValueUOM_s[DeviceValueUOM::FAHRENHEIT]);
    }
    return read_flash_string(DeviceValue::DeviceValueUOM_s[uom]);
}
```
and remove some other checks for the uom.
I see that in dallas there are also some `= (float)temperature / 10` that should be replaced by `round2()` to calculate fahrenheit.
I can make a PR, i have also some changes to thermostat/solar and more to analog and mqtt. Should i try to split in small PRs or is one PR with some commits ok?

proddy, 2022-02-16:
PR the whole thing and approve yourself is best.

SPCulhane, 2022-02-16:
looking forward to testing this when it's available. Thanks for the quick response!

proddy, 2022-02-17:
Michael made the changes in the latest b6 build

SPCulhane, 2022-02-17:
That was fast! At first, it was still reporting C but after disabling MQTT and deleting all the topics then re-enabling MQTT; we are good! Thank you!

[image]

MichaelDvP, 2022-02-17:
v3.4.0b6 also have a feature to remove all old HA configs from mqtt broker: You have to disable mqtt-discovery, wait some minutes, and reenable discovery again.

SPCulhane, 2022-02-17:
@MichaelDvP Thank you! I'll keep this in mind if ever needed for troubleshooting

closed 2022-02-18 by proddy