emsesp/EMS-ESP32 #277: WebUI dashboard on mobile devices looks ugly

proddy, 2022-01-02:
On small mobile devices (like smartphone) the new Dashboard table doesn't render nicely. Either introduce a scroll bar or hide columns as we did in earlier versions.

[image]

MichaelDvP, 2022-01-03:
If you allow line breaks in the description it looks better: `GBx72, Trendline, Cerapur, Greenstar Si/27i` instead of `GBx72/Trendline/Cerapur/Greenstar Si/27i`

proddy, 2022-01-03:
I had a test where it would limit the string to 10 chars and postfix with ellipsis (...) but it still didn't fit. There are just too many columns?

bbqkees, 2022-01-04:
My CSS is quite rusty but you can use CSS media queries to hide columns responsively if the page width gets too low, like on mobile in portrait mode.
That way you can hide the description column on mobile so the rest of the table will still fit.

proddy, 2022-01-15:
fixed in a13

closed 2022-01-15 by proddy