emsesp/EMS-ESP32 #278: Incorrect value for total energy supplied

JakeOfOz, 2022-01-03:
### Bug description
Version v3.3.1b1
Looking at the values for energy supplied and energy used, the **total energy supplied** seems off:

total energy supplied    **360 kWh**
total energy supplied heating    **6.984 kWh**
total energy warm supplied warm water    **2.483 kWh**
total energy supplied cooling    **1.038 kWh**

The total energy supplied is the same value as my **auxiliary electrical heater energy consumption total**. Do they get mixed up? I would expect the total energy supplied to be the sum of energy supplied heating, warm water, cooling (and pool, but that's 0 for me). 
I'm running v3.3.1b1

### Steps to reproduce
Just values that show up strange. Following @proddy 's question on Discord, this is the output from a few read 8 494 in telnet:
`ems-esp:$ read 8 494
000+11:12:21.001 N 1: [emsesp] Boiler(0x08) -> Me(0x0B), UBAEnergySupplied(0x494), data: 01 A3 02 9A 00 00 01 68 00 00 09 B3 00 00 1B 48 00 00 04 0E 00 00 00 00`
`ems-esp:$ read 8 494
000+11:12:29.251 N 2: [emsesp] Boiler(0x08) -> Me(0x0B), UBAEnergySupplied(0x494), data: 01 A3 02 A4 00 00 01 68 00 00 09 B3 00 00 1B 48 00 00 04 0E 00 00 00 00`
`ems-esp:$ read 8 494
000+11:12:30.476 N 3: [emsesp] Boiler(0x08) -> Me(0x0B), UBAEnergySupplied(0x494), data: 01 A3 02 A4 00 00 01 68 00 00 09 B3 00 00 1B 48 00 00 04 0E 00 00 00 00`
`ems-esp:$ read 8 494
000+11:12:43.930 N 4: [emsesp] Boiler(0x08) -> Me(0x0B), UBAEnergySupplied(0x494), data: 01 A3 02 B8 00 00 01 68 00 00 09 B3 00 00 1B 48 00 00 04 0E 00 00 00 00`
`ems-esp:$ read 8 494
000+11:13:03.052 N 5: [emsesp] Boiler(0x08) -> Me(0x0B), UBAEnergySupplied(0x494), data: 01 A3 02 CC 00 00 01 68 00 00 09 B3 00 00 1B 48 00 00 04 0E 00 00 00 00`
`ems-esp:$ read 8 494
000+11:16:35.481 N 7: [emsesp] Boiler(0x08) -> Me(0x0B), UBAEnergySupplied(0x494), data: 01 A3 03 9E 00 00 01 68 00 00 09 B3 00 00 1B 48 00 00 04 0E 00 00 00 00
`

### Expected behavior
I would expect the **total energy supplied** to be the sum of heating, warm water and cooling. The heat pump probably has it's own value for this, I suspect that in this case the value for the auxiliary heater consumption is shown.

### Screenshots
[image]

### Device information

System |  
-- | --
version | "3.3.1b1"
uptime | "000+11:37:00.874"
uptime (seconds) | 41820
freemem | 109
reset reason | "Software reset CPU / Software reset CPU"
Dallas sensors | 2
Network |  
connection | "Ethernet"
hostname | "ems-esp"
MAC | "A8:03:2A:A1:20:43"
IPv4 address | "192.168.X.XXX/255.255.255.0"
IPv4 gateway | "192.168.X.XXX"
IPv4 nameserver | "192.168.X.XXX"
Status |  
bus status | "connected"
bus protocol | "Buderus"
telegrams received | 132369
read requests sent | 13270
write requests sent | 1
incomplete telegrams | 0
tx fails | 0
rx line quality | 100
tx line quality | 100
MQTT | "connected"
MQTT publishes | 34321
MQTT publish fails | 0
Dallas reads | 16726
Dallas fails | 0
Devices |  
0 |  
type | "Boiler"
name | "Buderus Enviline/Compress 6000AW/Hybrid 7000iAW/SupraEco/Geo 5xx (DeviceID:0x08 ProductID:172, Version:01.18)"
handlers | "0x10 0x11 0xC2 0x14 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x494 0x495 0x48D 0x48F 0x48A"
1 |  
type | "Thermostat"
name | "RC300/RC310/Moduline 3000/1010H/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:13.07)"
handlers | "0xA3 0x06 0xA2 0x12 0x2A5 0x2B9 0x2AF 0x29B 0x471 0x2A6 0x2BA 0x2B0 0x29C 0x472 0x2A7 0x2BB 0x2B1 0x29D 0x473 0x2A8 0x2BC 0x2B2 0x29E 0x474 0x2F5 0x31B 0x31D 0x31E 0x23A 0x267 0x240"
2 |  
type | "Heatpump"
name | "HP Module (DeviceID:0x38, ProductID:200, Version:40.04)"
handlers | "0x42B 0x47B"
3 |  
type | "Gateway"
name | "KM200/MB LAN 2 (DeviceID:0x48, ProductID:189, Version:04.07)"

### Additional context
Running beta version **v3.3.1b1**, will try to revert to the latest stable version.

MichaelDvP, 2022-01-03:
I think this is an overflow in itoa [here](https://github.com/emsesp/EMS-ESP32/blob/3d8e1b8f869f08e29e2b564841d21a52c66950c0/src/helpers.cpp#L85), has to be  changed to int32_t. I had this for total burner time a few days ago. It's also in stable version.

JakeOfOz, 2022-01-03:
Just checked the values on the heat pump itself, they show up as incorrect too. So the EMS-ESP reads the values correctly it seems.

proddy, 2022-01-03:
ok, do you want to close this then?

closed 2022-01-05 by proddy