emsesp/EMS-ESP32 #280: Automatically reconnect to mqtt broker when connection is lost

lbbrhzn, 2022-01-04:
**Is your feature request related to a problem? Please describe.**
The connection to my MQTT broker sometimes fails, for instance when I reboot the server it runs on.
In this case the mqtt status of the ems-esp32 becomes 'disconnected', and I need to reset the connection by disabling and re-enabling MQTT before it connects again

**Describe the solution you'd like**
It would be nice if EMS-ESP would automatically attempt to reconnect to the MQTT broker in case the connection is dropped.

**Describe alternatives you've considered**
I could try to run the MQTT broker on a server that never goes down, but all servers need to be rebooted for the kernel to be updated.

**Additional context**
none

proddy, 2022-01-04:
I'm pretty sure it reconnects...I'll check

lbbrhzn, 2022-01-04:
For short connection drops it does reconnect. But some some reason this does not always work.

lbbrhzn, 2022-01-06:
I've experienced some more disconnects. Is there some logging I can enable to see why the connection is dropped?

proddy, 2022-01-06:
turn on the logging on the broker. EMS-ESP will only tell you the TCP connection was broken. What you also can try is downloading and running mosquitto on a pc/laptop and testing that. I think you'll find it will reconnect automatically. See https://emsesp.github.io/docs/#/Troubleshooting?id=mqtt

proddy, 2022-01-09:
https://emsesp.github.io/docs/#/Troubleshooting?id=integration-with-home-assistant-breaks-when-mqtt-broker-restarts

proddy, 2022-01-10:
I did some rigorous testing and confirm EMS-ESP always reconnects. There will be data loss if the Broker is not setup to have a persistent store. I've made a note in the docs about this. If it happens again then we can re-open this and dive into it again

closed 2022-01-10 by proddy

bsergei, 2023-03-28:
I had ems-esp uptime 29 days, and recently had some difficulties with my MQTT server (laggy, unresponsive). After restart the MQTT host I noticed MQTT connection lost (something like TCP connection lost in system logs) and EMS-ESP was not trying to reconnect to it in more than 20 minutes, so I had to restart it manually and it connected after restart. I am on firmware 3.4.4.