emsesp/EMS-ESP32 #281: Merging Victor-Mo's IRT-ESP?

mrwee, 2022-01-05:
Seems like a lot of good stuff has happened in EMS-ESP(32). Still owning an old boiler using the iRT protocol, I'm currently using Victor-Mo's IRT-ESP. Are there any plans / possibility of including his work in ESP-ESP32?

proddy, 2022-01-05:
Victor & his contributors would need to do the port to the new framework first. A lot has changed in the last 2 years and it won't be a trivial task as there's much new tech to learn (C++, STD libs, React, Typescript...). But I have offered to help out on numerous occasions. When that's done we can consider merging although I don't think there are many iRT users out there which is why BBQKees discontinued supporting his modded board. The long-term focus of this project is to get deeper into the new range of heat pumps.

bbqkees, 2022-01-05:
The current state of iRT-ESP is that there are about 15 parameters that can be read from the boiler, and for that purpose I think the 'old' firmware is still fine.
Of course it is nice to have the polished and feature rich EMS-ESP V3, but for the iRT boilers it won't actually make a real difference as the boiler itself cannot provide more parameters to interface with.
Also the current modded boards cannot actively control the thermostat, and those iRT boilers are now about 20 years or older.
I sold 5 or 6 iRT modded boards last your so it's also not worth the many hours of effort to port everything to EMS-ESP V3.

mrwee, 2022-01-05:
ok, if that's all the parameters we can get out of the old boilers, then it's probably not worth the effort, even if e.g. Home Assistant autodiscovery had been nice.
I'll just stay with the existing version and enjoy that I at least get some information out of the boiler, while I enviously follow all the nice new stuff you guys get with V3+ :)

closed 2023-02-19 by proddy