emsesp/EMS-ESP32 #282: Missing values for damped outdoor temperature

Tillmann-E, 2022-01-05:
**Bug description**
Reading the damped outdoor temperature (thermostat) fails after a couple of hours.
After restarting its working again for some hours. 

**Steps to reproduce**
Has been working with version 3.1.1. Error occurs since update to version 3.3.0.

System log for the time of the error. Last value received should be around 2022-01-04T00:04:20.000Z.
```
2022-01-03 20:26:11.977 W 55: [telegram] Incomplete Rx: 90 0B FF 00 01 40 32 00 01 00 00 00 00 00 FF 01 F6 06 FF 00
2022-01-03 20:44:05.246 W 56: [telegram] Incomplete Rx: 90 88 00 2A 00 00 00 00 00 03 00 00 00 C6 00 C6 80 00 00 80 00 80 00 80 00 00
2022-01-03 22:25:07.487 W 57: [telegram] Incomplete Rx: 88 88 00 2A 00 00 00 00 00 03 00 00 00 C7 00 C7 80 00 00 80 00 80 00 80 00 00
2022-01-03 23:05:00.499 W 58: [telegram] Incomplete Rx: FE 88 00 2A 00 00 00 00 00 03 00 00 00 CF 00 00 80 00 00 80 00 80 00 80 00 00
2022-01-03 23:08:29.687 W 59: [telegram] Incomplete Rx: FE 88 00 2A 00 00 00 00 00 03 00 00 00 CF 00 00 80 00 00 80 00 80 00 80 00 00
2022-01-04 00:05:07.469 W 60: [telegram] Incomplete Rx: 90 0B FF 00 01 B9 FF 2E 28 26 00 01 00 FF FF 05 28 01 E1 20
2022-01-04 00:16:09.681 E 61: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B 90 FF 00 20 01 B9
2022-01-04 00:32:17.354 E 62: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B 90 FF 00 20 01 3A
2022-01-04 00:32:28.616 E 63: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B 90 FF 00 20 01 40
2022-01-04 02:23:17.591 E 64: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B 90 FF 00 20 02 1B
```
**Expected behavior**
Constant values for damped outdoor temperature. 
**Screenshots**
[image]

**Device information**
| **System**                  |                                                                                                                                                                                   |
|-----------------------------|---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------|
| System.version              | 3.3.0                                                                                                                                                                                 |
| System.uptime               | 001+22:58:57.318                                                                                                                                                                      |
| System.uptime (seconds)     | 169137                                                                                                                                                                                |
| System.freemem              | 147                                                                                                                                                                                   |
| System.reset reason         | Software reset CPU / Software reset CPU                                                                                                                                               |
| System.Dallas sensors       | 4                                                                                                                                                                                     |
| Network.connection          | WiFi                                                                                                                                                                                  |
| Network.hostname            | ems-esp                                                                                                                                                                               |
| Network.SSID                | XXX                                                                                                                                                                              |
| Network.BSSID               | E8:DF:70:21:D8:ED                                                                                                                                                                     |
| Network.RSSI                | -71                                                                                                                                                                                   |
| Network.MAC                 | 08:3A:F2:52:DC:84                                                                                                                                                                     |
| Network.IPv4 address        | 192.168.XX.XX/255.255.255.0                                                                                                                                                           |
| Network.IPv4 gateway        | 192.168.XX.XX                                                                                                                                                                      |
| Network.IPv4 nameserver     | 192.168.XX.XX                                                                                                                                                                          |
| Status.bus status           | connected                                                                                                                                                                             |
| Status.bus protocol         | HT3                                                                                                                                                                                   |
| Status.telegrams received   | 129434                                                                                                                                                                                |
| Status.read requests sent   | 17390                                                                                                                                                                                 |
| Status.write requests sent  | 0                                                                                                                                                                                     |
| Status.incomplete telegrams | 20                                                                                                                                                                                    |
| Status.tx fails             | 7                                                                                                                                                                                     |
| Status.rx line quality      | 100                                                                                                                                                                                   |
| Status.tx line quality      | 100                                                                                                                                                                                   |
| Status.MQTT                 | connected                                                                                                                                                                             |
| Status.MQTT publishes       | 5130                                                                                                                                                                                  |
| Status.MQTT publish fails   | 0                                                                                                                                                                                     |
| Status.Dallas reads         | 135308                                                                                                                                                                                |
| Status.Dallas fails         | 0                                                                                                                                                                                     |
[... cut ...]
| Devices.1.handlers          | 0xA3 0x06 0xA2 0x12 0x2A5 0x2B9 0x2AF 0x29B 0x471 0x2A6 0x2BA 0x2B0 0x29C 0x472 0x2A7 0x2BB 0x2B1 0x29D 0x473 0x2A8 0x2BC 0x2B2 0x29E 0x474 0x2F5 0x31B 0x31D 0x31E 0x23A 0x267 0x240 |
| Devices.2.type              | Controller                                                                                                                                                                            |
| Devices.2.name              | Controller (DeviceID:0x09, ProductID:152, Version:02.08)                                                                                                                              |

proddy, 2022-01-05:
what are your system settings - are you publishing the data via MQTT or polling the API and at what frequency? If it's "on change" then perhaps EMS-ESP is not detecting a change. Does it always happen after the same # hours? 

Next time you notice no data is coming in go to the console do a `read 10 23a` to see if there is actual data.

Tillmann-E, 2022-01-05:
I'm publishing every 10 seconds via MQTT to home assistant. 
The number of hours are not always the same.

I'm going to check this next time, but I think there ist no actual data, because the value in the EMS-ESP UI hast also been static.

proddy, 2022-01-05:
yes, we need to figure out if the interface board or the Rx UART has stopped working

MichaelDvP, 2022-01-09:
The issue is that the boiler stops replying to 0x23A here:
`2022-01-04 00:32:17.354 E 62: [telegram] Last Tx Read operation failed after 3 retries. Ignoring request: 8B 90 FF 00 20 01 3A`
This generates a [empty dummy reply](https://github.com/emsesp/EMS-ESP32/blob/b65866217aab44f1501ef5fafefa53125cfba6f8/src/telegram.cpp#L589) and [toggle the fetch](https://github.com/emsesp/EMS-ESP32/blob/b65866217aab44f1501ef5fafefa53125cfba6f8/src/emsdevice.cpp#L1163) of this telegram. After this the telegram is not fetched anymore. We have introduced this, because some boilers/thermostats does not reply to some telegrams (or version request) and emsesp will endless query this telegrams. This dummy reply is also used for Tado-  and modem-detection.
Maybe 3 read-retries are not enough, or too fast if the bus is currently disturbed. We can try with a delayed send of the last retry, like the delay for post-send-queries, [insert here:](https://github.com/emsesp/EMS-ESP32/blob/b65866217aab44f1501ef5fafefa53125cfba6f8/src/telegram.cpp#L606)
` delayed_send_ = (retry_count_ < MAXIMUM_TX_RETRIES) ? 0 : (uuid::get_uptime() + POST_SEND_DELAY);`

Tillmann-E, 2022-01-09:
this ist the result to read 10 23a
`001+21:08:32.493 N 0: [emsesp] Thermostat(0x10) -> Me(0x0B), RC300OutdoorTemp(0x23A), data: 00 14`
with an outdoor temperature of 1.8 °C a damped outdoor temperature of 2.0 °C seems fine.

proddy, 2022-01-19:
@Tillmann-E did you try the latest v3.4 build to see if it resolved your issue?

Tillmann-E, 2022-01-29:
I updated to 3.3.1 a week ago and it's working now.
Thank you!

commit b0bd0eda (2022-01-10): Merge pull request #287 from MichaelDvP/v3.4

fix #282, check received status before toggling fetch-flag on empty t…
commit b4f9302d (2022-01-10): fix #282, check received status before toggling fetch-flag on empty telegrams.
closed 2022-01-29 by Tillmann-E