emsesp/EMS-ESP32 #283: IPv6 slaac support

kroon040, 2022-01-05:
Hi,

I see the ipv6 DHCP support is enabled. I got ipv6 working from my provider and it is working using ipv6 with Slaac for getting ipv6 adresses. Is the an option to support it with ems-esp32? the esp-idf support it but is not default enabled

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-lwip-ipv6

proddy, 2022-01-05:
This project, like most ESP8266/ESP32 projects, uses the Arduino core (https://github.com/espressif/arduino-esp32) and not built using the native ESP-IDF (which is something I've been considering). Unfortunately, the package comes with a pre-built SDK that's built _without_ the CONFIG_LWIP_IPV6_AUTOCONFIG parameter set to enable SLAAC.  The only way around this is to build a custom lwip library. It's not hard to do and add to platformio. I even think Tasmota may do something similar. Worth investigating if you have time. I'm pretty busy these days :-)

proddy, 2022-05-05:
this should be possible now with #473

closed 2023-02-18 by proddy