emsesp/EMS-ESP32 #284: Entities get renamed in Home Assistant?

bartplessers, 2022-01-07:
Hi,

It's not the first time that I noticed that the "entities" in homeassistant get renamed.

Currently my thermostat has following ID:

`sensor.thermostat_hc1_selected_room_temperature_2`
 notice the "_2" at the end.

I'm not sure what caused this change, I upgraded hass several times, but also ems-esp32...

Does anybody else experience the same issues?

Any idea what's the cause for this change in ID?

It is quite annoying because I register long term statistics in InfluxDB and those monitors are broken if the ID of an entity changes... 🥲

Kind regards,

Bart

proddy, 2022-01-07:
I get this all the time too and haven't been able to figure out how to remove the old ones. Usually, remove all the discovery entries from MQTT and restarting HA fixes it. There are a few threads on the HA forum and discord about how to clean up old lingering sensors from the DB but I haven't tried this yet. I'll dig into it over the weekend

proddy, 2022-01-16:
but saying that, I haven't seen this problem in the last month or two with the latest v3.4

bartplessers, 2022-01-23:
Hi @proddy 

Unfortunately:
[image]
Things went wrong 2022-01-18 15:32
No "inside temp" nor "set temp" from that point registered anymore.

Have you had any time to take a look at this issue? It drives me crazy...

grtz
B

proddy, 2022-01-24:
Are the entities there in the ems-esp? (Web, api, mqtt, console etc) or just missing in ha?

bartplessers, 2022-01-24:
# API
## http://192.168.3.118/api/thermostat
```

{
  "errorcode": "(0)",
  "datetime": "21:16:26 24/01/2022",
  "floordry": "off",
  "dampedoutdoortemp": 3.5,
  "floordrytemp": 0,
  "building": "medium",
  "minexttemp": -10,
  "wwsettemp": 60,
  "wwmode": "own_prog",
  "wwsettemplow": 45,
  "wwcircmode": "own_prog",
  "wwchargeduration": 15,
  "wwcharge": "off",
  "wwextra1": 0,
  "wwdisinfect": "off",
  "wwdisinfectday": "tu",
  "wwdisinfecttime": 120,
  "hc1": {
    "seltemp": 0,
    "currtemp": 17.8,
    "mode": "auto",
    "modetype": "eco",
    "ecotemp": 17,
    "manualtemp": 21,
    "comforttemp": 21,
    "summertemp": 17,
    "designtemp": 75,
    "offsettemp": 0,
    "minflowtemp": 25,
    "maxflowtemp": 75,
    "roominfluence": 4,
    "curroominfl": -4,
    "nofrosttemp": 5,
    "targetflowtemp": 0,
    "heatingtype": "radiator",
    "summersetmode": "auto",
    "summermode": "off",
    "controlmode": "optimized",
    "program": "prog_1",
    "fastheatup": 0
  }
}

```

## http://192.168.3.118/api/boiler
```
{
  "heatingactive": "off",
  "tapwateractive": "off",
  "selflowtemp": 5,
  "selburnpow": 0,
  "heatingpumpmod": 0,
  "outdoortemp": 2.3,
  "curflowtemp": 32.2,
  "rettemp": 13.2,
  "syspress": 1.4,
  "burngas": "off",
  "flamecurr": 0,
  "heatingpump": "off",
  "fanwork": "off",
  "ignwork": "off",
  "heatingactivated": "on",
  "heatingtemp": 82,
  "pumpmodmax": 71,
  "pumpmodmin": 71,
  "pumpdelay": 1,
  "burnminperiod": 10,
  "burnminpower": 0,
  "burnmaxpower": 100,
  "boilhyston": -6,
  "boilhystoff": 6,
  "curburnpow": 0,
  "burnstarts": 2885,
  "burnworkmin": 225600,
  "heatworkmin": 217451,
  "ubauptime": 647955,
  "lastcode": "6L(229) 13.03.2021 04:54",
  "servicecode": "0H",
  "servicecodenumber": 203,
  "maintenancemessage": " ",
  "maintenance": "off",
  "maintenancetime": 6000,
  "maintenancedate": "01.01.2015",
  "wwseltemp": 50,
  "wwsettemp": 53,
  "wwtype": "layered buffer",
  "wwcomfort": "hot",
  "wwflowtempoffset": 40,
  "wwmaxpower": 100,
  "wwcircpump": "off",
  "wwchargetype": "3-way valve",
  "wwhyston": -5,
  "wwhystoff": 0,
  "wwdisinfectiontemp": 70,
  "wwcircmode": "2x3min",
  "wwcirc": "off",
  "wwcurtemp": 52.9,
  "wwcurtemp2": 52.9,
  "wwcurflow": 0,
  "wwstoragetemp1": 20.9,
  "wwstoragetemp2": 53,
  "wwactivated": "on",
  "wwonetime": "off",
  "wwdisinfect": "off",
  "wwcharging": "off",
  "wwrecharging": "off",
  "wwtempok": "on",
  "wwactive": "off",
  "wwheat": "on",
  "wwstarts": 762,
  "wwworkm": 8149
}
```

## MQTT /ems-esp/boiler-data

`
{"id":208,"heatingactive":"off","tapwateractive":"off","selflowtemp":5,"selburnpow":0,"heatingpumpmod":0,"outdoortemp":2.3,"curflowtemp":31.4,"rettemp":12.8,"syspress":1.4,"burngas":"off","flamecurr":0,"heatingpump":"off","fanwork":"off","ignwork":"off","heatingactivated":"on","heatingtemp":82,"pumpmodmax":71,"pumpmodmin":71,"pumpdelay":1,"burnminperiod":10,"burnminpower":0,"burnmaxpower":100,"boilhyston":-6,"boilhystoff":6,"curburnpow":0,"burnstarts":2885,"burnworkmin":225600,"heatworkmin":217451,"ubauptime":647969,"lastcode":"6L(229) 13.03.2021 04:54","servicecode":"0H","servicecodenumber":203,"maintenancemessage":" ","maintenance":"off","maintenancetime":6000,"maintenancedate":"01.01.2015"}
`

- In HA, I only have `sensor.thermostat_damped_outdoor_temperature`, but no other entity with "outdoor" in its name, i.e. `boiler_outside_temperature`
- in MQTT, `/ems-esp/boiler-data` contains `outdoortemp`
- In InfluxDB, I **can** find `boiler_outside_temperature` and it has values !
- In Grafana, `boiler_outside_temperature` is selectable, but has no values

I just don't understand this.
In MQTT everything seems OK, but on HA, the entity is not recognized?

bart

proddy, 2022-01-24:
and using MQTTExplorer and searching for `outdoortemp` doesn't show a `homeassistant/sensor/ems-esp/boiler_outdoortemp` topic?

bartplessers, 2022-01-30:
Hi @proddy ,

sorry for the delay.

About this issue: unfortunately, it happened again lately.
As you can see: suffix "_2" to most of the entities in HA.
[image]

As you can see, my current room temperature entity has following id:

`sensor.thermostat_hc1_current_room_temperature_2`

Not all of the entities are renamed. I created a group with all ems-esp entities, and it looks now something like

[image]

Because the group is manually defined, it can not find `sensor.thermostat_hc1_current_room_temperature` anymore...

Of course I can updated all my HA internal id's, but this is not maintainable...

bartplessers, 2022-02-24:
Hi @proddy ,

FYI: It just happened again :-(

Now, I stopped my MQTT-broker and deleted the full mosquitto.db
After restarting HA , the entities still exists in HA.

Then I deleted manually all entities ending on "_2" and "_3", started my MQTT-broker again.

FYI
Before cleaning, I found out that on `R:\usr\share\hassio\homeassistant\.storage\core.entity_registry`, there are double entities, i.e.:

- sensor.boiler_ww_active_time
- sensor.boiler_ww_active_time_2

```

            {
                "area_id": null,
                "capabilities": {},
                "config_entry_id": "a8ffde57673a7cd71630b2e77d54ae55",
                "device_class": null,
                "device_id": "42a3cb86c146816f73f3e381286179c3",
                "disabled_by": null,
                "entity_category": null,
                "entity_id": "sensor.boiler_ww_active_time",
                "icon": null,
                "id": "1bf35f0887e59e27b717b71515fdb432",
                "name": null,
                "options": {},
                "original_device_class": null,
                "original_icon": "mdi:clock-outline",
                "original_name": "Boiler ww active time",
                "platform": "mqtt",
                "supported_features": 0,
                "unique_id": "boiler_ww_wwworkm",
                "unit_of_measurement": "minutes"
            },
			
			

            {
                "area_id": null,
                "capabilities": null,
                "config_entry_id": "a8ffde57673a7cd71630b2e77d54ae55",
                "device_class": null,
                "device_id": "42a3cb86c146816f73f3e381286179c3",
                "disabled_by": null,
                "entity_category": null,
                "entity_id": "sensor.boiler_ww_active_time_2",
                "icon": null,
                "id": "2de05d44e43cd888a37b5aed39af0bf2",
                "name": null,
                "options": {},
                "original_device_class": null,
                "original_icon": "mdi:clock-outline",
                "original_name": "Boiler ww active time",
                "platform": "mqtt",
                "supported_features": 0,
                "unique_id": "boiler_wwworkm",
                "unit_of_measurement": "minutes"
            },

```

Don't know why/how those entities have a double definition, but if you can avoid this, I would be very gratefull!

kind regards,
Bart

swa72, 2022-03-01:
Anyone with a hint on how to resolve this reliably? For some reason or the other, my ems-esp rebooted yesterday, some entities disappeared from HA and some _2 were created. Odd.

proddy, 2022-03-02:
I'm not really sure what causes it and to be honest it's been > 6months since I last had that same issue.

bartplessers, 2022-03-03:
I quiet sure it's a HA issue, and not an ems-esp one. 
When I had a problem with one of my instances of HA, I created a fresh one, and connected to the same mqqt-broker. No problem with the new instance. So there was no problem on the mqtt-tree.
If we only could find why ems-esp triggers the problem on HA, the solution would not be far away I think...

pswid, 2022-06-22:
I have also observed it many times and I think I know the reason.. Using mqqt explorer I noticed that after ems-esp restart (or firmware update) "HA discovery" topics are deleted and created from scratch. I suspect that new entities with the `_2` suffix appear when the discovery topic is recreated before the HA has time to detect the removal of the original one (in most cases there are more than 100 ems-esp topics, so it's a "lot of work" for HA). HA thinks that a second entity with with already existing name appears, so in order to keep the names unique, it adds a `_2` suffix to the newly detected one.

I see three solutions:
- not deleting valid discovery topics during ems-esp reboot
- adding some delay between deleting and recreating discovery topics
- use a faster server for HA

proddy, 2022-07-10:
That root cause does sound very plausible. Except I've noticed this problem with previous versions of EMS-ESP before we auto-removed all the HA discovery topics. And the original issue was logged back in January also before the removal of MQTT topics on startup, so it could also be something else.

closed 2023-02-19 by proddy