emsesp/EMS-ESP32 #288: Some comments on 3.4

MichaelDvP, 2022-01-10:
I have not read all the changes in code yet, but there are some small thing i'd like to comment: 

[This](https://github.com/emsesp/EMS-ESP32/blob/ff6738eef48d2e9c26ef470edb5a69e8513ee896/src/web/WebLogService.cpp#L108-L112) should be removed. I've made an experiment with log-level `trace` showing only telegrams (a suggestion from tp1de), but this only makes sense with selectable leves debug/trace/all, but you have only debug/all and `all` does not show all.

[This](https://github.com/emsesp/EMS-ESP32/blob/ff6738eef48d2e9c26ef470edb5a69e8513ee896/src/emsdevice.cpp#L1192-L1202) works, but not as expected. My build was older than the `factor` and i've used negative values as divider to allow multipications in Helpers::render/round. Now the strings are [negative](https://github.com/emsesp/EMS-ESP32/blob/ff6738eef48d2e9c26ef470edb5a69e8513ee896/src/locale_EN.h#L176-L178) and divider is always used. That's good for now, because `EMSdevice::publish_value`, `EMSdevice::generate_values_web`, and `EMSdevice::generate_values_web_all`  does not check for factor and always use the negative divider. But i think it should be consistent for all functions.

That's all for now.

proddy, 2022-01-10:
thanks for reviewing. As soon as I get off the v3.4 branch I'll go back to doing PRs so changes are easier to review and track. Right now it's a bit of a random blast of changes.

The first, yes, I noticed that and wasn't sure, which is why I made the comment but forgot to ask you. I've removed it now.

The second, yes it should be consistent and I'll think of a way to resolve that later when my brain is functioning

MichaelDvP, 2022-01-14:
Another point: The display of api-access-token is missing.  The `generateToken()` function is there, but never called.

MichaelDvP, 2022-01-15:
BTW: the ft-solar branch can be deleted. I think it's no need to merge to v3.3 and in v3.4 it is already included.

proddy, 2022-01-15:
btw today I'm looking at the ArduinoJson changes. in the latest version, they handle NUL strings which break a few things. I used a lot of std::strings in EMS-ESP because it was clean and all new and sexy, but probably more efficient to go back to C const char *'s to avoid all that unnecessary .c_str() conversions every time.

proddy, 2022-01-15:
> BTW: the ft-solar branch can be deleted. I think it's no need to merge to v3.3 and in v3.4 it is already included.

you should be able to delete it, no?

proddy, 2022-01-16:
> Another point: The display of api-access-token is missing. The `generateToken()` function is there, but never called.

It's back again

tp1de, 2022-01-18:
On 3.4 I get for seltemp per hc on api a max value of 14.5 °C returned. ???? 
My ioBroker reacts with a warning that actual selected temp is out of range !!!
.... it looks like that the values are now 50% of the values before .... 
{
"name": "seltemp",
"fullname": "hc1 selected room temperature",
"circuit": "hc1",
"value": 22,
"type": "number",
"min": 2.5,
"max": 14.5,
"uom": "°C",
"writeable": true,
"visible": true
}

proddy, 2022-01-18:
Same here. http://ems-esp.local/api/thermostat/seltemp shows 14.5.

MichaelDvP, 2022-01-18:
Yes also seen this, the min max have limits in °C and should not be scaled by divider. emsdevices line 1065:
```
            // limits are not scaled with divider and temperatures are °C
             if (dv.min != 0 || dv.max != 0) {
                json[min] = Helpers::round2(dv.min, 0, fahrenheit);
                json[max] = Helpers::round2(dv.max, 0, fahrenheit);
```

tp1de, 2022-01-18:
the field tempautotemp has half of actual value and min/max

MichaelDvP, 2022-01-18:
> the field tempautotemp has half of actual value and min/max

On your thermostat it is scaling 1/2, but others have 1/10 or 1/100 scaling, thats why stored limits are not scaled and limit for settemp is set to 5°C and 29°C. But actually the output uses the scaling.

tp1de, 2022-01-18:
I understand that it is quite difficult to scope with all different installations.
But this is one of the most used parameters I believe. Heating in automatic mode an temp is temporarily overwritten.
Maybe to be covered by settings?

tp1de, 2022-01-18:
its actual value as well and not just the limits:
{
  "name": "tempautotemp",
  "fullname": "hc1 temporary set temperature automode",
  "circuit": "hc1",
  "value": -0.5,
  "type": "number",
  "min": -63.5,
  "max": 63.5,
  "uom": "°C",
  "writeable": true,
  "visible": true
}

MichaelDvP, 2022-01-19:
Yes, the tempautotemp on RC300 is a piece of shit, scales with divider 2 but setting `-1` should send as 0xFF and 0xFF means deactivated. Setting it as UINT will show 0xFF as `not_set` and tempautotemp is not published. Then the HA users has complained the missing value on next auto-switchtime (@proddy: is this solved?). Handle it as INT now shows -0,5°C as value for deactivated. But sending -1 to emsesp should set the right value (0xFF).
If the HA problem is solved, we can set set thermostat.cpp line 3256 to 
```
register_device_value(tag, &hc->tempautotemp, DeviceValueType::UINT, FL_(div2), FL_(tempautotemp), DeviceValueUOM::DEGREES, MAKE_CF_CB(set_tempautotemp), -1, 29);
```
(**sic!** minimum -1 and UINT is right in this case)
Alternative there is (unused) code in line 987,ff to spoof a value 0 if 0xFF is received.

proddy, 2022-01-19:
> Then the HA users has complained the missing value on next auto-switchtime (@proddy: is this solved?)

yes, HA is fixed. Not seen any more errors. I solved it by dynamically adding and removing the HA MQTT Discovery topics. So when that piece of shit tempautotemp disappears, the MQTT topic will also be removed. I'm glad I did this since it turns out that MQTT Discovery is becoming a universal standard and also used by Domoticz now. Which is also a piece of.....

proddy, 2022-01-19:
@MichaelDvP when shall we take the plunge and merge the v3.4 branch into dev?

tp1de, 2022-01-19:
Isn't it a combination of numeric value range (5 ...30°C) and enum format: 0:off  -1:automatic /na ?

MichaelDvP, 2022-01-19:
@proddy IMO you should release current dev as main v3.3.1 and make v3.4 to beta dev. I think 3.4 is stable now and some more testers will help to find hidden issues.
@tp1de No it's a simple value. EMS-Bus has only a fixed type per field.

proddy, 2022-01-19:
> @proddy IMO you should release current dev as main v3.3.1 and make v3.4 to beta dev. I think 3.4 is stable now and some more testers will help to find hidden issues.

ok. I'll start updating the doc and screen shots too

MichaelDvP, 2022-01-19:
BTW: I've seen on Discord that some people (also @tp1de) have issues with reboot. There  was a long discussion in #156, with result of a instable DCDC regulator. In #289 i had also issues with my regulator (but no reboots) and reducing wifi-tx-power to ~10dBm helped. v3.4 have more sendings with individual mqtt, maybe more people getting this issue with premium gateway and MH-ET.
Add something to troubleshouting?

proddy, 2022-01-19:
yes, Kees said he would mention it on his website but I couldn't find it (https://bbqkees-electronics.nl/wiki/gateway/troubleshooting.html). I'll add to our doc site and mention this to those people on Discord. Thanks for reminding me about this issue

tp1de, 2022-01-19:
Interesting comments. So syslog all in hex produces a lot of traffic. I switched syslog out and it seems to be stable until now (20 hours). I will test reducing wifi-tx-power as well. Thanks.

proddy, 2022-01-24:
merged into dev

closed 2022-01-24 by proddy