emsesp/EMS-ESP32 #289: incomplete telegrams

MichaelDvP, 2022-01-10:
I had only very few incomplete telegrams in the past, but with v3.4 i recognize a larger number. A lot after boot, and fewer in normal operation, but with error-rate of ~0,5% (>99% quality). I checked what has changed. This is a gateway premium with MH-ET-D1-32 and resetted to factory defaults with v3.4. 
Result of all tests is that the wifi-sending of esp32 influences the comparator of bus-receiver. Disable the AP and the high error-rate on startup is vanished. Reduce tx-power from 20dBm to 12dBm and rx-quality is 100% with max. 1 error in 24h. 
I think i did not see this with the new gateway and it's MH-ET specific, but maybe add a hint to faq/doc here and @bbqkees.
If high tx-power is needed, maybe a metal shield between antenna and comparator will help.

proddy, 2022-01-10:
that's good dectective work! To be honest I've noticed this too but largely ignore it since I've been getting Rx errors since version 1.0. 

Do you think this is something in the newer espressif arduinio core that is causing this in v3.4?

Another thing I did change is removing the delay(1) from the emsesp() loop. Maybe that will give the Watchdog timers time to catchup?

bbqkees, 2022-01-10:
I never really tested a lot with the MH-ET Live chip. I tried it with the Premium II base board so I know it works, but did not do any stress testing.
I also never noticed any problems with the new S32 and E32 models so far.

Below my test boiler with E32 with 3.4a9 running for about 1,5 day:
[image]

And my house boiler with an E32 running 3.3 for +40 days:
[image]

The position of the ESP32 antenna with respect to the comparator chip is not really different for the S32/E32 and the Premium II with MH-ET Live. Also the antenna design is the same for all, even though they use an different brand of ESP32 chip.

Could it just be the green voltage regulator on the Premium II board that is bothered by interference from the ESP32 antenna?
The newer Gateway models all use a much more sophisticated regulator, which is also a tiny bit further away from the antenna.

bbqkees, 2022-01-10:
I'll add a note to the MH-ET upgrade instruction page in the wiki.

proddy, 2022-01-11:
I'll do a compare between versions a9 and a11 and see if anything has changed related to Rx/Tx

bbqkees, 2022-01-11:
I did mess with the thermostat cable yesterday. I tidied it up and now no incomplete telegrams.
So probably false alarm on a11.

bbqkees, 2022-01-11:
Still running fine so indeed false alarm on a11.

[image]

MichaelDvP, 2022-01-12:
I think it's not version dependent. This was only due to a factory reset, i had reduced tx-power before. And maybe not EMC from the antenna. It could be the transient current from sending, these peaks are much higher on esp32 than esp8266. I'll check if some extra capacity in 5V or 3.3V line will help. I have not seen this with gateway32S, so probably the 5V converter makes the difference.
But reducing the tx-power is a simple and effective way to reduce incomplete telegrams on MH-ET-systems.

bbqkees, 2022-01-12:
Yes that makes sense indeed. I think the green voltage regulator was a good match for the ESP8266 but the upgrade to the MH-ET board on the Premium II is more of a 'hack'. It works for a lot of people but the Premium II was not originally designed for such an upgrade.
I can consider including a better regulator for those who buy a complete MH-ET upgrade from me.

MichaelDvP, 2022-01-12:
BTW: to compete this information: With tx-power below 10dBm the esp32 also works buspowered with a only few incomplete telegrams.

closed 2022-01-12 by MichaelDvP