emsesp/EMS-ESP32 #290: Make prefix for MQTT Discovery configurable in v3.4

bbqkees, 2022-01-12:
HA's MQTT Discovery is quickly becoming the defacto standard for MQTT integrations.
This means other home automation systems like Domoticz are starting to implement it now.

Currently the discovery topic prefix is fixed to 'homeassistant', however someone with Domoticz or f.i. IOBroker does not run HA.
Therefore it would be nice if the prefix can be configured from within the settings.

Also the tickbox 'Enable Home Assistant integration' should be renamed (back) to 'Enable MQTT Discovery', as it is not only HA who uses the Discovery protocol.
[image]

proddy, 2022-01-12:
ok

proddy, 2022-01-13:
I'll default the 'discovery' topic prefix to `homeassistant` for backwards compatibility. Can domoticz live with the template conditionally code for the thermostat modes or should also be an option in the front-end?

proddy, 2022-01-16:
added this like:

[image]

bbqkees, 2022-01-17:
Yes that is fine.
Thanks.

closed 2022-01-17 by bbqkees