emsesp/EMS-ESP32 #294: Problem with system with more than 4 mixers.

Shmul1, 2022-01-15:
Good afternoon.

My heating system has the following equipment:
1. 1 x Junkers CerapurMaxx ZBR 100-3
2. 1 x Gateway MB LAN 2
3. 1 x CW400 Thermostat
4. 7 x MM100 Mixing modules

I use 5 MM100 modules for heating, 1 MM100 module for hot water and 1 MM100 module for heating the pool.
If I control the system via the Thermostat CW400, then everything works fine for me.
Now I decided to connect the heating system to the Home Assistant via MQTT and I bought BBQKees Gateway E32. I can see all my equipment perfectly in the EMS-ESP web interface, but there is one problem.
Mixer with 1-4 are shown as h1-h4, 5 becomes wwc1 Mixer, Mixer 6 becomes wwc2, but the mixer 7 (id8 for hot water) is wwc1!!!
Thus in EMS-ESP got two mixer wwc1, MQTT and considers them as one and constantly reads the information for sensors from one to the other mixer!!!
Is it possible to rename the mixer so that they are displayed correctly via MQTT?
Is it possible, if there are more than 4 mixers in the system, then, for example, mixers 1-8 should be assigned the values h1-h8, and mixers 9 - wwc1, 10 - wwc2 (for CW800 users, for example)?

proddy, 2022-01-15:
wow, that's one hell of a system. Could you send over the devices you have? http://ems-esp.local/api/system

proddy, 2022-01-15:
I'm also thinking of extending the customization service we have in v3.4 so that each device can be renamed

Shmul1, 2022-01-15:
{
  "System": {
    "version": "3.3.1b1",
    "uptime": "000+01:42:37.927",
    "uptime (seconds)": 6157,
    "freemem": 111,
    "reset reason": "Power on reset / APP CPU reseted by PRO CPU",
    "Dallas sensors": 0
  },
  "Network": {
    "connection": "WiFi",
    "hostname": "ems-esp",
    "SSID": "HomeNet",
    "BSSID": "74:4D:28:F6:F4:64",
    "RSSI": -66,
    "MAC": "8C:CE:4E:95:14:84",
    "IPv4 address": "192.168.10.25/255.255.255.0",
    "IPv4 gateway": "192.168.10.1",
    "IPv4 nameserver": "8.8.8.8"
  },
  "Status": {
    "bus status": "connected",
    "bus protocol": "HT3",
    "telegrams received": 20836,
    "read requests sent": 3504,
    "write requests sent": 0,
    "incomplete telegrams": 42,
    "tx fails": 0,
    "rx line quality": 100,
    "tx line quality": 100,
    "MQTT": "connected",
    "MQTT publishes": 8899,
    "MQTT publish fails": 0,
    "Dallas reads": 0,
    "Dallas fails": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Junkers Topline/GB162 (DeviceID:0x08 ProductID:115, Version:05.09)",
      "handlers": "0x10 0x11 0xC2 0x14 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "Thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:18.05)",
      "handlers": "0xA3 0x06 0xA2 0x12 0x2A5 0x2B9 0x2AF 0x29B 0x471 0x2A6 0x2BA 0x2B0 0x29C 0x472 0x2A7 0x2BB 0x2B1 0x29D 0x473 0x2A8 0x2BC 0x2B2 0x29E 0x474 0x2F5 0x31B 0x31D 0x31E 0x23A 0x267 0x240"
    },
    {
      "type": "Mixer",
      "name": "MM100 (DeviceID:0x20, ProductID:160, Version:24.05)",
      "handlers": "0x2D7"
    },
    {
      "type": "Mixer",
      "name": "MM100 (DeviceID:0x28, ProductID:160, Version:24.05)",
      "handlers": "0x331"
    },
    {
      "type": "Mixer",
      "name": "MM100 (DeviceID:0x23, ProductID:160, Version:24.05)",
      "handlers": "0x2DA"
    },
    {
      "type": "Mixer",
      "name": "MM100 (DeviceID:0x22, ProductID:160, Version:24.05)",
      "handlers": "0x2D9"
    },
    {
      "type": "Mixer",
      "name": "MM100 (DeviceID:0x21, ProductID:160, Version:24.05)",
      "handlers": "0x2D8"
    },
    {
      "type": "Mixer",
      "name": "MM100 (DeviceID:0x24, ProductID:160, Version:24.05)",
      "handlers": "0x2DB"
    },
    {
      "type": "Mixer",
      "name": "MM100 (DeviceID:0x25, ProductID:160, Version:24.05)",
      "handlers": "0x2DC"
    },
    {
      "type": "Gateway",
      "name": "KM200/MB LAN 2 (DeviceID:0x48, ProductID:189, Version:04.07)"
    },
    {
      "type": "Controller",
      "name": "BC10 (DeviceID:0x09, ProductID:114, Version:02.01)"
    }
  ]
}

MichaelDvP, 2022-01-15:
One thing is a simple naming issue, we have only names for h1-hc4, as the thermostats only have 4 heatingcircuits. (we need to insert TAG_HC5, -TAG_HC8)
The mixers device ID 0x20-0x27 are already assigned as heating, 0x28 and 0x29 as warm water. Now it's interesting what your mixers 0x24 and 0x25 sends, can you make a log from the telegrams (more than a minute).
I think ID 0x24 (mixer 5 is wrongly named wwc1 but shows right values. ID 0x25 (named wwc2) is the pool? Then the handler 0x2DC is not right.

Shmul1, 2022-01-15:
can you make a log from the telegrams (more than a minute).

Can you tell me please, what should I do exactly, Iam first day user :)))

Shmul1, 2022-01-15:
2022-01-15 19:09:34.935 D 256: [telegram] Sending read Tx [#108], telegram: 8B 90 FF 00 20 01 9C
2022-01-15 19:09:34.992 D 257: [emsesp] Last Tx read successful
2022-01-15 19:09:35.685 D 258: [telegram] Sending read Tx [#109], telegram: 8B 90 FF 00 20 01 A7
2022-01-15 19:09:35.781 D 259: [emsesp] Last Tx read successful
2022-01-15 19:09:36.585 D 260: [telegram] Sending read Tx [#129], telegram: 8B 90 FF 19 20 01 A7
2022-01-15 19:09:36.661 D 261: [emsesp] Last Tx read successful
2022-01-15 19:09:37.661 D 262: [telegram] Sending read Tx [#110], telegram: 8B 90 FF 00 20 01 BB
2022-01-15 19:09:37.746 D 263: [emsesp] Last Tx read successful
2022-01-15 19:09:38.385 D 264: [telegram] Sending read Tx [#111], telegram: 8B 90 FF 00 20 01 B1
2022-01-15 19:09:38.452 D 265: [emsesp] Last Tx read successful
2022-01-15 19:09:39.035 D 266: [telegram] Sending read Tx [#112], telegram: 8B 90 FF 00 20 01 9D
2022-01-15 19:09:39.095 D 267: [emsesp] Last Tx read successful
2022-01-15 19:09:39.735 D 268: [telegram] Sending read Tx [#113], telegram: 8B 90 FF 00 20 01 A8
2022-01-15 19:09:39.836 D 269: [emsesp] Last Tx read successful
2022-01-15 19:09:40.409 D 270: [telegram] Sending read Tx [#130], telegram: 8B 90 FF 19 20 01 A8
2022-01-15 19:09:40.484 D 271: [emsesp] Last Tx read successful
2022-01-15 19:09:41.185 D 272: [telegram] Sending read Tx [#114], telegram: 8B 90 FF 00 20 01 BC
2022-01-15 19:09:41.273 D 273: [emsesp] Last Tx read successful
2022-01-15 19:09:41.885 D 274: [telegram] Sending read Tx [#115], telegram: 8B 90 FF 00 20 01 B2
2022-01-15 19:09:41.947 D 275: [emsesp] Last Tx read successful
2022-01-15 19:09:42.536 D 276: [telegram] Sending read Tx [#116], telegram: 8B 90 FF 00 20 01 9E
2022-01-15 19:09:42.598 D 277: [emsesp] Last Tx read successful
2022-01-15 19:09:43.067 D 278: [mqtt] Publishing topic ems-esp/boiler_data (#10911, retain=0, retry=1, size=608, pid=1)
2022-01-15 19:09:43.168 D 279: [mqtt] Publishing topic ems-esp/boiler_data_ww (#10912, retain=0, retry=1, size=234, pid=1)
2022-01-15 19:09:43.235 D 280: [telegram] Sending read Tx [#117], telegram: 8B 90 FF 00 20 01 F5
2022-01-15 19:09:43.269 D 281: [mqtt] Publishing topic ems-esp/thermostat_data (#10913, retain=0, retry=1, size=145, pid=1)
2022-01-15 19:09:43.320 D 282: [emsesp] Last Tx read successful
2022-01-15 19:09:43.370 D 283: [mqtt] Publishing topic ems-esp/thermostat_data_hc1 (#10914, retain=0, retry=1, size=357, pid=1)
2022-01-15 19:09:43.471 D 284: [mqtt] Publishing topic ems-esp/thermostat_data_hc2 (#10915, retain=0, retry=1, size=350, pid=1)
2022-01-15 19:09:43.572 D 285: [mqtt] Publishing topic ems-esp/thermostat_data_hc3 (#10916, retain=0, retry=1, size=357, pid=1)
2022-01-15 19:09:43.673 D 286: [mqtt] Publishing topic ems-esp/thermostat_data_hc4 (#10917, retain=0, retry=1, size=350, pid=1)
2022-01-15 19:09:43.774 D 287: [mqtt] Publishing topic ems-esp/mixer_data_hc1 (#10918, retain=0, retry=1, size=79, pid=1)
2022-01-15 19:09:43.875 D 288: [mqtt] Publishing topic ems-esp/mixer_data_wwc1 (#10919, retain=0, retry=1, size=51, pid=1)
2022-01-15 19:09:43.910 D 289: [telegram] Sending read Tx [#118], telegram: 8B 90 FF 00 20 02 1B
2022-01-15 19:09:43.959 D 290: [emsesp] Last Tx read successful
2022-01-15 19:09:43.976 D 291: [mqtt] Publishing topic ems-esp/mixer_data_hc4 (#10920, retain=0, retry=1, size=80, pid=1)
2022-01-15 19:09:44.077 D 292: [mqtt] Publishing topic ems-esp/mixer_data_hc3 (#10921, retain=0, retry=1, size=79, pid=1)
2022-01-15 19:09:44.178 D 293: [mqtt] Publishing topic ems-esp/mixer_data_hc2 (#10922, retain=0, retry=1, size=80, pid=1)
2022-01-15 19:09:44.279 D 294: [mqtt] Publishing topic ems-esp/mixer_data_wwc1 (#10923, retain=0, retry=1, size=79, pid=1)
2022-01-15 19:09:44.380 D 295: [mqtt] Publishing topic ems-esp/mixer_data_wwc2 (#10924, retain=0, retry=1, size=79, pid=1)
2022-01-15 19:09:44.610 D 296: [telegram] Sending read Tx [#119], telegram: 8B 90 FF 00 20 01 3A
2022-01-15 19:09:44.655 D 297: [emsesp] Last Tx read successful
2022-01-15 19:09:45.235 D 298: [telegram] Sending read Tx [#120], telegram: 8B 90 FF 00 20 01 40
2022-01-15 19:09:45.315 D 299: [emsesp] Last Tx read successful
2022-01-15 19:09:46.011 D 300: [telegram] Sending read Tx [#121], telegram: 8B A8 FF 00 20 02 31
2022-01-15 19:09:46.083 D 301: [emsesp] Last Tx read successful
2022-01-15 19:09:47.085 D 302: [telegram] Sending read Tx [#122], telegram: 8B A3 FF 00 20 01 DA
2022-01-15 19:09:47.147 D 303: [emsesp] Last Tx read successful
2022-01-15 19:09:47.735 D 304: [telegram] Sending read Tx [#123], telegram: 8B A2 FF 00 20 01 D9
2022-01-15 19:09:47.796 D 305: [emsesp] Last Tx read successful
2022-01-15 19:09:48.435 D 306: [telegram] Sending read Tx [#124], telegram: 8B A1 FF 00 20 01 D8
2022-01-15 19:09:48.497 D 307: [emsesp] Last Tx read successful
2022-01-15 19:09:49.085 D 308: [telegram] Sending read Tx [#125], telegram: 8B A4 FF 00 20 01 DB
2022-01-15 19:09:49.147 D 309: [emsesp] Last Tx read successful
2022-01-15 19:09:49.604 D 310: [emsesp] No telegram type handler found for ID 0x255 (src 0x23)
2022-01-15 19:09:50.335 D 311: [telegram] Sending read Tx [#126], telegram: 8B A5 FF 00 20 01 DC
2022-01-15 19:09:50.397 D 312: [emsesp] Last Tx read successful
2022-01-15 19:09:53.086 D 313: [mqtt] Publishing topic ems-esp/boiler_data (#10925, retain=0, retry=1, size=608, pid=1)
2022-01-15 19:09:53.187 D 314: [mqtt] Publishing topic ems-esp/boiler_data_ww (#10926, retain=0, retry=1, size=234, pid=1)
[... cut ...]
2022-01-15 19:11:04.015 D 542: [mqtt] Publishing topic ems-esp/mixer_data_hc3 (#11034, retain=0, retry=1, size=79, pid=1)
2022-01-15 19:11:04.116 D 543: [mqtt] Publishing topic ems-esp/mixer_data_hc2 (#11035, retain=0, retry=1, size=80, pid=1)
2022-01-15 19:11:04.217 D 544: [mqtt] Publishing topic ems-esp/mixer_data_wwc1 (#11036, retain=0, retry=1, size=79, pid=1)
2022-01-15 19:11:04.318 D 545: [mqtt] Publishing topic ems-esp/mixer_data_wwc2 (#11037, retain=0, retry=1, size=79, pid=1)
2022-01-15 19:11:06.380 D 546: [emsesp] No telegram type handler found for ID 0x255 (src 0x21)

MichaelDvP, 2022-01-15:
Good, but use log level `ALL` to see the telegrams, You can download to a text-file and drop the file here.

Shmul1, 2022-01-15:
ok

Shmul1, 2022-01-15:
[log.txt](https://github.com/emsesp/EMS-ESP32/files/7875451/log.txt)

Shmul1, 2022-01-15:
0x20-0x23 and 0x25 - hc
0x24 - pool
0x28 - wwc

Shmul1, 2022-01-15:
EMS Devices screen:
0x24 & 0x28 - wwc1
0x25 - wwc2

MichaelDvP, 2022-01-15:
Ok, looks like the pool behave like a normal heating circuit and telegram 0x2DB is right:
```
2022-01-15 19:13:50.658 T 1155: [emsesp] Me(0x0B) <- Mixer(0x24), MMPLUSStatusMessage_HC(0x2DB), data: 20
2022-01-15 19:13:50.720 D 1156: [emsesp] Last Tx read successful
2022-01-15 19:13:50.720 T 1157: [emsesp] Mixer(0x24) -> Me(0x0B), MMPLUSStatusMessage_HC(0x2DB), data: 01 00 00 80 00 46 08 00 08
```
Then we only need to add the names. The telegram contains no measured temperature, flow-setpoint is 70°C, 

@proddy should i make a pr to 3.4, or dev-branch?

Shmul1, 2022-01-15:
I have made a little video. First, the sensors receive data from 0x28, and then from 0x2

https://user-images.githubusercontent.com/51746237/149632083-16b32c5b-d03f-4cc3-8af5-60b1d4bed712.mp4

4

MichaelDvP, 2022-01-15:
I've added to proddys v3.4alpha, please check if this solves the issue. After uploading the bin you need to refresh the page in your browser.
[EMS-ESP-3_4_0a13-ESP32.zip](https://github.com/emsesp/EMS-ESP32/files/7875538/EMS-ESP-3_4_0a13-ESP32.zip)

Shmul1, 2022-01-15:
thanks, uploading

Shmul1, 2022-01-15:
Now I see all mixers in HA, thank you! This firmware is a very big progress in your work!

proddy, 2022-01-15:
thanks @MichaelDvP for the quick fix

MichaelDvP, 2022-01-16:
Next step is to add the heating circuits to the thermostat. I've found on NorbertS' list that this thermostat have 8 ciircuits in row. I'm curious if the buffers for web/mqtt are large enought for all these circuits. Please check dashboard for thermostat, mqtt and [http://ems-esp.local/api/thermostat](http://ems-esp.local/api/thermostat)
Here is the bin with more thermostat hcs. 
[EMS-ESP-3_4_0a13-ESP32.zip](https://github.com/emsesp/EMS-ESP32/files/7876587/EMS-ESP-3_4_0a13-ESP32.zip)

Shmul1, 2022-01-16:
{
  "errorcode": "A35(3125)",
  "datetime": "10:46:55 16.01.2022",
  "floordry": 0,
  "dampedoutdoortemp": 0.9,
  "floordrytemp": 0,
  "building": 1,
  "minexttemp": -20,
  "wwsettemp": 60,
  "wwmode": 4,
  "wwsettemplow": 45,
  "wwcircmode": 3,
  "wwchargeduration": 60,
  "wwcharge": "off",
  "wwextra1": 0,
  "wwdisinfecting": "on",
  "wwdisinfectday": 1,
  "wwdisinfecttime": 120,
  "hc2": {
    "seltemp": 21,
    "mode": 1,
    "modetype": 1,
    "ecotemp": 29.5,
    "manualtemp": 21,
    "comforttemp": 30,
    "summertemp": 17,
    "designtemp": 45,
    "offsettemp": 0,
    "minflowtemp": 25,
    "maxflowtemp": 43,
    "roominfluence": 3,
    "curroominfl": 0,
    "nofrosttemp": 5,
    "targetflowtemp": 45,
    "heatingtype": 3,
    "summersetmode": 1,
    "summermode": "off",
    "controlmode": 6,
    "program": 0,
    "tempautotemp": -0.5,
    "fastheatup": 0
  },
  "hc4": {
    "seltemp": 21,
    "mode": 1,
    "modetype": 1,
    "ecotemp": 29.5,
    "manualtemp": 21,
    "comforttemp": 30,
    "summertemp": 17,
    "designtemp": 45,
    "offsettemp": 0,
    "minflowtemp": 25,
    "maxflowtemp": 43,
    "roominfluence": 3,
    "curroominfl": 0,
    "nofrosttemp": 5,
    "targetflowtemp": 45,
    "heatingtype": 3,
    "summersetmode": 1,
    "summermode": "off",
    "controlmode": 6,
    "program": 0,
    "tempautotemp": -0.5,
    "fastheatup": 0
  },
  "hc1": {
    "seltemp": 21,
    "mode": 1,
    "modetype": 1,
    "curroominfl": 0,
    "targetflowtemp": 45,
    "summermode": "off"
  },
  "hc3": {
    "seltemp": 21,
    "mode": 1,
    "modetype": 1,
    "ecotemp": 29.5,
    "manualtemp": 21,
    "comforttemp": 30,
    "summertemp": 17,
    "designtemp": 45,
    "offsettemp": 0,
    "minflowtemp": 25,
    "maxflowtemp": 43,
    "roominfluence": 3,
    "curroominfl": 0,
    "nofrosttemp": 5,
    "targetflowtemp": 45,
    "heatingtype": 3,
    "summersetmode": 1,
    "summermode": "off",
    "controlmode": 6,
    "program": 0,
    "tempautotemp": -0.5,
    "fastheatup": 0
  },
  "hc5": {
    "seltemp": 21,
    "mode": 1,
    "modetype": 1,
    "ecotemp": 15,
    "manualtemp": 21,
    "comforttemp": 30,
    "summertemp": 17,
    "designtemp": 75,
    "offsettemp": 0,
    "minflowtemp": 25,
    "maxflowtemp": 75,
    "roominfluence": 3,
    "curroominfl": 0,
    "nofrosttemp": 5,
    "targetflowtemp": 70,
    "heatingtype": 1,
    "summersetmode": 1,
    "summermode": "off",
    "controlmode": 6,
    "program": 0,
    "tempautotemp": -0.5,
    "fastheatup": 0
  },
  "hc6": {
    "seltemp": 21,
    "mode": 1,
    "modetype": 1,
    "ecotemp": 29.5,
    "manualtemp": 21,
    "comforttemp": 30,
    "summertemp": 17,
    "designtemp": 45,
    "offsettemp": 0,
    "minflowtemp": 25,
    "maxflowtemp": 43,
    "roominfluence": 3,
    "curroominfl": 0,
    "nofrosttemp": 5,
    "targetflowtemp": 45,
    "heatingtype": 3,
    "summersetmode": 1,
    "summermode": "off",
    "controlmode": 6,
    "program": 0,
    "tempautotemp": -0.5,
    "fastheatup": 0
  }
}

MichaelDvP, 2022-01-16:
Looks good, only hc1 have to few values, wait a minute to allow emsesp to read all values. Does web-dashboard also show all values for all hc? Also check Settings -> Customization for the thermostat.

Shmul1, 2022-01-16:
Yes? dashboard shows all. Customization ok.
I have restarted EMS-ESP but hc1 still have to few values,
[telegrams.txt](https://github.com/emsesp/EMS-ESP32/files/7876637/telegrams.txt)
.

MichaelDvP, 2022-01-16:
> I have restarted EMS-ESP but hc1 still have to few values,

After restart it always take some minutes to collect all data from bus. 
Do you use mqtt? Then check the mqtt-publish for thermostat and mixer, especially for (MQTT) nested formatting and (Settings) Enum-format Value. I think the 4k buffer for mqtt is to small for nested formatting.

proddy, 2022-01-16:
the API call to http://ems-esp.local/api/thermostat uses a large buffer (EMSESP_JSON_SIZE_XXLARGE_DYN in WebAPIService.cpp) so I would expect to see all the hc1 values in that call

Shmul1, 2022-01-16:
I have changed Enum-format from Index to Value and now I have: Boiler entities 43 (was 56), thermostat entities 133 (was 133). But  http://ems-esp.local/api/thermostat now shows "null", customization is empty for all devices. I try to watch entities by tapping on device and the list of sensors do not appear.
After that I changed  Enum-format to Index back,  Now I have: Boiler entities 42, thermostat entities 133. situation with API, sensor list and customization the same, it doesn't work

Shmul1, 2022-01-16:
I do all these actions with a delay of more than 5 minutes

Shmul1, 2022-01-16:
Now I have uploaded firmware and the problem with customization, API and entity list disappeared

Shmul1, 2022-01-16:
Enum-format - Value 
[thermostats.txt](https://github.com/emsesp/EMS-ESP32/files/7876873/thermostats.txt)

MichaelDvP, 2022-01-16:
I'm confused that hc1 is not sorted to front and it contains only values from the hc1-monitor telegram. @proddy any idea why?
@Shmul1 Please make another >1 min log with loglevel ALL i like to see if the other telegrams are empty or not fetched.

@proddy what do you think about inserting in `EMSESP::publish_device_values`
```
        if (doc.overflowed()) {
            LOG_WARNING(F("buffer overflow, please use individual topics"));
        }
```

Shmul1, 2022-01-16:
[log (1).txt](https://github.com/emsesp/EMS-ESP32/files/7877034/log.1.txt)

MichaelDvP, 2022-01-16:
Thanks, i see the problem. 
1. @proddys new hextoa() cuts the high byte from type-id, but this is cosmetic.
2. the tx-queue overflows and the first messages are ignored, Maybe we should wait for empty queue before generating the messages for next device. Quick solution is to enlarge the queue.

MichaelDvP, 2022-01-16:
This is the quick solution, i think now all values should be there. I've added the suggested warning for mqtt overflow.
[EMS-ESP-3_4_0a13-ESP32.zip](https://github.com/emsesp/EMS-ESP32/files/7877275/EMS-ESP-3_4_0a13-ESP32.zip)

proddy, 2022-01-16:
thanks @MichaelDvP for those fixes

Shmul1, 2022-01-16:
Good job, thank you!

Shmul1, 2022-01-16:
[thermostats.txt](https://github.com/emsesp/EMS-ESP32/files/7877778/thermostats.txt)

MichaelDvP, 2022-01-18:
@Shmul1 Have you already checked the mqtt-output?

Shmul1, 2022-01-18:
Yes, of course, I sent the log back on Sunday. The firmware works well, but there is one more thing. In your firmware, warm water is prepared only through a circuit in the boiler, and in my case through a separate circuit "wwc1", so the sensors associated with warm water do not work. Is it possible to fix this?

MichaelDvP, 2022-01-18:
>  Is it possible to fix this?

It need some work, but i think it's possible. The wwc has two telegrams:
`Mixer(0x28) -> All(0x00), MMPLUSStatusMessage_WWC(0x331), data: 02 4E 00 3C 00 3C 3C 46 02 03 04 00 3C`
here we know only 024E as flowtemp, 00 as pumpstatus and offset 11: 00 as status.  The other bytes are unknown, one of the 3C could be the settemp (60°C)?
`Thermostat(0x10) -> Mixer(0x28), ?(0x33B), data: 01 02 00`
is a setting, but we do not know what is set by the thermostat. 
You have to log the telegrams and change the values you want to have on the thermostat, if we know the new value and find it in a telegram ``Thermostat(0x10) -> Mixer(0x28), ...`, we can add the value and the setting.

Shmul1, 2022-01-18:
Ok, I will do it today

Shmul1, 2022-01-18:
I have changed different settings and made the lo
[log (2).txt](https://github.com/emsesp/EMS-ESP32/files/7891112/log.2.txt)

Shmul1, 2022-01-18:
Please check if this is enough

MichaelDvP, 2022-01-18:
I see changing values in the unknown telegram 0x313 on offsets 4, 5, 10 but as written before i need a reference what value has changed and to what value is has changed. First change was offset 10 to 0x41, maybe a temperature to 65°C, or a bitfield bits 0 and 7 set?

Shmul1, 2022-01-18:
Maybe it's better to make a video and make a log at the same time?

Shmul1, 2022-01-18:
[log (3).txt](https://github.com/emsesp/EMS-ESP32/files/7891554/log.3.txt)

https://user-images.githubusercontent.com/51746237/150002075-d7fe4a78-8a1c-42b4-a639-54342391ad82.mp4

MichaelDvP, 2022-01-19:
Ok, here are some more values. Some settings (like disinfect day, etc. are already in thermostat. Some values you have skipped, and values you have shown on video but not changed are not logged. But it's a good idea with the video.
[EMS-ESP-3_4_0a16-ESP32.zip](https://github.com/emsesp/EMS-ESP32/files/7897722/EMS-ESP-3_4_0a16-ESP32.zip)
Please check the values and settings, i think we will need some more iterations to have all working.

Shmul1, 2022-01-19:
Good afternoon. Unfortunately, I had to leave on a business trip before Saturday and now I can only do everything remotely. I'm ready to make more videos on Saturday and Sunday. I installed the firmware, there is integration in HA, but I can test it in more detail only at night. Thank you for your work
[log.txt](https://github.com/emsesp/EMS-ESP32/files/7899194/log.txt)
.

proddy, 2022-01-26:
> I'm confused that hc1 is not sorted to front and it contains only values from the hc1-monitor telegram. @proddy any idea why? 

@MichaelDvP this was a bug. I was sorting the vector but always returning the last hc in the list. It still needs to be fixed.  https://github.com/emsesp/EMS-ESP32/blob/8c4e5e5185e967a719fc0f8ccc48679fba5af976/src/devices/thermostat.cpp#L335.

proddy, 2022-01-26:
Hi @Shmul1, if you don't mind I'd like to use some of your data as test data for EMS-ESP. I don't have any mixers, gateways or even a thermostat than can support more than one heating circuit! When you have time would you mind sending over some data which I can re-use?
http://ems-esp.local/api/thermostat
http://ems-esp.local/api/mixer

thanks

Shmul1, 2022-01-26:
Hello, sure, no problem.  I'm the only one who can't make a video yet, I'm still on a business trip, otherwise I'll do what I can do remotely.

Shmul1, 2022-01-26:
[mixer.txt](https://github.com/emsesp/EMS-ESP32/files/7941255/mixer.txt)
[mixer_info.txt](https://github.com/emsesp/EMS-ESP32/files/7941256/mixer_info.txt)
[thermostat.txt](https://github.com/emsesp/EMS-ESP32/files/7941257/thermostat.txt)
[thermostat_info.txt](https://github.com/emsesp/EMS-ESP32/files/7941258/thermostat_info.txt)

commit 6e457c2a (2022-01-15): Merge pull request #296 from MichaelDvP/v3.4

fix #294
closed 2022-02-18 by proddy