emsesp/EMS-ESP32 #295: [Question] Required MQTT commands

FireWizard52, 2022-01-15:
Actually this question applies version 2, but as the repository for ESP8286 is read-only, I have to put it here, but it might be applicable for this version as well.

I use the device (with ESP8286) together with a Nefit Smartline and the Moduline 400 Thermostat in combination with Domoticz.

As the latest betas of Domoticz support MQTT Auto Discovery (like Home Assistant) the Domoticz plugin of BBQKees has been declared obsolete and he suggested to use Auto Discovery with MQTT.
However this will not work with the latest version 2.2.2. for EMS8286 as this version does njot supportAuto Discovery.

This is not a problem, as you can do it manually, but now I am stuck with the correct commands to set the mode and the setpoint in the Thermostat.
The Wiki and documentation is not really clear on this.

> EMS-ESP will subscribe to specific topics depending on the EMS devices attached. For example boiler, thermostat etc. Commands can be sent to EMS-ESP via these topics using the payload format:

What are the specific topics that the EMS-ESP (with ESP8286) subscribes to?

I tried ems-esp/thermostat and ems-esp/thermostat_cmd with the payload:

[image]

The link in the documentation for "Commands" points to a non existing page

proddy, 2022-01-15:
to be honest I don't know anymore. v2 was a long time ago. Why not upgrade to v3 and you'll benefit from all the new MQTT discovery code plus a really nice UI

MichaelDvP, 2022-01-15:
It should be topic: `ems-esp/thermostat` and payload `{"cmd"="mode", "data"="auto", "hc"=1}` for hc1 as default there is no need to send hc. 
But the esp32 version is much better, i recomment to spend a few € and upgrade.

FireWizard52, 2022-01-15:
Hello @proddy and @MichaelDvP

Thank you both for your quick reply.
 
Michael, I assume you mean payload {"cmd":"mode", "data":"auto", "hc":1}

As said, I tested that topic and payload before. It appeared that I had to restart the EMS-ESP.
After a restart (and I have no idea why) everything worked as expected. So the issue is solved.

> Why not upgrade to v3 and you'll benefit from all the new MQTT discovery code plus a really nice UI

and

> But the esp32 version is much better, i recomment to spend a few € and upgrade.

I bought this unit in March 2020, so less than 2 years ago. I'm satisfied with the unit and the software.
Of course I can invest again something like 80 €, but what do I gain?
Okay, I get MQTT Auto Discovery, but the majority of published topics I do not use in Domoticz.
Perhaps I get a better GUI, which I hardly use.
But the most important reason is that my boiler has almost reach EOL. It has been installed in 2003, so quite old for a boiler.
As long as it works, it is okay, but I do not make extra costs. If I buy a supported new boiler I can invest in a new EMS-ESP version.

But the issue is solved andthanks for your replies,

Regards

MichaelDvP, 2022-01-16:
If itworks for you, it's ok, but if you want to upgrade you dont need a new gateway, you can change only the processor-board to a MH-ET mini,which is ~5€ from aliexpress or 9€ fro amazon/ebay ([This one](https://www.amazon.de/AZDelivery-Bluetooth-Internet-Entwicklungsboard-kompatibel/dp/B08BTRQNB3/ref=sr_1_33_sspa?crid=3R65G4O5QYHK3&keywords=esp32%2Bmini%2Bd1&qid=1642320579&sprefix=esp32%2Bmin%2Caps%2C412&sr=8-33-spons&smid=A1X7QLRQH87QA3&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUEySklIVkhRNFhFVDI2JmVuY3J5cHRlZElkPUEwMTE1MzM4MTlHRkFNVkZYOU5IUyZlbmNyeXB0ZWRBZElkPUEwNzMxMzM0MVVORUU5NkxaRldQOSZ3aWRnZXROYW1lPXNwX2F0Zl9uZXh0JmFjdGlvbj1jbGlja1JlZGlyZWN0JmRvTm90TG9nQ2xpY2s9dHJ1ZQ&th=1)). You have to solder the pins and flash via usb, if you dont want that, bbqkees also sell soldered/programmed modules for 15€,

FireWizard52, 2022-01-16:
Hello @MichaelDvP 

Thanks for the information. It is something to consider and also the module of bbqkees is a not too expensive solution.
Probably the delivery is much quicker than the other two. currently

Regards

closed 2023-02-19 by proddy