emsesp/EMS-ESP32 #299: Climate control component for dhw in HA not present due to non functional  select.boiler_dhw_comfort entity

pausenpepe, 2022-01-16:
Entity select.boiler_dhw_comfort entity is not working properly and therefore no hot water control climate component is created.
Platform: 3.4.0a14
Home Assistant 2021.12.9
Hardware: ems-esp S32
Hardware heater: Buderus GB152
Hardware thermostat: RC35
http://ems-esp.local/api/system 
[system.txt](https://github.com/emsesp/EMS-ESP32/files/7878001/system.txt)
http://ems-esp.local/api/thermostat
[thermostat.txt](https://github.com/emsesp/EMS-ESP32/files/7878002/thermostat.txt)

BR Dirk

proddy, 2022-01-17:
The thermostat and boiler are different entities and not related. Let's look at both

- **non-functional boiler_dhw_comfort** :  What is being shown in the list and what is not working? Does it work via the EMS-ESP WebUI and Console. Are there errors in your Home Assistant logs? Any errors in the EMS-ESP logs?

- **climate control in HA** : If you have MQTT Discovery correctly configured then the climate control should be created. Check in HA->Configuration->Devices->EMS-ESP Thermostat. If not check your MQTT broker and use an application like MQTTExplorer to see if a topic exists with `climate` in the name. In HA you need to build your own lovelace dashboards. See https://www.home-assistant.io/getting-started/

MichaelDvP, 2022-01-17:
With [BC10 controller](https://bbqkees-electronics.nl/wp-content/uploads/ems-devices/bc10-e1581426090307.jpg) the `wwcomfort` is set by the left hardware rotary knob an can not be set by software.

pausenpepe, 2022-01-17:
**- non-functional boiler_dhw_comfort** : What is being shown in the list and what is not working? Does it work via the EMS-ESP WebUI and Console. Are there errors in your Home Assistant logs? Any errors in the EMS-ESP logs?
Within EMS-ESP Boiler UI the Boiler dhw comfort is not working properly (last item):
[image]
Selecting e.g. ECO
[image]
Leads to:
[image]
Which looks ok but it is probably not transferred to the gateway. After a brower refresh it is still set to "hot"
[image]

From HA /config/logs it is showing only awarning for an missing attribute for tankbottomtemp
Logger: homeassistant.helpers.template
Source: helpers/template.py:1779
First occurred: 5:20:48 PM (150 occurrences)
Last logged: 5:22:39 PM

Template variable warning: 'dict object' has no attribute 'tankbottomtemp' when rendering '{{value_json.tankbottomtemp}}'
Template variable warning: 'dict object' has no attribute 'tankmaxtemp' when rendering '{{value_json.tankmaxtemp}}'
Template variable warning: 'dict object' has no attribute 'solarpumpmodulation' when rendering '{{value_json.solarpumpmodulation}}'
Template variable warning: 'dict object' has no attribute 'tankheated' when rendering '{{value_json.tankheated}}'
Template variable warning: 'dict object' has no attribute 'activated' when rendering '{{value_json.activated}}'

 Inside EMS-ESP there are no error inside the log if i choose log level ERROR.
 If check log level debug: 
000+00:07:01.175 D 307: [mqtt] Publishing topic ems-esp/heartbeat (#514, retain=0, retry=1, size=255, pid=1)
000+00:07:01.180 D 308: [telegram] Sending read Tx [#184], telegram: 0B 90 3D 00 20
000+00:07:01.381 D 309: [emsesp] Last Tx read successful
000+00:07:01.767 D 310: [command] Calling command 'boiler/wwcomfort' (comfort) with value intelligent
000+00:07:01.858 D 311: [telegram] Sending write Tx [#190], telegram: 0B 08 33 09 EC
000+00:07:01.858 D 312: [emsesp] Process setting 0x33 for device 0x08
000+00:07:01.883 D 313: [emsesp] Last Tx write successful
000+00:07:01.883 D 314: [telegram] Sending post validate read, type ID 0x33 to dest 0x08
000+00:07:02.456 D 315: [emsesp] Process setting 0x33 for device 0x08
000+00:07:02.508 D 316: [telegram] Sending read Tx [#191], telegram: 0B 88 33 00 20
000+00:07:02.508 D 316: [telegram] Sending read Tx [#191], telegram: 0B 88 33 00 20
000+00:07:02.553 D 317: [emsesp] Last Tx read successful
000+00:07:02.589 D 318: [mqtt] Publishing topic ems-esp/boiler_data (#515, retain=0, retry=1, size=813, pid=1)
000+00:07:02.690 D 319: [mqtt] Publishing topic ems-esp/boiler_data_ww (#516, retain=0, retry=1, size=503, pid=1)
000+00:07:03.357 D 320: [telegram] Sending read Tx [#189], telegram: 0B 90 3D 1B 20
000+00:07:03.478 D 321: [emsesp] Last Tx read successful
000+00:07:03.782 D 322: [telegram] Sending read Tx [#185], telegram: 0B 90 3F 00 20
000+00:07:04.006 D 323: [emsesp] Last Tx read successful
000+00:07:04.307 D 324: [telegram] Sending read Tx [#192], telegram: 0B 90 3F 1B 20
000+00:07:04.510 D 325: [emsesp] Last Tx read successful
000+00:07:04.757 D 326: [telegram] Sending read Tx [#193], telegram: 0B 90 3F 36 20
000+00:07:04.885 D 327: [emsesp] Last Tx read successful
000+00:07:05.207 D 328: [telegram] Sending read Tx [#194], telegram: 0B 90 3F 51 20
000+00:07:05.361 D 329: [emsesp] Last Tx read successful
000+00:07:05.635 D 330: [telegram] Sending read Tx [#186], telegram: 0B 90 A5 00 20
000+00:07:05.757 D 331: [emsesp] Last Tx read successful
000+00:07:06.181 D 332: [telegram] Sending read Tx [#195], telegram: 0B 90 A5 1B 20
000+00:07:06.318 D 333: [emsesp] Last Tx read successful
000+00:07:06.636 D 334: [telegram] Sending read Tx [#187], telegram: 0B 90 37 00 20
000+00:07:06.713 D 335: [emsesp] Last Tx read successful
000+00:07:07.006 D 336: [telegram] Sending read Tx [#188], telegram: 0B B0 96 00 20
000+00:07:07.119 D 337: [emsesp] Last Tx read successful

It looks like HA is sending the data, but it is processed inside the EMS-ESP.
The EMS-ESP API Dashboard is still reporting dhw comfort as set to hot.

I hope this info helps to understand the situation better.

pausenpepe, 2022-01-17:
### Proddy wrotes:
> climate control in HA : If you have MQTT Discovery correctly configured then the climate control should be created. Check in HA->Configuration->Devices->EMS-ESP Thermostat.

The normal heater thermostat "Thermostat hc1" is working like a charm:
[image]
It would nice to have the same control type for dhw

proddy, 2022-01-17:
Michael is correct, your GB152 has a BC10 which is controlled manually with dials. It doesn't support comfort and I couldn't find it in the manual either. EMS-ESP can only simulate the functions that are supported by the actual EMS device.

You could in theory create a climate control in Home Assistant based on the "Home Assistant HVAC" control, but it's not what is it intended for. But you can try with a few lines of code. See how to do this in https://www.home-assistant.io/integrations/climate.mqtt/

pausenpepe, 2022-01-17:
Thanks for the clarification :) @proddy and @MichaelDvP 
I will check for getting a little bit into the climate.mqtt documentation.
I just checked the german manual for the BC10 and can confirm it. For the RC35 the setting for the dhw temperatur has to be set to Aut, which enables the RC35 to control the dhw temperature.
see:
[https://buderus-de-de.boschtt-documents.com/download/pdf/file/6720640987.pdf?token=ak073p2fcplq6r6uqu4ctf23r6](url)
BR Dirk

closed 2022-01-17 by pausenpepe

MichaelDvP, 2022-01-18:
I've checked with my GB125 (not 152) with BC10, the boiler accepts the command (N5), keeps the setting (N6), but the BC10-controller instantly overwrite to hot again(N7). Knob is set to auto to allow changing dhw temperature over the bus (RC35/emsesp).
```
ems-esp:# watch 33
Watching incoming telegrams, displayed in decoded format
Filtering only telegrams that match a deviceID or telegram type of 0x33
ems-esp:# 000+03:14:01.610 N 1: [emsesp] Me(0x0B) <- Boiler(0x08), UBAParameterWW(0x33), data: 1B
ems-esp:# 000+03:14:01.728 N 2: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameterWW(0x33), data: 08 FF 30 FB FF 28 FF 07 3C 00 00
cems-esp:# call boiler wwcomfort eco
ems-esp:# 000+03:14:08.025 N 3: [emsesp] Me(0x0B) -> Boiler(0x08), UBAParameterWW(0x33), data: D8 (offset 9)
ems-esp:# 000+03:14:09.173 N 4: [emsesp] Me(0x0B) <- Boiler(0x08), UBAParameterWW(0x33), data: 1B
ems-esp:# 000+03:14:09.217 N 5: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameterWW(0x33), data: 08 FF 30 FB FF 28 FF 07 3C D8 00
ems-esp:# 000+03:14:18.202 N 6: [emsesp] Boiler(0x08) -> All(0x00), UBAParameterWW(0x33), data: 08 FF 30 FB FF 28 FF 07 3C D8 00
ems-esp:# 000+03:14:18.772 N 7: [emsesp] Controller(0x09) -> Boiler(0x08), UBAParameterWW(0x33), data: 00 (offset 9)
ems-esp:# 000+03:14:29.284 N 8: [emsesp] Boiler(0x08) -> All(0x00), UBAParameterWW(0x33), data: 08 FF 30 FB FF 28 FF 07 3C 00 00
ems-esp:# call boiler wwcomfort intelligent
ems-esp:# 000+03:14:51.640 N 9: [emsesp] Me(0x0B) -> Boiler(0x08), UBAParameterWW(0x33), data: EC (offset 9)
ems-esp:# 000+03:14:51.919 N 10: [emsesp] Me(0x0B) <- Boiler(0x08), UBAParameterWW(0x33), data: 1B
ems-esp:# 000+03:14:51.947 N 11: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameterWW(0x33), data: 08 FF 30 FB FF 28 FF 07 3C EC 00
ems-esp:# 000+03:15:01.550 N 12: [emsesp] Me(0x0B) <- Boiler(0x08), UBAParameterWW(0x33), data: 1B
ems-esp:# 000+03:15:01.631 N 13: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameterWW(0x33), data: 08 FF 30 FB FF 28 FF 07 3C EC 00
ems-esp:# 000+03:15:01.733 N 14: [emsesp] Boiler(0x08) -> All(0x00), UBAParameterWW(0x33), data: 08 FF 30 FB FF 28 FF 07 3C EC 00
ems-esp:# 000+03:15:01.989 N 15: [emsesp] Controller(0x09) -> Boiler(0x08), UBAParameterWW(0x33), data: 00 (offset 9)
ems-esp:# 000+03:15:12.015 N 16: [emsesp] Boiler(0x08) -> All(0x00), UBAParameterWW(0x33), data: 08 FF 30 FB FF 28 FF 07 3C 00 00
```

pausenpepe, 2022-01-18:
@MichaelDvP 
Exactly the same on my side, only dhw temperature can be controlled via the bus. dhw comfort is set fix inside BC10  and not accessible from outside.  
Thanks for being more accurate than me :)

proddy, 2022-01-18:
maybe a good thing to add the wiki under https://emsesp.github.io/docs/#/Troubleshooting?id=general

MichaelDvP, 2022-01-19:
@proddy You have already added this on [14.11.2021](https://emsesp.github.io/docs/#/Troubleshooting?id=changing-a-value-on-an-ems-device-doesn39t-work) and [15.12.2021](https://emsesp.github.io/docs/#/Troubleshooting?id=changing-a-value-works-at-first-but-is-then-reset-to-it39s-original-value).

proddy, 2022-01-19:
yes, but expanding with more examples like setting the comfort.

On a side note, we have a lot of BC10's in the device_library.h which are probably not really BC10 controllers (the external chunky dial units).  For example, my Nefit Trendline HRC30 (combi boiler) has a built in controller with productID 190 which shows up as a BC10. I think we need to rename them, but to what? @bbqkees ideas too?