emsesp/EMS-ESP32 #300: Moduline 300 'crashes' when in Manual mode

joepvanboekel, 2022-01-17:
**Bug description**
Moduline 300, does not show set temperature from EMS-ESP or vice versa under some circumstances

**Steps to reproduce**
step 1: set hc1 mode from auto to manual
step 2: thermostat can set 'hc1 selected room temperature' and is synced to ems-esp
<repeat step 2 as many times as you want, it stays ok>
step 3: ems-esp can set 'hc1 selected room temperature' and is NOT synced to thermostat
step 4: thermostat can set 'hc1 selected room temperature' and is NOT synced to ems-esp, also Boiler does not respond to set temperature from thermostat anymore

you can reset this behaviour by settings thermostat physically to AUTO and to MANUAL back, until you perform step3 again.

**Expected behavior**
Thermostat should always reflect / and or set the correct 'hc1 selected room temperature'

**Device information**
{
  "System": {
    "version": "3.3.1b0",
    "uptime": "004+12:58:13.002",
    "uptime (seconds)": 392293,
    "freemem": 164,
    "reset reason": "Software reset CPU / Software reset CPU",
    "Dallas sensors": 0
  },
  "Network": {
    "connection": "Ethernet",
    "hostname": "ems-esp",
    "MAC": "11:22:33:44:55:66",
    "IPv4 address": "192.168.39.184/255.255.255.0",
    "IPv4 gateway": "192.168.39.1",
    "IPv4 nameserver": "192.168.39.1"
  },
  "Status": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "telegrams received": 187482,
    "read requests sent": 24,
    "write requests sent": 23,
    "incomplete telegrams": 1,
    "tx fails": 2,
    "rx line quality": 100,
    "tx line quality": 92,
    "MQTT": "connected",
    "MQTT publishes": 59258,
    "MQTT publish fails": 0,
    "Dallas reads": 0,
    "Dallas fails": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Nefit GBx72/Trendline/Cerapur/Greenstar Si/27i (DeviceID:0x08 ProductID:123, Version:06.08)",
      "handlers": "0x10 0x11 0xC2 0x14 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "Thermostat",
      "name": "RC20/Moduline 300 (DeviceID:0x17, ProductID:77, Version:03.03)",
      "handlers": "0xA3 0x06 0xA2 0x12 0x91 0xA8"
    },
    {
      "type": "Controller",
      "name": "BC10 (DeviceID:0x09, ProductID:190, Version:01.04)"
    }
  ]
}

MichaelDvP, 2022-01-18:
Does set temperature in auto-mode work?

For this thermostat there are only a few parameters known. 
Monitor message in 0x91, 15 bytes long and we know only seltemp in offset 1 and roomtemp in offset 2.
Set message in 0xA8, min 28 byte long, we read only mode from offset 23 and set the seltemp in offset 28. Maybe offset 28 is only for auto-mode and manual temp is stored in a different offset.

Please start a log with loglevel `ALL` and change the temperature on the thermostat in manual mode. Post the log here.

proddy, 2022-01-18:
I have an RC20/Moduline 300
 - set to Auto from the WebUI
 - set the selected room temperature
 - it is set correctly, also visible on the thermostat

I can't reproduce it @joepvanboekel . We'll need to the logs like Michael mentioned

MichaelDvP, 2022-01-18:
@proddy have you tried in manual mode? I understand it, that setting temperature on emsesp does notwork in manual mode, that's why ask if it works in automode.

proddy, 2022-01-18:
sorry, was a typo above. I set to manual (or heat as its in HA) and it worked. I know it works because its the only thermostat I have in my house for the last 10 years and what I've been using since EMS-ESP version 0.9 :-)

joepvanboekel, 2022-01-18:
log(1)
RC20 change to MANUAL
RC20 change settemp to 15
EMS-ESP shows 'hc1 selected room temperature' changed to 15

log(2)
EMSESP change 'hc1 selected room temperature' to 20
RC20 does not show 20 degrees on screen
RC20 change settemp from 15 to 24 degrees
EMSESP does not seem to receive updated 24degree setting

[log (1).txt](https://github.com/emsesp/EMS-ESP32/files/7891932/log.1.txt)
[log (2).txt](https://github.com/emsesp/EMS-ESP32/files/7891930/log.2.txt)

proddy, 2022-01-18:
I stand corrected. @joepvanboekel is right, when switching to manual mode on the RC20 unit and changing the setpoint temperature using the dial, the setpoint isn't been updated in EMS-ESP. I'll need to trace the telegrams and see what is happening.

MichaelDvP, 2022-01-19:
Looks like setting temperature in manual mode is in offset 29 and temperature for automode in offset28. But the auto-setpoint should be set to 0 to make manual setpoint active. The log does not contain the mode switching, but i think the offset 28 is cleared by mode-switch on thermostat. In this case switching the mode by emsesp should also not work (keep the auto-temperature).
From log 1: mode is manual and temp 19.5, auto-temp 0: 
`Thermostat(0x17) -> All(0x00), RC20Set(0xA8), data: 00 00 27 02 (offset 27)`
thermostat changes manual temperature: 
`Thermostat(0x17) -> All(0x00), RC20Set(0xA8), data: 1E (offset 29)`

Log 2: emsesp changes temperture:
`Me(0x0B) -> Thermostat(0x17), RC20Set(0xA8), data: 28 (offset 28)`
now auto-temp and manual temp have values and monitor reports  the auto-temp back:
`Thermostat(0x17) -> All(0x00), RC20Set(0xA8), data: 00 28 1E 02 (offset 27)`
`Thermostat(0x17) -> All(0x00), RC20Monitor(0x91), data: 90 28 00 CD 1A 00 00 00 00 05 07 00 CD 00`
changing manual temp on thermostat only writes to offset 29 and offset 28 keeps active:
`Thermostat(0x17) -> All(0x00), RC20Set(0xA8), data: 30 (offset 29)` 
`Thermostat(0x17) -> All(0x00), RC20Monitor(0x91), data: 90 28 00 CD 1A 00 00 00 00 05 07 00 CE 00`

MichaelDvP, 2022-01-19:
I think we need a log when switching on the thermostat from auto to manual and back to auto.
Clearing offset 28 when switching to manual should be no problem, but from where we get the auto-temperature when switching  from manal to auto?

joepvanboekel, 2022-01-20:
auto > manual > change temp to 23 > auto
[log (3).txt](https://github.com/emsesp/EMS-ESP32/files/7908074/log.3.txt)

auto > manual > auto
[log (4).txt](https://github.com/emsesp/EMS-ESP32/files/7908127/log.4.txt)

proddy, 2022-01-20:
@joepvanboekel thanks for testing. The easiest way is to use the telnet Console and type `watch 17` and then go to the Thermostat, manually setting it to Manual mode using the button and then dialing in another temperature using the up/down buttons. Remember to mark in the log when this was done. You can just hit enter on the Console to do a newline break. Then copy & paste the log to here

MichaelDvP, 2022-01-21:
Yes, watch 17 makes a easier to read. The compile with debug flag produces a lot of extra messages that we dont need in this case. Also the weblog buffer overflows when set to 100 and show the long [DEBUG] messages, so we are missing 10 messages, with (murphy) the messge i'm searching for.

@joepvanboekel Please repeat these steps with `loglevel info` and `watch 17`. And wait to get the full messages0x91 and 0xA8 before changing a parameter.

What i see in your log 3: (telegram -> comment)
`2022-01-20 20:47:03.142 T 774665: [emsesp] Thermostat(0x17) -> All(0x00), RC20Monitor(0x91), data: 80 23 00 BB 1A 00 FF FF 00 05 07 00 BB 00`
-> Setpoint 17,5 °C, i guess that was mode auto, We need 0xA8 to be sure.
`2022-01-20 20:47:04.911 T undefined: [undefined] undefined`
->These are the missing messages, the change to mode manual (i guess)
`2022-01-20 20:47:18.713 T 774689: [emsesp] Thermostat(0x17) -> All(0x00), RC20Set(0xA8), data: 01 00 FF F6 01 06 00 01 0D 01 00 FF FF 01 02 02 02 00 00 05 1E 05 1E 01 1C 00 FF`
`2022-01-20 20:47:19.006 T 774691: [emsesp] Thermostat(0x17) -> All(0x00), RC20Set(0xA8), data: 00 00 30 02 (offset 27)`
-> Now we are in mode manual (offset 23 set to 1) and setpoint auto is cleared (offset 28: 0) and setpoint manual 24°C (offset 29: 30). 
`2022-01-20 20:47:45.611 T 774784: [emsesp] Thermostat(0x17) -> All(0x00), RC20Set(0xA8), data: 2E (offset 29)`
-> Setpoint manual changed to 23°C
`2022-01-20 20:48:03.408 T 774820: [emsesp] Thermostat(0x17) -> All(0x00), RC20Monitor(0x91), data: 80 2E 00 BB 1A 00 00 00 64 5A 09 00 BB 00`
-> and reported in monitor Telegram (offset 1: 2E)
 `2022-01-20 20:48:12.492 T 774838: [emsesp] Thermostat(0x17) -> All(0x00), RC20Set(0xA8), data: 02 (offset 23)`
-> switch to auto mode
`2022-01-20 20:49:03.640 T 774935: [emsesp] Thermostat(0x17) -> All(0x00), RC20Monitor(0x91), data: 80 2A 00 BB 1A 00 00 00 64 5A 04 00 BC 00`
-> Now setpoint is reported as 21°C (offset 1: 2A)

But the full telegrams 0xA8 are missing between the changes, I don't see what happend to offset 28 and 29.

The actual v3.4 have a change to set manual temperature from emsesp, does this work now?

joepvanboekel, 2022-01-21:
ems-esp:$ set to manual
008+13:14:37.205 N 33: [emsesp] Thermostat(0x17) -> All(0x00), RC20Set(0xA8), data: 01 00 FF F6 01 06 00 01 0D 01 00 FF FF 01 02 02 02 00 00 05 1E 05 1E 01 1C 00 FF
008+13:14:37.512 N 34: [emsesp] Thermostat(0x17) -> All(0x00), RC20Set(0xA8), data: 00 00 2E 02 (offset 27)
008+13:14:38.874 N 35: [emsesp] Thermostat(0x17) -> Boiler(0x08), UBASetPoints(0x1A), data: 5A 64 00 00
008+13:15:07.267 N 36: [emsesp] Thermostat(0x17) -> All(0x00), RC20Monitor(0x91), data: 80 2E 00 C9 1A 00 00 01 64 5A 09 00 C9 00

ems-esp:$ set temp to 18
008+13:15:14.137 N 37: [emsesp] Thermostat(0x17) -> All(0x00), RCTime(0x06), data: 16 01 14 15 1A 31 04 00 00 00 00 00 00
008+13:15:16.361 N 38: [emsesp] Thermostat(0x17) -> All(0x00), RC20Set(0xA8), data: 24 (offset 29)
008+13:15:17.235 N 39: [emsesp] Thermostat(0x17) -> All(0x00), RCOutdoorTemp(0xA3), data: 00 00 00
008+13:15:18.604 N 40: [emsesp] Thermostat(0x17) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00 00 00
008+13:15:28.723 N 41: [emsesp] Thermostat(0x17) -> Boiler(0x08), UBAFlags(0x35), data: 11 00
008+13:16:00.379 N 42: [emsesp] Thermostat(0x17) -> All(0x00), RCError(0xA2), data: 00 00 00 00 00 00 00 00 00 00 00 00 00

ems-esp:$ set to manual
008+13:16:07.630 N 43: [emsesp] Thermostat(0x17) -> All(0x00), RC20Monitor(0x91), data: 80 24 00 C9 1A 00 00 01 00 5A 09 00 CA 00
008+13:16:14.511 N 44: [emsesp] Thermostat(0x17) -> All(0x00), RCTime(0x06), data: 16 01 14 15 1B 31 04 00 00 00 00 00 00
008+13:16:17.481 N 45: [emsesp] Thermostat(0x17) -> All(0x00), RCOutdoorTemp(0xA3), data: 00 00 00
008+13:16:18.861 N 46: [emsesp] Thermostat(0x17) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00 00 00
008+13:16:28.843 N 47: [emsesp] Thermostat(0x17) -> Boiler(0x08), UBAFlags(0x35), data: 11 00
008+13:17:08.132 N 48: [emsesp] Thermostat(0x17) -> All(0x00), RC20Monitor(0x91), data: 80 24 00 CA 1A 00 00 00 00 5A 09 00 CB 00
008+13:17:14.752 N 49: [emsesp] Thermostat(0x17) -> All(0x00), RCTime(0x06), data: 16 01 14 15 1C 32 04 00 00 00 00 00 00
008+13:17:17.975 N 50: [emsesp] Thermostat(0x17) -> All(0x00), RCOutdoorTemp(0xA3), data: 00 00 00
008+13:17:19.230 N 51: [emsesp] Thermostat(0x17) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00 00 00

is this useable ?

joepvanboekel, 2022-01-21:
ems-esp:$ set to auto
Too many arguments for command
008+13:19:43.620 N 63: [emsesp] Thermostat(0x17) -> All(0x00), RC20Set(0xA8), data: 02 (offset 23)
008+13:19:46.621 N 64: [emsesp] Thermostat(0x17) -> Boiler(0x08), UBASetPoints(0x1A), data: 5A 23 00 00
008+13:20:00.880 N 65: [emsesp] Thermostat(0x17) -> All(0x00), RCError(0xA2), data: 00 00 00 00 00 00 00 00 00 00 00 00 00
008+13:20:09.268 N 66: [emsesp] Thermostat(0x17) -> All(0x00), RC20Monitor(0x91), data: 80 2A 00 CB 1A 00 00 04 24 5A 04 00 CB 00
008+13:20:16.254 N 67: [emsesp] Thermostat(0x17) -> All(0x00), RCTime(0x06), data: 16 01 14 15 1F 33 04 00 00 00 00 00 00
008+13:20:18.979 N 68: [emsesp] Thermostat(0x17) -> All(0x00), RCOutdoorTemp(0xA3), data: 00 00 00
008+13:20:30.220 N 69: [emsesp] Thermostat(0x17) -> Boiler(0x08), UBAFlags(0x35), data: 11 00
008+13:20:46.847 N 70: [emsesp] Thermostat(0x17) -> Boiler(0x08), UBASetPoints(0x1A), data: 5A 24 00 00
ems-esp:$ set to manual
Too many arguments for command
008+13:21:07.119 N 71: [emsesp] Thermostat(0x17) -> All(0x00), RC20Set(0xA8), data: 01 (offset 23)
008+13:21:08.499 N 72: [emsesp] Thermostat(0x17) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00 00 00
008+13:21:09.779 N 73: [emsesp] Thermostat(0x17) -> All(0x00), RC20Monitor(0x91), data: 80 24 00 CB 1A 00 00 04 00 5A 09 00 CB 00
008+13:21:16.640 N 74: [emsesp] Thermostat(0x17) -> All(0x00), RCTime(0x06), data: 16 01 14 15 20 33 04 00 00 00 00 00 00
008+13:21:19.223 N 75: [emsesp] Thermostat(0x17) -> All(0x00), RCOutdoorTemp(0xA3), data: 00 00 00
008+13:21:30.354 N 76: [emsesp] Thermostat(0x17) -> Boiler(0x08), UBAFlags(0x35), data: 11 00
ems-esp:$ set temp to 20
Too many arguments for command
008+13:21:36.351 N 77: [emsesp] Thermostat(0x17) -> All(0x00), RC20Set(0xA8), data: 28 (offset 29)
008+13:22:08.727 N 78: [emsesp] Thermostat(0x17) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00 00 00
008+13:22:10.136 N 79: [emsesp] Thermostat(0x17) -> All(0x00), RC20Monitor(0x91), data: 80 28 00 CB 1A 00 00 03 00 5A 09 00 CC 00
008+13:22:17.124 N 80: [emsesp] Thermostat(0x17) -> All(0x00), RCTime(0x06), data: 16 01 14 15 21 34 04 00 00 00 00 00 00
008+13:22:19.724 N 81: [emsesp] Thermostat(0x17) -> All(0x00), RCOutdoorTemp(0xA3), data: 00 00 00
008+13:22:30.476 N 82: [emsesp] Thermostat(0x17) -> Boiler(0x08), UBAFlags(0x35), data: 11 00
ems-esp:$ set to auto
Too many arguments for command
008+13:22:36.119 N 83: [emsesp] Thermostat(0x17) -> All(0x00), RC20Set(0xA8), data: 02 (offset 23)
008+13:22:38.504 N 84: [emsesp] Thermostat(0x17) -> Boiler(0x08), UBASetPoints(0x1A), data: 5A 1F 00 00
008+13:23:10.757 N 85: [emsesp] Thermostat(0x17) -> All(0x00), RC20Monitor(0x91), data: 80 2A 00 CC 1A 00 00 02 1C 5A 04 00 CD 00
008+13:23:17.389 N 86: [emsesp] Thermostat(0x17) -> All(0x00), RCTime(0x06), data: 16 01 14 15 22 34 04 00 00 00 00 00 00
008+13:23:19.973 N 87: [emsesp] Thermostat(0x17) -> All(0x00), RCOutdoorTemp(0xA3), data: 00 00 00

can you work with this data ?

proddy, 2022-01-21:
perfect!

MichaelDvP, 2022-01-22:
> can you work with this data ?

Sorry, not really, I expected to see a complete 0xA8 telegram between each change, but it seems these telegrams are sended very rarly (i guess every 10 minutes), we have to read them by command. 
But the change-messages are as expected and the v3.4 changes should work.
> where do I get this (beta?) firmware?

It's the v3.4 branch from here ( to compile yourself), Proddy publishes bin-files on discord announcements, and here is a fresh compiled bin:
[EMS-ESP-3_4_0a17-ESP32.zip](https://github.com/emsesp/EMS-ESP32/files/7917923/EMS-ESP-3_4_0a17-ESP32.zip)

If it does not work, or if you have time to satisfy my curiosity do in terminal (no watch needed, wait for response before sending next command):
```
read 17 A8
read 17 91
call thermostat mode auto
read 17 A8
read 17 91
call thermostat seltemp 22
read 17 A8
read 17 91
call thermostat mode manual
read 17 A8
read 17 91
call thermostat seltemp 23
read 17 A8
read 17 91
call thermostat mode auto
read 17 A8
read 17 91
```
If the software does not work make the same with changes on thermostat instead of call command.

joepvanboekel, 2022-01-22:
new firmware does not seem to have wanted effect, the call command is not working on telnet?
so I performed the commands from the physical thermostat for you:

> ems-esp:$ read 17 A8
000+00:08:34.401 I 11: [emsesp] Thermostat(0x17) -> Me(0x0B), RC20Set(0xA8), data: 01 00 FF F6 01 06 00 01 0D 01 00 FF FF 01 02 02 02 00 00 05 1E 05 1E 00 1C 00 FF
000+00:08:35.709 I 12: [emsesp] Thermostat(0x17) -> Me(0x0B), RC20Set(0xA8), data: 00 00 26 02 (offset 27)
ems-esp:$ read 17 91
000+00:08:40.099 I 13: [emsesp] Thermostat(0x17) -> Me(0x0B), RC20Monitor(0x91), data: 90 1C 00 D4 1A 00 00 00 00 5A 08 00 D5 00
ems-esp:$ thermostat set to auto
Command not found
ems-esp:$ read 17 A8
000+00:09:07.776 I 14: [emsesp] Thermostat(0x17) -> Me(0x0B), RC20Set(0xA8), data: 01 00 FF F6 01 06 00 01 0D 01 00 FF FF 01 02 02 02 00 00 05 1E 05 1E 02 1C 00 FF
000+00:09:08.210 I 15: [emsesp] Thermostat(0x17) -> Me(0x0B), RC20Set(0xA8), data: 00 00 26 02 (offset 27)
ems-esp:$ read 17 91
000+00:09:12.727 I 16: [emsesp] Thermostat(0x17) -> Me(0x0B), RC20Monitor(0x91), data: 90 2A 00 D5 1A 00 00 05 00 5A 04 00 D6 00
ems-esp:$ thermostat set temp to 22
Command not found
ems-esp:$ read 17 A8
000+00:09:59.026 I 17: [emsesp] Thermostat(0x17) -> Me(0x0B), RC20Set(0xA8), data: 01 00 FF F6 01 06 00 01 0D 01 00 FF FF 01 02 02 02 00 00 05 1E 05 1E 02 1C 00 FF
000+00:09:59.458 I 18: [emsesp] Thermostat(0x17) -> Me(0x0B), RC20Set(0xA8), data: 00 2C 26 02 (offset 27)
ems-esp:$ read 17 91
000+00:10:02.230 I 19: [emsesp] Thermostat(0x17) -> Me(0x0B), RC20Monitor(0x91), data: 80 2C 00 D5 1A 00 00 05 1D 5A 07 00 D6 00
ems-esp:$ thermostat set to manual
Command not found
000+00:10:18.047 W 20: [telegram] Incomplete Rx: 09 F0 09 F8 F8
ems-esp:$ read 17 A8
000+00:10:32.158 I 21: [emsesp] Thermostat(0x17) -> Me(0x0B), RC20Set(0xA8), data: 01 00 FF F6 01 06 00 01 0D 01 00 FF FF 01 02 02 02 00 00 05 1E 05 1E 01 1C 00 FF
000+00:10:32.708 I 22: [emsesp] Thermostat(0x17) -> Me(0x0B), RC20Set(0xA8), data: 00 00 26 02 (offset 27)
ems-esp:$ read 17 91
000+00:10:37.593 I 23: [emsesp] Thermostat(0x17) -> Me(0x0B), RC20Monitor(0x91), data: 80 26 00 D6 1A 00 00 08 00 5A 09 00 D7 00
ems-esp:$ thermostat set temp to 23
Command not found
000+00:10:57.779 W 24: [telegram] Incomplete Rx: FE FE FF
ems-esp:$ read 17 A8
000+00:11:15.281 I 25: [emsesp] Thermostat(0x17) -> Me(0x0B), RC20Set(0xA8), data: 01 00 FF F6 01 06 00 01 0D 01 00 FF FF 01 02 02 02 00 00 05 1E 05 1E 01 1C 00 FF
000+00:11:15.851 I 26: [emsesp] Thermostat(0x17) -> Me(0x0B), RC20Set(0xA8), data: 00 00 2E 02 (offset 27)
ems-esp:$ read 17 91
000+00:11:18.482 I 27: [emsesp] Thermostat(0x17) -> Me(0x0B), RC20Monitor(0x91), data: 80 2E 00 D7 1A 00 00 0C 64 5A 09 00 D8 00
ems-esp:$ thermostat set to auto
Command not found
ems-esp:$ read 17 A8
000+00:11:50.282 I 28: [emsesp] Thermostat(0x17) -> Me(0x0B), RC20Set(0xA8), data: 01 00 FF F6 01 06 00 01 0D 01 00 FF FF 01 02 02 02 00 00 05 1E 05 1E 02 1C 00 FF
000+00:11:50.836 I 29: [emsesp] Thermostat(0x17) -> Me(0x0B), RC20Set(0xA8), data: 00 00 2E 02 (offset 27)
ems-esp:$ read 17 91
000+00:11:53.101 I 30: [emsesp] Thermostat(0x17) -> Me(0x0B), RC20Monitor(0x91), data: 80 2A 00 D8 1A 00 00 0D 00 5A 04 00 D9 00

MichaelDvP, 2022-01-23:
> new firmware does not seem to have wanted effect, the call command is not working on telnet?

As written in manual you need to be superuser, command `su` password as set in web-security (default: ems-esp-neo).
Does setting temp now work from web?

Some more analysis: 
- It starts with mode off and setpoint 1C in 0x91, I see 1C in offset 24 of 0xA8, maybe this can be written
- 0xA8 offset 28 is `tempautotemp` and goes to 0 after each mode change, the auto temperatures are stored elsewhere (unknown telegram)
- 0xA8 offset 29 is the manual temp, this should work now.
- 0x91 offset 10 could be modetype in automode 4 (day-high) or 7(temporary changed), manualmode 9, mode-off 8. should be checked for automode off/night/...
- 0x91 offset 8 is a percent value and goes up if temperature setpoint is rised. Maybe the caculated setpoint for boiler burnpower

I think this is all i can analyse without having this thermostat.

MichaelDvP, 2022-01-26:
@joepvanboekel Have you tried with the [new build](https://github.com/emsesp/EMS-ESP32/releases). I've made a mistake in first implementation and think now it should work.

I've searched the manual of moduline 300 and see that there are 4 auto-temperature levels: night(T1), day-low(T2), day-middle(T3), day-high(T4). Can make a log in terminal with `watch 17` and change these temperatures on the thermostat. You have to change each temperature, not only showw it, but you can change it and in next step change it back.
The log will have a new telegram `Thermostat(0x17)->All(0x00) ?(0x..)`  Please type in terminal a read for this telegram `read 17 ..` where the `..` is the number from above `?(0x..)`. Thanks.

randommen96, 2022-02-10:
Hey Michael,

I just tested the latest build as I ran into the same issue, as mentioned in the Discord it is working now and changes are reflected both ways in manual mode.

I didn't try the other modes, if needed I can test them too, to confirm there are no breaking changes.

Do you still want the log you described to Joep?

Kind regards,
David

randommen96, 2022-02-11:
> Thanks.
>
> Do you still want the log you described to Joep?
>
> Not for this bug, but if you are interrested in changing the different
> (auto) T1-T4 levels via emsesp, we can add these parameters if can fetch
> them in a log.

I will try to do so this weekend and report back with the log.

randommen96, 2022-02-11:
Hi Michael,

I tried your steps, changing T1 - T4 back and forth on the thermostat while watching / reading the terminal.
However most of them give back the same value 90, but sometimes 8F and A2 (with RCError).

Can you work with this?

`T1
000+00:15:26.222 N 170: [emsesp] Thermostat(0x17) -> All(0x00), ?(0x90), data: 26 (offset 3)
000+00:15:36.471 N 171: [emsesp] Thermostat(0x17) -> All(0x00), ?(0x90), data: 24 (offset 3)

ems-esp:$ read 17 90
000+00:18:54.125 N 210: [emsesp] Me(0x0B) <- Thermostat(0x17), ?(0x90), data: 20
000+00:18:54.284 I 211: [emsesp] Thermostat(0x17) -> Me(0x0B), ?(0x90), data: 01 01 02 24 24 28 28 1E 0E 00 01 5A 32 05 4B 2D 00 28 00 3C FF 11 00 05

T2
000+00:15:44.996 N 172: [emsesp] Thermostat(0x17) -> All(0x00), RCError(0xA2), data: 00 00 00 00 00 00 00 00 00 00 00 00 00
000+00:15:46.846 N 173: [emsesp] Thermostat(0x17) -> All(0x00), ?(0x90), data: 2A (offset 4)

ems-esp:$ read 17 A2
000+00:18:33.008 N 208: [emsesp] Me(0x0B) <- Thermostat(0x17), RCError(0xA2), data: 20
000+00:18:33.129 I 209: [emsesp] Thermostat(0x17) -> Me(0x0B), RCError(0xA2), data: 00 00 00 00 00 00 00 00 00 00 00 00 00

T3
000+00:15:57.096 N 174: [emsesp] Thermostat(0x17) -> All(0x00), ?(0x90), data: 24 (offset 4)
000+00:16:00.101 N 175: [emsesp] Thermostat(0x17) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00 00 00
000+00:16:01.556 N 176: [emsesp] Me(0x0B) <- Thermostat(0x17), RC20Set(0xA8), data: 20
000+00:16:01.667 N 177: [emsesp] Thermostat(0x17) -> Me(0x0B), RC20Set(0xA8), data: 01 00 FF F6 01 06 00 01 0D 03 00 FF FF 01 02 02 02 00 00 05 24 05 24 01 0E 00 00
000+00:16:01.748 N 178: [emsesp] Thermostat(0x17) -> All(0x00), RC20Monitor(0x91), data: 80 24 00 C8 00 00 00 03 00 05 09 00 C8 00
000+00:16:01.996 N 179: [emsesp] Thermostat(0x17) -> All(0x00), RCTime(0x06), data: 16 02 17 0B 0B 32 04 00 00 00 00 00 00
000+00:16:02.383 N 180: [emsesp] Me(0x0B) <- Thermostat(0x17), RC20Set(0xA8), data: 20 (offset 27)
000+00:16:02.478 N 181: [emsesp] Thermostat(0x17) -> Me(0x0B), RC20Set(0xA8), data: 00 00 24 (offset 27)
000+00:16:03.723 N 182: [emsesp] Thermostat(0x17) -> All(0x00), RCOutdoorTemp(0xA3), data: 00 00 00
000+00:16:03.971 N 183: [emsesp] Thermostat(0x17) -> Boiler(0x08), UBAFlags(0x35), data: 01 00

000+00:16:07.347 N 184: [emsesp] Thermostat(0x17) -> All(0x00), ?(0x90), data: 2B (offset 5)

ems-esp:$ read 17 90
000+00:18:54.125 N 210: [emsesp] Me(0x0B) <- Thermostat(0x17), ?(0x90), data: 20
000+00:18:54.284 I 211: [emsesp] Thermostat(0x17) -> Me(0x0B), ?(0x90), data: 01 01 02 24 24 28 28 1E 0E 00 01 5A 32 05 4B 2D 00 28 00 3C FF 11 00 05

T4
000+00:16:17.597 N 185: [emsesp] Thermostat(0x17) -> All(0x00), ?(0x90), data: 28 (offset 5)
000+00:16:27.847 N 186: [emsesp] Thermostat(0x17) -> All(0x00), ?(0x90), data: 2D (offset 6)
000+00:16:38.097 N 187: [emsesp] Thermostat(0x17) -> All(0x00), ?(0x90), data: 28 (offset 6)

ems-esp:$ read 17 90
000+00:18:54.125 N 210: [emsesp] Me(0x0B) <- Thermostat(0x17), ?(0x90), data: 20
000+00:18:54.284 I 211: [emsesp] Thermostat(0x17) -> Me(0x0B), ?(0x90), data: 01 01 02 24 24 28 28 1E 0E 00 01 5A 32 05 4B 2D 00 28 00 3C FF 11 00 05
`

I also got back 8F one time, but i'm unsure which T<number> it was.
`
000+00:05:35.395 N 48: [emsesp] Thermostat(0x17) -> All(0x00), ?(0x8F), data: 04 5D 01 78 24 5D 21 6E 43 5D 41 78 64 5D 61 78 84 5D 81 78 E7 90 E7 90 E7 90 E7
000+00:05:35.774 N 49: [emsesp] Thermostat(0x17) -> All(0x00), ?(0x8F), data: 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 (offset 27)
000+00:05:36.274 N 50: [emsesp] Thermostat(0x17) -> All(0x00), ?(0x8F), data: E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 (offset 54)
000+00:05:36.626 N 51: [emsesp] Thermostat(0x17) -> All(0x00), ?(0x8F), data: 90 E7 90 01 00 00 01 01 00 01 01 00 01 01 00 01 01 00 00 (offset 81)

ems-esp:$ read 17 8F
000+00:21:30.933 N 241: [emsesp] Me(0x0B) <- Thermostat(0x17), ?(0x8F), data: 20
000+00:21:31.045 I 242: [emsesp] Thermostat(0x17) -> Me(0x0B), ?(0x8F), data: 04 5D 01 78 24 5D 21 6E 43 5D 41 78 64 5D 61 78 84 5D 81 78 E7 90 E7 90 E7 90 E7
000+00:21:31.262 N 243: [emsesp] Me(0x0B) <- Thermostat(0x17), ?(0x8F), data: 20 (offset 27)
000+00:21:31.428 I 244: [emsesp] Thermostat(0x17) -> Me(0x0B), ?(0x8F), data: 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 (offset 27)
000+00:21:31.681 N 245: [emsesp] Me(0x0B) <- Thermostat(0x17), ?(0x8F), data: 20 (offset 54)
000+00:21:31.799 I 246: [emsesp] Thermostat(0x17) -> Me(0x0B), ?(0x8F), data: E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 90 E7 (offset 54)
000+00:21:32.006 N 247: [emsesp] Me(0x0B) <- Thermostat(0x17), ?(0x8F), data: 20 (offset 81)
000+00:21:32.155 I 248: [emsesp] Thermostat(0x17) -> Me(0x0B), ?(0x8F), data: 90 E7 90 01 00 00 01 01 00 01 01 00 01 01 00 01 01 00 00 (offset 81)
`

Kind regards,
David

randommen96, 2022-02-12:
> Thanks, so we have the T1-T4 in 0x90, offset 3-6. this telegram seems to
> hold some more settings and temperatures.
> The 0x8F is the time schedule of the program. always 2 bytes for one
> entry: first byte upper 3 bits for dayofweek, lower 3 bits for temperatures
> T1-T4, second byte the time in steps of 10 minutes. Interresting the bytes
> offset 84 to 99, in RC35 these are settings for pause, party, holiday and
> vacation modes. Does your thermostat have theses settings?

Not in standalone mode, but it looks like it does when the RC20 is used as
a remote for the RC35, from the manual: "If holiday mode is  active  for a
heating  circuit when  using an  RC35 (RC20 as  remote control),  the
holiday  setting  can be  adjusted  on the RC20, but not the temporary
room  setting."

commit d3ca5569 (2022-01-25): fix #300 moduline 300 seltemp, next try
closed 2023-02-19 by proddy