emsesp/EMS-ESP32 #301: Add Olimex ESP32-POE-ISO to board profiles

NoNeedForAName123, 2022-01-17:
Could you please add the Olimex ESP32-POE-ISO board to the profiles list.
Unfortunately the ethernet module needs other settings for ETH_CLK_MOD & ETH_PHY_POWER
More information:  https://github.com/OLIMEX/ESP32-POE/issues/16#issuecomment-707502051

It's needs:
power  = 12;
...
clock_mode = ETH_CLOCK_GPIO17_OUT;

Thx

bbqkees, 2022-01-18:
So is this correct then and did you test it yourself?

phy_addr   = 0;
power      = 12;
mdc        = 23;
mdio       = 18;
type       = ETH_PHY_LAN8720;
clock_mode = ETH_CLOCK_GPIO17_OUT;

proddy, 2022-01-18:
yes this should work for the ESP32-POE. I'm thinking whether to add it as a new Board Profile or just extend the CUSTOM section to all the clock mode, power and address to be changed

proddy, 2022-01-18:
added to v3.4 (EMS-ESP-3_4_0a16-ESP32.bin)

NoNeedForAName123, 2022-01-19:
Just as Information I also tried a [LilyGO T-ETH-POE](https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-POE) board. It works with the same ethernet configuration. 

Thx for your support!

closed 2022-01-19 by proddy