emsesp/EMS-ESP32 #303: Add additional configurable MQTT json key:value to support influxdb database selection

hansij66, 2022-01-18:
**Is your feature request related to a problem? Please describe.**
No - this is a feature request

I am using EMS-ESP in conjunction with Telegraf and influxdb to store selected data from EMS-ESP.
I have multiple MQTT sources (eg, weather, lights, smart meter,...), and I would like to store their data in specific influxdb databases (unique per MQTT source)

Snippet from telegraf configuration:
```
[[outputs.influxdb]]
  database = "telegraf" 
  database_tag = "database"
```

Telegraf selects (in above config example) the influxdb database based on the MQTT json key:value "database". For MQTT sources under my control, this works, I add eg a database:weather key:value.
For MQTT sources (like EMS-ESP, all data ends up in one influxdb database "telegraf"
I have not been able to find other ways in telegraf to specify a influxdb per [[inputs.mqtt_consumer]]

I do realize that this is scope creep beyond the original intend of EMS-ESP, and understand if it is not implemented.

Cheers,
Hans

proddy, 2022-01-18:
Hi Hans, that's pretty hardcore! A few users have Home Assistant feeding the EMS-ESP data directly into InfluxDB and then using Grafana or Chronograf to see visually or run queries against the timeline. It's something that can setup in minutes. Haven't heard of anyone using Telegraf though. One method could be to create a script to poll the data from the REST APIs using the Telegraf HTTP plugin and avoiding MQTT all together? 

On your specific request, how would it work exactly? Like where would this key:value json object be part of? A separate payload?

hansij66, 2022-01-18:
Hi Paul,

 One of my other MQTT clients produces this:

{"V1_sags":5,"V1_swells":0,"V2_sags":5,"V2_swells":0,"V3_sags":4,"V3_swells":0,"database":"dsmr","long_power_failures":12,"power_failures":1373,"timestamp":1642538642}

the "database":"dsmr" is used by telegraf to put the data in dsmr database
in influxdb.

But if this is a niche usecase, I will look for other options...you don't
want to pollute your great app :)
Cheers, Hans

hansij66, 2022-12-31:
I found the proper way to configure this in telegraf.conf

closed 2022-12-31 by hansij66