emsesp/EMS-ESP32 #306: Changing the central heating max power on the boiler

proddy, 2022-01-19:
This one has been bugging me for a while.

I have a Nefit HRC30/CW5 combi boiler (does both DHW and heating) which is rated at 30kW max power. On the front-panel you can adjust the central heating power (the default is 22.5 kW). I don't need so much power for my house (3 floors, 13 radiators) so I always lower it to save gas. Anyway, I've been trying to find how to adjust the power via telegrams but with no luck.

What I've observed:

The kW value seems to be part of telegram's 0x18 (offset 10) and 0x34 (offset 2) as a single byte (* 2).

But when I manually change the power to say 26.4 kW (hex 0x0108) it starts getting weird with 0x18 now having a value of 0xBF and 0x18 a value of 0xBE.

I'll keep investigating, just wanted to share for now

MichaelDvP, 2022-01-19:
Hmm, 18 and 34 are monitor telegrams and not writeable. I think the max. heater power is in telegram 0x16 offset 2 as percent value (burnmaxpower). My (unmodulated, oil) boiler is fixed to 100% and not changeable.

proddy, 2022-01-19:
genius! Indeed, I remember seeing some old videos of other Nefit boilers where you only adjust the burn power % and it will show you the kW on the display window. Mine was set at 75%. I changed it to 40% and the kW dropped to 12 kW so that's how to change it. Case closed - thanks!!

closed 2022-01-19 by proddy