emsesp/EMS-ESP32 #309: Show Tx Reads and Writes to avoid confusion

proddy, 2022-01-20:
There was a comment in Discord to why there are so many Tx telegrams sent which may cause panic thinking EMS-ESP is changing things on the Boiler constantly. It's obviously because of the scheduled fetch as some data is not in any of the broadcasted telegrams. 

I was thinking of some small changes:

- In the Status page show both Tx Read and Writes. Maybe call it "Telegrams Read request (Tx)" and "Telegrams Write commands (Tx)"
- Add to the Console command `show devices` to also show which Telegram ID's are included in the scheduled fetch

@MichaelDvP what do you think?

MichaelDvP, 2022-01-20:
Yes, in `api/system/info` and `show ems` it is already included, makes sense to add it also to status page.
In `show devices` the ids could be colored green:broadcasted, yellow: fetched, red: not received

bbqkees, 2022-01-20:
I can understand the confusion for a lot of people.
Sometimes you have to 'write' something to 'read' something which only makes sense if you understand the low level stuff.

Although it will improve it somewhat, I think if you show the TX reads and writes separately it will not make it more clear immediately for the non-tech users.

proddy, 2022-01-21:
I suggest just: 

- EMS Reads (Tx)
- EMS Writes (Tx)

Also I noticed you added back the Tx-Mode. I don't think it has a purpose here. It's static and a duplicate of the Tx Mode setting. Also most users don't understand or care about it. Suggest we remove it.

MichaelDvP, 2022-01-21:
> Also I noticed you added back the Tx-Mode.

You have removed the call, but not the (now unused) function, i thought you've forgot to add the call back (like it happend to temp sensors first). It is redundant, but if people realize tx-issues on this page, it is a hint what can be changed.

proddy, 2022-01-21:
fair point. Alternative solutions
 1) changing DashboardStatus.tsx::busStatus() to show 'Tx Errors - check TxMode setting" when there are stability errors.
 2) or showing a <MessageBox/> at the bottom of the page that pops up if Rx & Tx is 0 or the quality < 80%. Otherwise its hidden

MichaelDvP, 2022-01-21:
I vote for 1.

proddy, 2022-01-21:
me too. Then call it something less scary like "Tx instability - check your Tx-Mode setting"

do you want to include this with your other changes? Then I'll merge into dev

MichaelDvP, 2022-01-21:
Ok, i have to add to console too, 
`instability` is not right if the device does not react on a telegram. What about "Tx Errrors - try a different Tx-Mode.
But you can change the texts later. Also in info there is a mix of capitals and lower case.

proddy, 2022-01-21:
we can change the text later. I just though the word 'error' was a bit harsh and may scare people. Specially after paying 80 euros for a gateway :)

proddy, 2022-01-22:
its included and works nicely

closed 2022-01-22 by proddy