emsesp/EMS-ESP32 #311: (v3.4) Solar module entity not showing in HA

proddy, 2022-01-21:
@PhillyGilly reported in HA the main config for the solar entity is not created.

Can you check using MQTTExplorer and search for 'solar'. You should one topic like

`ems-esp/homeassistant/sensor/ems-esp/solar_collectortemp/config payload={"~":"ems-esp","uniq_id":"solar_collectortemp","stat_t":"~/solar_data","name":"Solar collector temperature (TS1)","val_tpl":"{{value_json.collectortemp}}","unit_of_meas":"°C","state_class":"measurement","device_class":"temperature","dev":{"ids":["ems-esp-solar"],"name":"EMS-ESP Solar","mf":"","mdl":"SM100/MS100","via_device":"ems-esp"}}`

pausenpepe, 2022-01-22:
@proddy 
MQTTExplorer is showing the following:

`homeip
homeassistant
sensor
ems-esp
solar
config = {"uniq_id":"solar","ic":"mdi:home-automation","stat_t":"ems-esp/solar_data","name":"Solar EMS Product ID","val_tpl":"{{value_json.id}}","dev":{"name":"EMS-ESP Solar","sw":"3.3.1b1","mf":"","mdl":"SM10","ids":["ems-esp-solar"]}}
solar_collectortemp
config = {"~":"ems-esp","uniq_id":"solar_collectortemp","stat_t":"~/solar_data","name":"Solar collector temperature (TS1)","val_tpl":"{{value_json.collectortemp}}","unit_of_meas":"°C","state_class":"measurement","device_class":"temperature","dev":{"ids":["ems-esp-solar"],"name":"EMS-ESP Solar","mf":"","mdl":"SM10","via_device":"ems-esp"}}
solar_tankbottomtemp
config = {"~":"ems-esp","uniq_id":"solar_tankbottomtemp","stat_t":"~/solar_data","name":"Solar tank bottom temperature (TS2)","val_tpl":"{{value_json.tankbottomtemp}}","unit_of_meas":"°C","ic":"mdi:coolant-temperature","device_class":"temperature","dev":{"ids":["ems-esp-solar"]}}
solar_pumpworktime
config = {"~":"ems-esp","uniq_id":"solar_pumpworktime","stat_t":"~/solar_data","name":"Solar pump working time","val_tpl":"{{value_json.pumpworktime}}","unit_of_meas":"minutes","ic":"mdi:clock-outline","state_class":"total_increasing","dev":{"ids":["ems-esp-solar"]}}
solar_tankmaxtemp
config = {"~":"ems-esp","uniq_id":"solar_tankmaxtemp","stat_t":"~/solar_data","name":"Solar maximum tank temperature","val_tpl":"{{value_json.tankmaxtemp}}","unit_of_meas":"°C","ic":"mdi:coolant-temperature","dev":{"ids":["ems-esp-solar"]}}
solar_solarpumpmodulation
config = {"~":"ems-esp","uniq_id":"solar_solarpumpmodulation","stat_t":"~/solar_data","name":"Solar pump modulation (PS1)","val_tpl":"{{value_json.solarpumpmodulation}}","unit_of_meas":"%","ic":"mdi:percent-outline","device_class":"power_factor","dev":{"ids":["ems-esp-solar"]}}
solar_pumpminmod
config = {"~":"ems-esp","uniq_id":"solar_pumpminmod","stat_t":"~/solar_data","name":"Solar minimum pump modulation","val_tpl":"{{value_json.pumpminmod}}","unit_of_meas":"%","ic":"mdi:percent-outline","dev":{"ids":["ems-esp-solar"]}}
solar_turnondiff
config = {"~":"ems-esp","uniq_id":"solar_turnondiff","stat_t":"~/solar_data","name":"Solar pump turn on difference","val_tpl":"{{value_json.turnondiff}}","unit_of_meas":"°C","ic":"mdi:coolant-temperature","dev":{"ids":["ems-esp-solar"]}}
solar_turnoffdiff
config = {"~":"ems-esp","uniq_id":"solar_turnoffdiff","stat_t":"~/solar_data","name":"Solar pump turn off difference","val_tpl":"{{value_json.turnoffdiff}}","unit_of_meas":"°C","ic":"mdi:coolant-temperature","dev":{"ids":["ems-esp-solar"]}}
solar_solarpower
config = {"~":"ems-esp","uniq_id":"solar_solarpower","stat_t":"~/solar_data","name":"Solar actual solar power","val_tpl":"{{value_json.solarpower}}","unit_of_meas":"W","state_class":"measurement","device_class":"power","dev":{"ids":["ems-esp-solar"]}}
solar_energylasthour
config = {"~":"ems-esp","uniq_id":"solar_energylasthour","stat_t":"~/solar_data","name":"Solar energy last hour","val_tpl":"{{value_json.energylasthour}}","unit_of_meas":"Wh","state_class":"measurement","device_class":"energy","dev":{"ids":["ems-esp-solar"]}}
solar_maxflow
config = {"~":"ems-esp","uniq_id":"solar_maxflow","stat_t":"~/solar_data","name":"Solar maximum solar flow","val_tpl":"{{value_json.maxflow}}","unit_of_meas":"l/min","ic":"mdi:water-boiler","dev":{"ids":["ems-esp-solar"]}}
solar_wwmintemp
config = {"~":"ems-esp","uniq_id":"solar_wwmintemp","stat_t":"~/solar_data","name":"Solar ww minimum temperature","val_tpl":"{{value_json.wwmintemp}}","unit_of_meas":"°C","ic":"mdi:coolant-temperature","dev":{"ids":["ems-esp-solar"]}}
solar_cylbottomtemp
config = {"~":"ems-esp","uniq_id":"solar_cylbottomtemp","stat_t":"~/solar_data","name":"Solar cylinder bottom temperature (TS2)","val_tpl":"{{value_json.cylbottomtemp}}","unit_of_meas":"°C","state_class":"measurement","device_class":"temperature","dev":{"ids":["ems-esp-solar"]}}
solar_solarpumpmod
config = {"~":"ems-esp","uniq_id":"solar_solarpumpmod","stat_t":"~/solar_data","name":"Solar pump modulation (PS1)","val_tpl":"{{value_json.solarpumpmod}}","unit_of_meas":"%","state_class":"measurement","device_class":"power_factor","dev":{"ids":["ems-esp-solar"]}}
solar_data11
config = {"~":"ems-esp","uniq_id":"solar_data11","stat_t":"~/solar_data","name":"Solar unknown datafield 11","val_tpl":"{{value_json.data11}}","ic":"mdi:counter","state_class":"measurement","dev":{"ids":["ems-esp-solar"]}}
solar_data12
config = {"~":"ems-esp","uniq_id":"solar_data12","stat_t":"~/solar_data","name":"Solar unknown datafield 12","val_tpl":"{{value_json.data12}}","ic":"mdi:counter","state_class":"measurement","dev":{"ids":["ems-esp-solar"]}}
solar_data1
config = {"~":"ems-esp","uniq_id":"solar_data1","stat_t":"~/solar_data","name":"Solar unknown datafield 1","val_tpl":"{{value_json.data1}}","ic":"mdi:counter","state_class":"measurement","dev":{"ids":["ems-esp-solar"]}}
solar_data0
config = {"~":"ems-esp","uniq_id":"solar_data0","stat_t":"~/solar_data","name":"Solar unknown datafield 0","val_tpl":"{{value_json.data0}}","ic":"mdi:counter","state_class":"measurement","dev":{"ids":["ems-esp-solar"]}}
binary_sensor
ems-esp
solar_solarpump
config = {"~":"ems-esp","uniq_id":"solar_solarpump","stat_t":"~/solar_data","name":"Solar pump (PS1)","val_tpl":"{{value_json.solarpump}}","payload_on":"on","payload_off":"off","state_class":"measurement","dev":{"ids":["ems-esp-solar"]}}
solar_collectorshutdown
[... cut ...]
Template variable warning: 'dict object' has no attribute 'tankmaxtemp' when rendering '{{value_json.tankmaxtemp}}'
Template variable warning: 'dict object' has no attribute 'solarpumpmodulation' when rendering '{{value_json.solarpumpmodulation}}'
Template variable warning: 'dict object' has no attribute 'tankheated' when rendering '{{value_json.tankheated}}'
Template variable warning: 'dict object' has no attribute 'activated' when rendering '{{value_json.activated}}'`

I hope this will help to nail it down.

MichaelDvP, 2022-01-22:
these are renamed values from v3.3, read: [https://github.com/emsesp/EMS-ESP32/issues?q=dict+object+in%3Abody+](https://github.com/emsesp/EMS-ESP32/issues?q=dict+object+in%3Abody+)

proddy, 2022-01-22:
that's right. Those names are no longer used, for example `tankbottomtemp` is now `cylbottomtemp` and you don't have that value anyway. Follow the guide on https://emsesp.github.io/docs/#/Troubleshooting?id=home-assistant and try again

PhillyGilly, 2022-01-22:
@proddy The devices and sensors tab on the EMS-ESP web interface has miraculously reappeared over-night. It's very neat.
However in MQTT explorer I am not getting exactly the message you described. 
I have got homeassistant/sensor/ems-esp/solar_collectortemp/config with payload:
{"~":"ems-esp","uniq_id":"solar_collectortemp","stat_t":"~/solar_data","name":"Solar collector temperature (TS1)","val_tpl":"{{value_json.collectortemp}}","unit_of_meas":"°C","state_class":"measurement","device_class":"temperature","dev":{"ids":["ems-esp-solar"]}}

MichaelDvP, 2022-01-22:
I dont use HA, and can't say if it is working, but if enabled it in emsesp i get in mqtt for SM10 module (after removing all old HA publishes):
```
{
  "~": "ems",
  "uniq_id": "solar_collectortemp",
  "stat_t": "~/solar_data",
  "name": "Solar collector temperature (TS1)",
  "val_tpl": "{{value_json.collectortemp}}",
  "unit_of_meas": "°C",
  "state_class": "measurement",
  "device_class": "temperature",
  "dev": {
    "ids": [
      "ems-esp-solar"
    ],
    "name": "EMS-ESP Solar",
    "mf": "",
    "mdl": "SM10",
    "via_device": "ems-esp"
  }
}
```

proddy, 2022-01-22:
yes it works in my offline tests too

proddy, 2022-01-22:
so I think it just needed a refresh of the WebUI and a cleanup of the old MQTT topics

PhillyGilly, 2022-01-22:
All good now. Several attempts at clearing my MQTT of all solar related topics combined with restarting my MQTT service and restarting EMS-ESP got me there. I can see that it was the lack of id info in the payload which meant the device part wasn't working. Maybe it was my migration path that caused the problem. I went from EMS-ESP-3_3_0-ESP32 to EMS-ESP-3_4_0a16-ESP32. Or maybe my ESP was a bit "dozy" as it wasn't displaying the devices yesterday. Many thanks.

pausenpepe, 2022-01-23:
> so I think it just needed a refresh of the WebUI and a cleanup of the old MQTT topics

Thanks now it is working there are no further warning related to EMS-ESP ;)
I am still a newbie to EMS-ESP and HA