emsesp/EMS-ESP32 #314: min/max incorrect in HA for Number entities

proddy, 2022-01-22:
As reported in Discord, in v3.4 the max values are not set to 100 for percentages. It's one line to fix but thought its better to separate out the min/max calculation into a separate class function which I did. Needs careful reviewing though.

proddy, 2022-01-22:
@MichaelDvP question :  is it possible for a % to be actually higher than 100? For example, the `selburnpower` having a value of say 115%? 

```json
http://ems-esp.local/api/boiler/selburnpow
{
  "name": "selburnpow",
  "fullname": "burner selected max power",
  "value": 32,
  "type": "number",
  "min": 0,
  "max": 100,
  "uom": "%",
  "writeable": true,
  "visible": true
}
```

MichaelDvP, 2022-01-22:
I dont think so, my boiler is fixed100% and not changeable. 
On the other hand heating manufactors also calculating efficiency to more than 100%, which is phyicaly impossible.
We have the optional min/max fields for each values. I think setting percent default to 0-100(UINT) and -100-100(INT) and set individual limits to values that are not standard. I think percent is always INT/UINT, but sometimes scaled with F_(mul5).

proddy, 2022-01-22:
I pushed a change - should be a PR (sorry), will get around to doing that later

MichaelDvP, 2022-01-22:
Commenting a commit or a PR is the same for me.

projectile25, 2022-01-31:
Hi, 

@proddy  - for me it always occurs when hot water is heated , v3.4.0b1 .
```
2022-01-31 12:59:39 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.boiler_burner_selected_max_power: 123 (range 0.0 - 100.0)
2022-01-31 12:59:49 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.boiler_burner_selected_max_power: 123 (range 0.0 - 100.0)
2022-01-31 12:59:59 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.boiler_burner_selected_max_power: 123 (range 0.0 - 100.0)
2022-01-31 13:00:09 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.boiler_burner_selected_max_power: 123 (range 0.0 - 100.0)
2022-01-31 13:00:19 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.boiler_burner_selected_max_power: 123 (range 0.0 - 100.0)
2022-01-31 13:00:29 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.boiler_burner_selected_max_power: 123 (range 0.0 - 100.0)
2022-01-31 13:00:39 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.boiler_burner_selected_max_power: 123 (range 0.0 - 100.0)
2022-01-31 13:00:49 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.boiler_burner_selected_max_power: 123 (range 0.0 - 100.0)
2022-01-31 13:00:59 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.boiler_burner_selected_max_power: 123 (range 0.0 - 100.0)
2022-01-31 13:01:09 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.boiler_burner_selected_max_power: 123 (range 0.0 - 100.0)
2022-01-31 13:01:19 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.boiler_burner_selected_max_power: 123 (range 0.0 - 100.0)
2022-01-31 13:01:29 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.boiler_burner_selected_max_power: 123 (range 0.0 - 100.0)
2022-01-31 13:01:39 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.boiler_burner_selected_max_power: 123 (range 0.0 - 100.0)
2022-01-31 13:01:49 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.boiler_burner_selected_max_power: 123 (range 0.0 - 100.0)
2022-01-31 13:01:59 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.boiler_burner_selected_max_power: 123 (range 0.0 - 100.0)
2022-01-31 13:02:09 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.boiler_burner_selected_max_power: 123 (range 0.0 - 100.0)
2022-01-31 13:02:19 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.boiler_burner_selected_max_power: 123 (range 0.0 - 100.0)
2022-01-31 13:02:29 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.boiler_burner_selected_max_power: 123 (range 0.0 - 100.0)
2022-01-31 13:02:39 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.boiler_burner_selected_max_power: 123 (range 0.0 - 100.0)
2022-01-31 13:02:49 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.boiler_burner_selected_max_power: 123 (range 0.0 - 100.0)
2022-01-31 13:02:59 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.boiler_burner_selected_max_power: 123 (range 0.0 - 100.0)
2022-01-31 13:03:09 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.boiler_burner_selected_max_power: 123 (range 0.0 - 100.0)
2022-01-31 13:03:19 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.boiler_burner_selected_max_power: 123 (range 0.0 - 100.0)
2022-01-31 13:03:29 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.boiler_burner_selected_max_power: 123 (range 0.0 - 100.0)
2022-01-31 13:03:39 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.boiler_burner_selected_max_power: 123 (range 0.0 - 100.0)
2022-01-31 13:03:49 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.boiler_burner_selected_max_power: 123 (range 0.0 - 100.0)
2022-01-31 13:03:59 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.boiler_burner_selected_max_power: 123 (range 0.0 - 100.0)
2022-01-31 13:04:09 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.boiler_burner_selected_max_power: 123 (range 0.0 - 100.0)
2022-01-31 13:04:19 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.boiler_burner_selected_max_power: 123 (range 0.0 - 100.0)
```

proddy, 2022-02-05:
Ok so we need to explicitly set the min and max for the Boiler's `selburnpow` to fix this.

proddy, 2022-02-12:
re-opening. The change hasn't been made yet.

commit 718f9a4f (2022-02-12): fixes #314
closed 2022-02-12 by proddy