emsesp/EMS-ESP32 #315: CLIMATE entity not being configured on Home Assistant discovery

guardianbs, 2022-01-22:
**Bug description**
ems-esp is only sending HA discovery config data for 6 thermostat SENSORS. It is not sending HA discovery config data for a CLIMATE entity. .

**Steps to reproduce**
Ran HA discovery.. Examined result.

**Expected behavior**
A CLIMATE entity discovery config should be published.

**Screenshots**
[image]

**Device information**
{
  "System": {
    "version": "3.4.0a16",
    "uptime": "001+04:16:27.506",
    "uptime (seconds)": 101787,
    "freemem": 120,
    "reset reason": "Software reset CPU / Software reset CPU",
    "Temperature sensors": 0
  },
  "Status": {
    "bus status": "connected",
    "bus protocol": "HT3",
    "telegrams received": 47863,
    "read requests sent": 5157,
    "write requests sent": 47,
    "incomplete telegrams": 3,
    "tx fails": 1,
    "rx line quality": 100,
    "tx line quality": 100,
    "MQTT": "connected",
    "MQTT publishes": 35557,
    "MQTT publish fails": 0,
    "Temperature sensors": 0,
    "Temperature sensor reads": 0,
    "Temperature sensor fails": 0,
    "Analog sensors": 0,
    "API calls": 3,
    "API fails": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3 (DeviceID:0x08, ProductID:95, Version:12.10)",
      "handlers": "0x10 0x11 0xC2 0x14 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A"
    },
    {
      "type": "Thermostat",
      "name": "Worcester CRF200S (DeviceID:0x18, ProductID:216, Version:01.03)",
      "handlers": "0xA3 0x06 0xA2 0x12 0x2A5 0x2A6 0x2A7 0x2A8"
    },
    {
      "type": "Controller",
      "name": "HT3 (DeviceID:0x09, ProductID:95, Version:12.10)"
    }
  ]
}

**Additional context**
*Add any other context about the problem here.*

proddy, 2022-01-23:
The HA climate entity will only be created when there is a valid current temperature, setpoint temperature and mode. These are the HVAC requirements within HA. Can you use MQTTExplorer and post here the values of the `ems-esp/thermostat_data` topic plus anything under `homeassistant/climate`

and of course your settings

proddy, 2022-01-23:
see also https://github.com/emsesp/EMS-ESP32/issues/299

guardianbs, 2022-01-23:
MQTT output

[image]
Not sure what you mean by 'your settings'

No homeassistant/climate entry.

The Climate entity used to be present before updating from 3.3 to 3.4.

proddy, 2022-01-23:
Settings as in http://ems-esp.local/api/system/settings (and mentioned in https://emsesp.github.io/docs/#/Support?id=support. I wanted to see if you had ha enabled and the prefix matches, but from the screenshot above it looks ok. 

It's strange the climate is missing. I can't reproduce or find anything in the code that would cause it. You could turn log level to ADD which will show you all the MQTT publishes and ctrl-F search for 'climate'. Or even easier if you have a syslog service running.

guardianbs, 2022-01-23:
Not sure how set log level to ADD but log level ALL for a few minutes produced the following.
The only topics published are

esp/boiler_data
esp/boiler_data_ww
esp/heartbeat
esp/thermostat_data

No mention of climate

ems-esp:$ log ADD
Invalid log level
ems-esp:$ log all
Log level: all
001+22:01:20.042 D 0: [mqtt] Publishing topic ems-esp/boiler_data (#56912, retain=0, retry=1, size=616, pid=1)
001+22:01:20.143 D 1: [mqtt] Publishing topic ems-esp/boiler_data_ww (#56913, retain=0, retry=1, size=484, pid=1)
001+22:01:20.244 D 2: [mqtt] Publishing topic ems-esp/thermostat_data (#56914, retain=0, retry=1, size=121, pid=1)
001+22:01:20.647 T 3: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 01 AD 00 00 80 02 00 C0 01 1D 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
001+22:01:20.861 T 4: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 36 01 1D 80 00 80 00 00 01 00 01 5E 32 01 4B 35 00
001+22:01:30.045 D 5: [mqtt] Publishing topic ems-esp/boiler_data (#56915, retain=0, retry=1, size=616, pid=1)
001+22:01:30.146 D 6: [mqtt] Publishing topic ems-esp/boiler_data_ww (#56916, retain=0, retry=1, size=484, pid=1)
001+22:01:30.247 D 7: [mqtt] Publishing topic ems-esp/thermostat_data (#56917, retain=0, retry=1, size=121, pid=1)
001+22:01:30.568 T 8: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 01 AD 00 00 80 02 00 C0 01 1D 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
001+22:01:30.809 T 9: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 36 01 1C 80 00 80 00 00 01 00 01 5E 32 01 4B 35 00
001+22:01:40.047 D 10: [mqtt] Publishing topic ems-esp/boiler_data (#56918, retain=0, retry=1, size=616, pid=1)
001+22:01:40.148 D 11: [mqtt] Publishing topic ems-esp/boiler_data_ww (#56919, retain=0, retry=1, size=484, pid=1)
001+22:01:40.249 D 12: [mqtt] Publishing topic ems-esp/thermostat_data (#56920, retain=0, retry=1, size=121, pid=1)
001+22:01:40.616 T 13: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 01 AD 00 00 80 02 00 C0 01 1C 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
001+22:01:40.830 T 14: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 36 01 1C 80 00 80 00 00 01 00 01 5E 32 01 4B 35 00
001+22:01:45.485 T 15: [emsesp] Thermostat(0x18) -> All(0x00), RCTime(0x06), data: 16 01 0C 17 2E 00 06 00
001+22:01:45.684 T 16: [emsesp] Thermostat(0x18) <- Boiler(0x08), Version(0x02), data: 0A
001+22:01:45.716 T 17: [emsesp] Boiler(0x08) -> Thermostat(0x18), Version(0x02), data: 5F 0C 0A 00 00 00 00 00 00 00
001+22:01:45.716 D 18: [emsesp] Updating details for already active deviceID 0x08
001+22:01:45.753 T 19: [emsesp] Thermostat(0x18) -> All(0x00), CRFMonitor(0x02A5), data: 00 B5 20 00 00 00 14
001+22:01:45.945 T 20: [emsesp] Thermostat(0x18) -> Boiler(0x08), UBAFlags(0x35), data: 01
001+22:01:45.945 D 21: [emsesp] Process setting 0x35 for device 0x08
001+22:01:45.967 T 22: [emsesp] Thermostat(0x18) -> Boiler(0x08), UBASetPoints(0x1A), data: 00 00
001+22:01:45.967 D 23: [emsesp] Process setting 0x1A for device 0x08
001+22:01:50.066 D 24: [mqtt] Publishing topic ems-esp/boiler_data (#56921, retain=0, retry=1, size=616, pid=1)
001+22:01:50.167 D 25: [mqtt] Publishing topic ems-esp/boiler_data_ww (#56922, retain=0, retry=1, size=484, pid=1)
001+22:01:50.268 D 26: [mqtt] Publishing topic ems-esp/thermostat_data (#56923, retain=0, retry=1, size=121, pid=1)
001+22:01:50.600 T 27: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 01 AD 00 00 80 02 00 C0 01 1C 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
001+22:01:50.815 T 28: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 36 01 1C 80 00 80 00 00 01 00 01 5E 32 01 4B 35 00
001+22:02:00.068 D 29: [mqtt] Publishing topic ems-esp/boiler_data (#56924, retain=0, retry=1, size=616, pid=1)
001+22:02:00.169 D 30: [mqtt] Publishing topic ems-esp/boiler_data_ww (#56925, retain=0, retry=1, size=484, pid=1)
001+22:02:00.270 D 31: [mqtt] Publishing topic ems-esp/thermostat_data (#56926, retain=0, retry=1, size=121, pid=1)
001+22:02:00.523 T 32: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 00 01 AB 00 00 80 02 00 C0 01 1C 80 00 80 00 FF FF FF 00 00 00 00 00 00 00
001+22:02:00.762 T 33: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 36 01 1C 80 00 80 00 00 01 00 01 5E 32 01 4B 35 00
001+22:02:02.849 D 34: [emsesp] Fetching values for deviceID 0x08
001+22:02:02.849 D 35: [telegram] Tx read request to device 0x08 for type ID 0x14
001+22:02:02.849 D 36: [telegram] Tx read request to device 0x08 for type ID 0x16
001+22:02:02.849 D 37: [telegram] Tx read request to device 0x08 for type ID 0x33
001+22:02:02.849 D 38: [emsesp] Fetching values for deviceID 0x18
001+22:02:02.849 D 39: [emsesp] Fetching values for deviceID 0x09
001+22:02:02.861 D 40: [telegram] Sending read Tx [#236], telegram: 8B 88 14 00 20
001+22:02:02.861 T 41: [emsesp] Me(0x0B) <- Boiler(0x08), UBATotalUptime(0x14), data: 20
001+22:02:02.899 D 42: [emsesp] Last Tx read successful
001+22:02:02.899 T 43: [emsesp] Boiler(0x08) -> Me(0x0B), UBATotalUptime(0x14), data: 32 53 94
001+22:02:03.021 D 44: [telegram] Sending read Tx [#237], telegram: 8B 88 16 00 20
001+22:02:03.035 T 45: [emsesp] Me(0x0B) <- Boiler(0x08), UBAParameters(0x16), data: 20
001+22:02:03.066 D 46: [emsesp] Last Tx read successful
001+22:02:03.067 T 47: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameters(0x16), data: FF 43 54 00 00 F6 03 01 03 64 0A 04
[... cut ...]
    "master_thermostat": 0,
    "shower_timer": false,
    "shower_alert": false,
    "rx_gpio": 5,
    "tx_gpio": 17,
    "dallas_gpio": 4,
    "pbutton_gpio": 33,
    "led_gpio": 2,
    "hide_led": false,
    "notoken_api": false,
    "readonly_mode": false,
    "fahrenheit": false,
    "dallas_parasite": false,
    "bool_format": 1,
    "enum_format": 1,
    "analog_enabled": false,
    "telnet_enabled": true,
    "phy_type": 1,
    "eth_power": 16,
    "eth_phy_addr": 1,
    "eth_clock_mode": 0
  }
}

proddy, 2022-01-23:
I think the best way forward is if I create a special debug version for you with some extra trace information to help pinpoint why it's failing

guardianbs, 2022-01-23:
Sure, OK. Let me know when its ready.

guardianbs, 2022-01-24:
@proddy  Just loaded v3.4.0b0 and Climate entity HA configuration has returned. :grin:

closed 2022-01-24 by guardianbs

proddy, 2022-01-24:
wonderful! To be honest, I have no idea why. I refactored some of the code but this is still a mystery why it didn't work. If you had multiple thermostats or heating circuits then I could probably think of a reason or two, but you only had hc1