emsesp/EMS-ESP32 #316: EMS-ESP not updating Dashboard with Home Assistant value change

guardianbs, 2022-01-22:
**Bug description**
When a change is made to a Boiler value a mqtt message is published but ems-esp would appear not to see it and no change is made to the Boiler value on the ems-esp Dashboard.

**Steps to reproduce**
Made a change to boiler_burner_max_power on both the the Lovelace UI and directly on Device / Entity. In both case HA sent the message to mqtt showing up in HA logs and MQTT Explorer. No change to ems-esp Dashboard and no sign of mqtt receipt in ems-esp 'log all' output.

**Expected behavior**
ems-esp Dashboard field should be updated and value sent to Boiler.

**Screenshots**
[image]

**Device information**
{
  "System": {
    "version": "3.4.0a16",
    "uptime": "001+04:16:27.506",
    "uptime (seconds)": 101787,
    "freemem": 120,
    "reset reason": "Software reset CPU / Software reset CPU",
    "Temperature sensors": 0
  },
  "Status": {
    "bus status": "connected",
    "bus protocol": "HT3",
    "telegrams received": 47863,
    "read requests sent": 5157,
    "write requests sent": 47,
    "incomplete telegrams": 3,
    "tx fails": 1,
    "rx line quality": 100,
    "tx line quality": 100,
    "MQTT": "connected",
    "MQTT publishes": 35557,
    "MQTT publish fails": 0,
    "Temperature sensors": 0,
    "Temperature sensor reads": 0,
    "Temperature sensor fails": 0,
    "Analog sensors": 0,
    "API calls": 3,
    "API fails": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Condens 2500/Logamax/Logomatic/Cerapur Top/Greenstar/Generic HT3 (DeviceID:0x08, ProductID:95, Version:12.10)",
      "handlers": "0x10 0x11 0xC2 0x14 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A"
    },
    {
      "type": "Thermostat",
      "name": "Worcester CRF200S (DeviceID:0x18, ProductID:216, Version:01.03)",
      "handlers": "0xA3 0x06 0xA2 0x12 0x2A5 0x2A6 0x2A7 0x2A8"
    },
    {
      "type": "Controller",
      "name": "HT3 (DeviceID:0x09, ProductID:95, Version:12.10)"
    }
  ]
}

ems-esp:$ show mqtt
MQTT is connected
MQTT publish errors: 0

MQTT topic subscriptions:
 ems-esp/system/#
 ems-esp/boiler/#
 ems-esp/thermostat/#

MQTT queue is empty
[... 196 lines cut ...]
001+02:31:01.638 D 220: [emsesp] Fetching values for deviceID 0x08
001+02:31:01.638 D 221: [telegram] Tx read request to device 0x08 for type ID 0x14
001+02:31:01.638 D 222: [telegram] Tx read request to device 0x08 for type ID 0x16
001+02:31:01.638 D 223: [telegram] Tx read request to device 0x08 for type ID 0x33
001+02:31:01.638 D 224: [emsesp] Fetching values for deviceID 0x18
001+02:31:01.638 D 225: [emsesp] Fetching values for deviceID 0x09
001+02:31:01.763 D 226: [telegram] Sending read Tx [#32], telegram: 8B 88 14 00 20
001+02:31:01.776 T 227: [emsesp] Me(0x0B) <- Boiler(0x08), UBATotalUptime(0x14), data: 20
001+02:31:01.802 D 228: [emsesp] Last Tx read successful
001+02:31:01.802 T 229: [emsesp] Boiler(0x08) -> Me(0x0B), UBATotalUptime(0x14), data: 32 4F 00
001+02:31:01.925 D 230: [telegram] Sending read Tx [#33], telegram: 8B 88 16 00 20
001+02:31:01.926 T 231: [emsesp] Me(0x0B) <- Boiler(0x08), UBAParameters(0x16), data: 20
001+02:31:01.968 D 232: [emsesp] Last Tx read successful
001+02:31:01.969 T 233: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameters(0x16), data: FF 43 54 00 00 F6 03 01 03 64 0A 04
001+02:31:02.106 D 234: [telegram] Sending read Tx [#34], telegram: 8B 88 33 00 20
001+02:31:02.119 T 235: [emsesp] Me(0x0B) <- Boiler(0x08), UBAParameterWW(0x33), data: 20
001+02:31:02.156 D 236: [emsesp] Last Tx read successful
001+02:31:02.156 T 237: [emsesp] Boiler(0x08) -> Me(0x0B), UBAParameterWW(0x33), data: 08 FF 36 00 00 00 00 02 46 00 FF FF
001+02:31:02.888 D 238: [mqtt] Publishing topic ems-esp/heartbeat (#33267, retain=0, retry=1, size=231, pid=1)
001+02:31:05.122 T 239: [emsesp] Boiler(0x08) -> All(0x00), UBADevices(0x07), data: 0B 00 01 00 00 00 00 00 00 00 00 00 00 00 00
001+02:31:05.352 T 240: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorFast(0x18), data: 43 02 A5 54 25 89 23 25 C0 02 0D 80 00 80 00 FF FF FF 00 00 00 00 00 02 18
001+02:31:05.604 T 241: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorSlow(0x19), data: 80 00 80 00 80 00 FF FF 00 5F 07 D8 29 17 55 AC 00 00 00 15 F7 87 06 8C 87 80 00
001+02:31:05.842 T 242: [emsesp] Boiler(0x08) -> All(0x00), UBAMonitorWW(0x34), data: 36 02 0D 80 00 A1 00 00 01 00 01 5E 25 01 4B 31 00
ems-esp:$ log info
Log level: info

MichaelDvP, 2022-01-23:
The command is sent to the boiler and boiler acknowledged, but keeps setting 84%. Maybe this is the max. setting for this boiler, try with lower values (you have only tried higher). The limits are stored in the boiler and can differ between boiler types, but it's not known how to read them (if they are readable). Emsesp can not check for out-of-limit values, it only sends to the boiler and reads back the new (or unchanged) value and reports this.

proddy, 2022-01-23:
EMS-ESP is not an intelligent device, it just mirrors what other EMS devices do. If you can change the boiler max power manually /physical to 90% on one of your devices (e.g. front panel of the boiler) and it stays there and works, but can't EMS-ESP doesn't then it's a bug. Otherwise its https://emsesp.github.io/docs/#/Troubleshooting?id=changing-a-value-on-an-ems-device-doesn39t-work

guardianbs, 2022-01-23:
@proddy. Yes, looks like Boiler is keeping a fixed value, so not really 'writeable'. When #317 is implemented it shouldn't be a problem as this parameter can be marked as 'read-only'.

'EMS-ESP is not an intelligent device' !! Bit harsh :grin:

closed 2022-01-23 by guardianbs

MichaelDvP, 2022-01-23:
> Yes, looks like Boiler is keeping a fixed value, so not really 'writeable'.

I dont think so, It's a modulated boiler, isn't it? Than it's probably writable but only in a limited (safe) range, try other values.

guardianbs, 2022-01-23:
Hmm, yes. Values less than 84% work. Anything over 84% reset to 84%. 
Thanks.