emsesp/EMS-ESP32 #317: Enable selection of type of Home Assistant Entity created

guardianbs, 2022-01-22:
**Is your feature request related to a problem? Please describe.**
Not really a problem. More of an inconvenience. With version 3.4 the discovery config data sent to Home Assistant has been enhanced to set up Number / Switch / Enum entities for parameters that are 'writeable' . This is good as it saves coding in HA for parameters that are to be updated. Unfortunately some parameters that appear 'writeable' are not, in as much as if you change the value it immediately gets set back to its original value by the device. This means the HA Number / Switch / Enum entity is superfluous and misleading. 'writeable' parameters that cannot be updated would be better if they were set up as sensor entity types. This would save having to set up copy template sensor entities for each of the superfluous Number / Switch / Enum entities.

**Describe the solution you'd like**
Extract to HA 'writeable' parameters that are not 'writeable' (as defined by the user) as sensor entity types.

**Describe alternatives you've considered**
Two possibilities. Firstly,The main Dashboard screen could be amended to allow the deletion of the 'writeable' flag. parameters would then appear not 'writeable' and be extracted and set up on HA as sensor entity types. Secondly, as version 3.4 now has the nice feature in Settings / Customization to allow entities to be ignored altogether and not extracted to HA, this screen could be amended to allow parameters to be marked as 'sensor only'. Presumably the table on  Settings / Customization is being stored and parsed when generating the HA config entries so those marked 'sensor only' could be extracted accordingly.

**Additional context**
*Add any other context or screenshots about the feature request here.*

proddy, 2022-01-23:
I can see the need for this. Some people may accidentally move one of the sliders in HA and forget what it was originally set to, and freak out. I've done this a few times.

I suggest extending the Customization Page to show two checkboxes, one for Disabled (excluded) and one for Read Only.

guardianbs, 2022-01-23:
Sounds good :grin:

proddy, 2022-03-27:
added to b10

proddy, 2022-03-30:
In the latest implementation, there is still one last thing that needs to be fixed. When an entity is marked to be read-only it should remove the HA config entry (which is number/switch) and mark it to be re-created (as a sensor) so it has immediate effect. Right now you need to restart EMS-ESP.

proddy, 2022-04-03:
made some final minor fixes into b11

closed 2022-04-03 by proddy