emsesp/EMS-ESP32 #318: API and Dashboard stops showing data

pypb, 2022-01-23:
**Bug description**
The gateway stops reporting data through the API and on the Dashboard after a varying amount of time, usually hours. The API returns null for any call, and clicking on any of the devices in the Dashboard returns nothing. Inspecting the POST call to /rest/deviceData shows it returns null as well. Rebooting the gateway will usually help, on some occasions it takes multiple reboots (might be a time factor, rather than the reboot itself - however it does not seem to recover without a reboot).

MQTT and syslog still works and you can still view information through telnet.

E32 gateway with firmware version 3.3.1. 3.3.0 had the same problem.

**Steps to reproduce**
Unknown.

**Expected behavior**
The API and Dashboard should continue to work, or return a HTTP error code with a helpful message if something is wrong with the backend.

**Screenshots**
```
% curl http://ems-esp.local/api/system/info
null%                                                                                                                                                                                                                                                         
```

**Device information**
IVT AirX 90, IVT AirModule E9

```
  "Devices": [
    {
      "type": "Boiler",
      "name": "Enviline/Compress 6000AW/Hybrid 7000iAW/SupraEco/Geo 5xx (DeviceID:0x08, ProductID:172, Version:01.10)",
      "handlers": "0x10 0x11 0xC2 0x14 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x494 0x495 0x48D 0x48F 0x48A"
    },
    {
      "type": "Thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:13.05)",
      "handlers": "0xA3 0x06 0xA2 0x12 0x2A5 0x2B9 0x2AF 0x29B 0x471 0x2A6 0x2BA 0x2B0 0x29C 0x472 0x2A7 0x2BB 0x2B1 0x29D 0x473 0x2A8 0x2BC 0x2B2 0x29E 0x474 0x2F5 0x31B 0x31D 0x31E 0x23A 0x267 0x240"
    },
    {
      "type": "Gateway",
      "name": "KM200/MB LAN 2 (DeviceID:0x48, ProductID:189, Version:04.07)"
    }
  ]
```

**Additional context**
Logs from the minutes leading up to the point where the API started returning null.

```
2022-01-23T08:28:38.864499+01:00 ems-esp emsesp 000+00:00:00.000 I 0: Starting Syslog
2022-01-23T08:29:12.387967+01:00 ems-esp mqtt 000+00:00:07.211 W 91: Lost device value for boiler_hpbrinein. Removing HA config
2022-01-23T08:29:12.389671+01:00 ems-esp mqtt 000+00:00:07.211 W 92: Lost device value for boiler_hpbrineout. Removing HA config
2022-01-23T08:29:12.391314+01:00 ems-esp mqtt 000+00:00:07.211 W 93: Lost device value for boiler_hpsuctiongas. Removing HA config
2022-01-23T08:29:12.392855+01:00 ems-esp mqtt 000+00:00:07.211 W 94: Lost device value for boiler_hphotgas. Removing HA config
2022-01-23T08:29:12.394380+01:00 ems-esp mqtt 000+00:00:07.211 W 95: Lost device value for boiler_hptc0. Removing HA config
2022-01-23T08:29:12.395931+01:00 ems-esp mqtt 000+00:00:07.211 W 96: Lost device value for boiler_hptc1. Removing HA config
2022-01-23T08:29:12.397465+01:00 ems-esp mqtt 000+00:00:07.211 W 97: Lost device value for boiler_hptc3. Removing HA config
2022-01-23T08:29:12.398950+01:00 ems-esp mqtt 000+00:00:07.211 W 98: Lost device value for boiler_hptr3. Removing HA config
2022-01-23T08:29:12.400499+01:00 ems-esp mqtt 000+00:00:07.211 W 99: Lost device value for boiler_hptr4. Removing HA config
2022-01-23T08:29:12.402077+01:00 ems-esp mqtt 000+00:00:07.211 W 100: Lost device value for boiler_hptr5. Removing HA config
2022-01-23T08:29:12.403572+01:00 ems-esp mqtt 000+00:00:07.211 W 101: Lost device value for boiler_hptr6. Removing HA config
2022-01-23T08:29:12.405075+01:00 ems-esp mqtt 000+00:00:07.211 W 102: Lost device value for boiler_hptr7. Removing HA config
2022-01-23T08:29:12.406542+01:00 ems-esp mqtt 000+00:00:07.211 W 103: Lost device value for boiler_hptl2. Removing HA config
2022-01-23T08:29:12.408027+01:00 ems-esp mqtt 000+00:00:07.211 W 104: Lost device value for boiler_hppl1. Removing HA config
2022-01-23T08:29:12.409542+01:00 ems-esp mqtt 000+00:00:07.211 W 105: Lost device value for boiler_hpph1. Removing HA config
2022-01-23T08:29:12.411111+01:00 ems-esp mqtt 000+00:00:07.211 W 106: Lost device value for boiler_poolsettemp. Removing HA config
2022-01-23T08:29:12.412543+01:00 ems-esp mqtt 000+00:00:07.211 W 107: Lost device value for boiler_wwseltemp. Removing HA config
2022-01-23T08:29:12.413966+01:00 ems-esp mqtt 000+00:00:07.211 W 108: Lost device value for boiler_wwsettemp. Removing HA config
2022-01-23T08:29:12.415402+01:00 ems-esp mqtt 000+00:00:07.211 W 109: Lost device value for boiler_wwseltemplow. Removing HA config
2022-01-23T08:29:12.416971+01:00 ems-esp mqtt 000+00:00:07.211 W 110: Lost device value for boiler_wwseltempoff. Removing HA config
2022-01-23T08:29:12.418548+01:00 ems-esp mqtt 000+00:00:07.211 W 111: Lost device value for boiler_wwseltempsingle. Removing HA config
2022-01-23T08:29:12.420189+01:00 ems-esp mqtt 000+00:00:07.211 W 112: Lost device value for boiler_wwtype. Removing HA config
2022-01-23T08:29:12.421837+01:00 ems-esp mqtt 000+00:00:07.211 W 113: Lost device value for boiler_wwcomfort. Removing HA config
2022-01-23T08:29:12.423487+01:00 ems-esp mqtt 000+00:00:07.211 W 114: Lost device value for boiler_wwflowtempoffset. Removing HA config
2022-01-23T08:29:12.425050+01:00 ems-esp mqtt 000+00:00:07.211 W 115: Lost device value for boiler_wwmaxpower. Removing HA config
2022-01-23T08:29:12.426700+01:00 ems-esp mqtt 000+00:00:07.211 W 116: Lost device value for boiler_wwcircpump. Removing HA config
2022-01-23T08:29:12.428280+01:00 ems-esp mqtt 000+00:00:07.211 W 117: Lost device value for boiler_wwchargetype. Removing HA config
2022-01-23T08:29:12.429908+01:00 ems-esp mqtt 000+00:00:07.211 W 118: Lost device value for boiler_wwhyston. Removing HA config
2022-01-23T08:29:12.431569+01:00 ems-esp mqtt 000+00:00:07.211 W 119: Lost device value for boiler_wwhystoff. Removing HA config
2022-01-23T08:29:12.433282+01:00 ems-esp mqtt 000+00:00:07.211 W 120: Lost device value for boiler_wwdisinfectiontemp. Removing HA config
2022-01-23T08:29:12.434957+01:00 ems-esp mqtt 000+00:00:07.211 W 121: Lost device value for boiler_wwcircmode. Removing HA config
2022-01-23T08:29:12.436597+01:00 ems-esp mqtt 000+00:00:07.211 W 122: Lost device value for boiler_wwcirc. Removing HA config
2022-01-23T08:29:12.438229+01:00 ems-esp mqtt 000+00:00:07.211 W 123: Lost device value for boiler_wwcurtemp. Removing HA config
2022-01-23T08:29:12.439931+01:00 ems-esp mqtt 000+00:00:07.211 W 124: Lost device value for boiler_wwcurtemp2. Removing HA config
2022-01-23T08:29:12.441890+01:00 ems-esp mqtt 000+00:00:07.211 W 125: Lost device value for boiler_wwcurflow. Removing HA config
2022-01-23T08:29:12.444876+01:00 ems-esp mqtt 000+00:00:07.395 W 126: Lost device value for boiler_wwstoragetemp1. Removing HA config
2022-01-23T08:29:12.446709+01:00 ems-esp mqtt 000+00:00:07.395 W 127: Lost device value for boiler_wwstoragetemp2. Removing HA config
2022-01-23T08:29:12.448396+01:00 ems-esp mqtt 000+00:00:07.395 W 128: Lost device value for boiler_wwactivated. Removing HA config
2022-01-23T08:29:12.450077+01:00 ems-esp mqtt 000+00:00:07.395 W 129: Lost device value for boiler_wwonetime. Removing HA config
2022-01-23T08:29:12.451764+01:00 ems-esp mqtt 000+00:00:07.395 W 130: Lost device value for boiler_wwdisinfect. Removing HA config
2022-01-23T08:29:12.453416+01:00 ems-esp mqtt 000+00:00:07.395 W 131: Lost device value for boiler_wwcharging. Removing HA config
2022-01-23T08:29:12.455076+01:00 ems-esp mqtt 000+00:00:07.395 W 132: Lost device value for boiler_wwrecharging. Removing HA config
2022-01-23T08:29:12.456786+01:00 ems-esp mqtt 000+00:00:07.395 W 133: Lost device value for boiler_wwtempok. Removing HA config
2022-01-23T08:29:12.458476+01:00 ems-esp mqtt 000+00:00:07.395 W 134: Lost device value for boiler_wwactive. Removing HA config
2022-01-23T08:29:12.460196+01:00 ems-esp mqtt 000+00:00:07.395 W 135: Lost device value for boiler_wwheat. Removing HA config
2022-01-23T08:29:12.461952+01:00 ems-esp mqtt 000+00:00:07.395 W 136: Lost device value for boiler_wwsetpumppower. Removing HA config
2022-01-23T08:29:12.463727+01:00 ems-esp mqtt 000+00:00:07.395 W 137: Lost device value for boiler_wwmixertemp. Removing HA config
2022-01-23T08:29:12.465448+01:00 ems-esp mqtt 000+00:00:07.395 W 138: Lost device value for boiler_wwtankmiddletemp. Removing HA config
2022-01-23T08:29:12.467180+01:00 ems-esp mqtt 000+00:00:07.395 W 139: Lost device value for boiler_wwstarts. Removing HA config
2022-01-23T08:29:12.468978+01:00 ems-esp mqtt 000+00:00:07.395 W 140: Lost device value for boiler_wwworkm. Removing HA config
2022-01-23T08:29:15.048192+01:00 ems-esp emsesp 000+00:00:10.000 I 141: Stopping Access Point
2022-01-23T08:29:21.099320+01:00 ems-esp emsesp 000+00:00:16.033 I 142: Recognized new thermostat with device ID 0x10
```

proddy, 2022-01-23:
never seen anything like this before. If MQTT and Console are working then it seems like the Web server is down. We'll need more details on the version, system settings etc. I expect something else is interfering with HTTP traffic, either firewall, IPv6 routing, DNS mixup, port blocking etc.

pypb, 2022-01-23:
HTTP server is still up. I can still navigate the web GUI when this happens and I can still restart the gateway through the Restart button under Settings. As far as I can tell, everything is still working except for the API and the Device information on the Dashboard.

Nothing is "interfering" with the HTTP traffic. curl would not return a payload (containing null) if there was a network connection error, it would exit with the appropriate error message (like connection refused etc.).

/api/system/info:

```
{
  "System": {
    "version": "3.3.1",
    "uptime": "000+01:29:37.390",
    "uptime (seconds)": 5377,
    "freemem": 114,
    "reset reason": "Software reset CPU / Software reset CPU",
    "Dallas sensors": 0
  },
  "Network": {
    "connection": "Ethernet",
    "hostname": "ems-esp",
    "MAC": "7C:87:CE:44:C9:13",
    "IPv4 address": "10.100.13.46/255.255.255.0",
    "IPv4 gateway": "10.100.13.1",
    "IPv4 nameserver": "10.100.13.1"
  },
  "Status": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "telegrams received": 15834,
    "read requests sent": 1548,
    "write requests sent": 0,
    "incomplete telegrams": 0,
    "tx fails": 0,
    "rx line quality": 100,
    "tx line quality": 100,
    "MQTT": "connected",
    "MQTT publishes": 1860,
    "MQTT publish fails": 0,
    "Dallas reads": 0,
    "Dallas fails": 0,
    "syslog IP": "10.100.12.31",
    "syslog active": true
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Enviline/Compress 6000AW/Hybrid 7000iAW/SupraEco/Geo 5xx (DeviceID:0x08, ProductID:172, Version:01.10)",
      "handlers": "0x10 0x11 0xC2 0x14 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x494 0x495 0x48D 0x48F 0x48A"
    },
    {
      "type": "Thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:13.05)",
      "handlers": "0xA3 0x06 0xA2 0x12 0x2A5 0x2B9 0x2AF 0x29B 0x471 0x2A6 0x2BA 0x2B0 0x29C 0x472 0x2A7 0x2BB 0x2B1 0x29D 0x473 0x2A8 0x2BC 0x2B2 0x29E 0x474 0x2F5 0x31B 0x31D 0x31E 0x23A 0x267 0x240"
    },
    {
      "type": "Gateway",
      "name": "KM200/MB LAN 2 (DeviceID:0x48, ProductID:189, Version:04.07)"
    }
  ]
}
```

proddy, 2022-01-23:
I would wait until 3.4 is released and then try again

pypb, 2022-01-23:
Does that mean you have identified something that may be the cause?

In the meantime, I have started a logged console session with debug log level in the hopes of capturing any interesting messages when it happens again.

Let me know if there is anything else I can do to help debug.

MichaelDvP, 2022-01-23:
It looks like a restart with uptime counting from 0, accesspoint online until I141 and newly registering devices from I141 on. Until all devices and values are registered (take 1-2 min) you will get empty value info. It's a bit strange that ha-config values are removed after reboot.
As always on reboots: check power supply, try USB-power, or try to reduce wifi-tx-power.

pypb, 2022-01-23:
> It looks like a restart with uptime counting from 0, accesspoint online until I141 and newly registering devices from I141 on. 

You're right, I didn't realize that column shows the uptime.

> Until all devices and values are registered (take 1-2 min) you will get empty value info. It's a bit strange that ha-config values are removed after reboot. 

The API actually continues to output null for hours after this happens, until I manually restart it.

> As always on reboots: check power supply, try USB-power, or try to reduce wifi-tx-power.

I'll try with an external power supply. Not using wifi.

pypb, 2022-01-23:
I have now tried rebooting the gateway multiple times manually and I can get this problem to appear every few reboots. Issuing call commands from the console has the same result as the API, they return nothing. Waiting does not help.

```
ems-esp:# call system info
000+00:12:34.278 I 9: [command] Calling system command 'info'
ems-esp:# show devices
These EMS devices are currently active:

(1) Boiler: Enviline/Compress 6000AW/Hybrid 7000iAW/SupraEco/Geo 5xx (DeviceID:0x08, ProductID:172, Version:01.10)
 This Boiler will respond to telegram type IDs: 0x10 0x11 0xC2 0x14 0x15 0x1C 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA 0x494 0x495 0x48D 0x48F 0x48A

(2) Thermostat: RC300/RC310/Moduline 3000/1010H/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:13.05)
 This Thermostat will respond to telegram type IDs: 0xA3 0x06 0xA2 0x12 0x2A5 0x2B9 0x2AF 0x29B 0x471 0x2A6 0x2BA 0x2B0 0x29C 0x472 0x2A7 0x2BB 0x2B1 0x29D 0x473 0x2A8 0x2BC 0x2B2 0x29E 0x474 0x2F5 0x31B 0x31D 0x31E 0x23A 0x267 0x240

(3) Gateway: KM200/MB LAN 2 (DeviceID:0x48, ProductID:189, Version:04.07)

ems-esp:# show mqtt
MQTT is connected
MQTT publish errors: 0

MQTT topic subscriptions:
 ems-esp/system/#
 ems-esp/boiler/#
 ems-esp/thermostat/#
 ems-esp/thermostat_hc1

MQTT queue is empty
```

pypb, 2022-01-26:
The gateway has been stable (no spontaneous reboots) since I connected an external power supply, so at least that part has been cleared up.

However, the issue with `call system info` sometimes returning nothing after a reboot is still there. I have done a bit of debugging by doing manual reboots and collecting logs using syslog.

Out of 8 reboots, five failed like above and three booted up fine. I noticed one difference between failures and success, the first couple of seconds never reach the syslog server on failed reboots.

The successful ones all start with:
```
emsesp 000+00:00:00.000 I 0: Starting Syslog
emsesp 000+00:00:00.000 I 1: Ethernet initialized
syslog 000+00:00:00.000 I 2: Log level set to TRACE
```
While the first log messages never reach the syslog server on failed reboots:
```
emsesp 000+00:00:05.962 D 44: New EMS device detected with ID 0x08. Requesting version information.
mqtt 000+00:00:07.214 W 186: Lost device value for boiler_hptr3. Removing HA config
```
Looking at the log through the web gui does not contain much:
```
2022-01-26 16:53:58.000 I 0: [emsesp] Last system reset reason Core0: Software reset CPU, Core1: Software reset CPU
2022-01-26 16:53:58.000 I 1: [emsesp] EMS Device library loaded with 87 records
2022-01-26 16:53:58.000 I 2: [emsesp] Starting Access Point with captive portal on 192.168.4.1
2022-01-26 16:54:04.863 I 3: [emsesp] Starting NTP
2022-01-26 16:54:04.863 I 4: [emsesp] Ethernet Connected with IP=10.242.13.46, speed 100 Mbps
2022-01-26 16:54:04.863 I 5: [emsesp] mDNS responder service started
2022-01-26 16:54:06.897 I 6: [emsesp] Recognized new boiler with device ID 0x08
2022-01-26 16:54:07.909 I 7: [emsesp] Recognized new thermostat with device ID 0x10
2022-01-26 16:54:08.000 I 8: [emsesp] Stopping Access Point
```
I could provide you with the complete logs if you like, but since the first few seconds are missing from all the failed examples they may not help much.

If you have any ideas on how to debug this further, I'd be happy to try it.

proddy, 2022-01-26:
Ok, please help write down the actual problem. Is it:

"with a BBQKees gateway which is powered by an external 5V USB source, after a hard reboot and waiting a few seconds, the Console command `call system info` sometimes shows no information back on the screen,  50% of the time?"

I have a zillion questions, but let's first get the problem statement accurate!

pypb, 2022-01-26:
> Ok, please help write down the actual problem. Is it:
>
> "with a BBQKees gateway which is powered by an external 5V USB source, after a hard reboot and waiting a few seconds, the Console command `call [...]

I'll attempt to clarify.

_After a reboot or power cycle, the BBQKees E32 gateway will sometimes end up in a state where:_

- _Issuing `call system info` from the console returns no information_
- _API calls return null_
- _Unable to expand devices in Dashboard_

_However, MQTT is still functional and the console command `show devices` still works and returns three active devices._

_It does not recover from this state in any reasonable amount of time (not for hours, at least). You have to perform one or several reboots until it functions normally. Approximately 50% of reboots will trigger the problem._

_This happens both when the gateway is powered by a 12V DC power adapter or by bus power._

_Gateway is running firmware 3.3.1 and is connected to the service jack on an IVT AirX 90 / AirModule E9._

proddy, 2022-01-26:
Great summary, thanks for that. Next step is to isolate pieces so we can home in on the root cause. Things to consider trying

1. with syslog, mqtt. wifi, ntp and ap disabled, does it still happen?
2. when you get nothing back from `call system info` what does `show system` tell you. Do you have >140kb left?
3. if it still fails, disconnect the EMS. Does it still happen?
4. final case is removed the ESP32, using that crazy USB/TTL converter, flipping the jumper, flashing it with a debug compiled firmware and use something like ptty to connect to the Serial and trace what is going on.
5. last resort, do you have an S32 or just an plain old ESP32 with wifi you can try out instead?
6. last last resort, ask BBQKees to send you a new one ;)

pypb, 2022-01-30:
> 1. with syslog, mqtt. wifi, ntp and ap disabled, does it still happen?
> 2. when you get nothing back from `call system info` what does `show system` tell you. Do you have >140kb left?

I think I have narrowed it down. The problem will only occur when MQTT and AP is both enabled (AP is set to `When Network Disconnected`).

- Everything disabled: OK
- Any combination of syslog, ntp & mqtt enabled: OK
- Any combination of syslog, ntp & AP enabled: OK
- Any combination with MQTT & AP enabled simultaneously: problem reproducible

I ran `show system` after each reboot, and _Free heap_ is a lot lower when `call system info` is NOT working.

- With MQTT and AP both enabled AND `call system info` NOT working, _Free heap_ is between 80k and 116k.
- With either MQTT or AP disabled with everything working, _Free heap_ is above 130k

proddy, 2022-02-05:
Ok, it's not memory related then. There is enough available heap.

MQTT enabled and AP enabled (when network disconnected) is the default settings everyone uses, so I'm confused why it fails on your setup since v3.3 which is a couple of months old. Are you getting a lot of wifi drop-outs and then constantly switching between AP and WiFi  perhaps? Check the wifi strength with `show system`

pypb, 2022-02-05:
I'm not using wifi, I have an ethernet cable connected (it's an E32 gateway with an ethernet port).

proddy, 2022-02-05:
ok. Then I'm out of ideas, sorry!

proddy, 2022-04-02:
@pypb are you still seeing issues, like clicking on the Dashboard sometimes shows no data back?

pypb, 2022-04-03:
> @pypb are you still seeing issues, like clicking on the Dashboard sometimes shows no data back?

No, as long as I keep WiFi disabled (AP set to Never) it's stable. Haven't tried enabling it with 3.4.0b, though.

closed 2022-05-05 by proddy