emsesp/EMS-ESP32 #321: tempautotemp disappeared

tp1de, 2022-01-24:
tempautotemp suddenly disappeared in actual dev version.

MichaelDvP, 2022-01-24:
No, only if not set, if it have a legal value it is shown. We discussed that before.

tp1de, 2022-01-24:
But if not seen within api it can't be set by api (iobroker) as well .....

proddy, 2022-01-24:
> tempautotemp suddenly disappeared in actual dev version.

That's because it does actually disappear. There was a long discussion about this (you really need to start watching this repo so you're up to date). And a result I had to code some logic into actually test and remove 'zombie' entities from the MQTT Discovery payload.

tp1de, 2022-01-24:
> And a result I had to code some logic into actually test and remove 'zombie' entities from the MQTT Discovery payload.

Why do you call it "zombie". On km200 it is called temporaryRoomSetpoint per hc is for me most used entity to change temporarily the hc room setpoint. 
On a RC310 you have to use it or change from automatic to manual mode. But then you have the risk to vhange it back again. We discussed this now a couple of times (sure I still remember).

Shall I provide the telegrams by changing out of km200?

proddy, 2022-01-24:
zombie because it would suddenly appear with a valid number, and hence included in the MQTT payload, and then later get a value of unset in the EMS telegram and then excluded. For example, in your case the number of Thermostat entities would bounce between 58 and 59. This confused Home Assistant as its expects static data. People were complaining of errors in the HA log because it couldn't find and render that sensor type. So now it's just removed and in the log you'll see the debug line saying 'lost device entity tempautotemp'

tp1de, 2022-01-24:
hc1: 01B9 offset 8:
 0: 48 10 FF 08 01 B9 00 D1
-1: 48 10 FF 08 01 B9 FF 2E
10: 48 10 FF 08 01 B9 14 C5
20: 48 10 FF 08 01 B9 28 F9

hc2: type: 01BA same offset

tp1de, 2022-01-24:
Everybody who has a RC310/RC300 ems+ Thermostat will need this entity. 
It shall always be present and should have valid range from 5 to 30 and two enum states 0:off and -1: automatic.
(in ioBroker it's a number with range from -1 to 30);

proddy, 2022-01-24:
it is present. I think we talking on different frequencies. I'm closing this issue.

tp1de, 2022-01-24:
?????

tp1de, 2022-01-24:
> it is present. where ???

[image]

MichaelDvP, 2022-01-25:
The tempautotemp is a (for emsesp) useless value and especially in RC300 confusing mathematical incorrect. 
It is only a temporarry setpoint, only valid if in auto mode and automatically cleared on any mode-change and on any switchpoint.
In emsesp the writing of `seltemp` makes an automatic selection of the right setpoint, in manual mode it writes to manualtemp, in auto mode it writes to tempautotemp. There is no need for an extra listing of tempautotemp, it's only confusing users.
The normal setpoints for manual/day/night/comfort/nofrost can be written any time and are stable, If we change the mode it becomes active. Tempautotemp is different, it works only if auto mode is selected before setting tempautotemp. If tempautotemp is set, `seltemp` shows the right, identical value, showing tempautotemp is completly redundant.

All other thermostats have tempautotemp zero if not active, only RC300 have the setting 0xFF for off and the stupid KM200 publishes this as -1 instead of -0.5.

proddy, 2022-01-25:
I don't want to add this. As Michael described it's a silly temporary value, probably a last minute hack by Buderus that I don't want to repeat in EMS-ESP. It's not writable and serves no purpose. The fact that KM200 displays this is irrelevant as again, it provides no value to anyone.

@tp1de I keep asking you this - what problem are you trying to solve? Why is it important for you and other users of iobroker? Other than trying to mimic the KM200 for your personal use what value does showing this entity have for others? Help me understand please.

MichaelDvP, 2022-01-25:
I think KM200 only mirrors the telegram fields and this makes it the only way to change temperature in auto mode. But i hope the guy who translates 0xFF to -1 for KM200 isn't a employee of bosch anymore.
@proddy Remove all `tempautotemp` to avoid confusion? Otherwise it will be a neverending discussion.

tp1de, 2022-01-25:
@proddy @MichaelDvP 
Please wait, you are too fast in conclusions . I will explain today .....

tp1de, 2022-01-25:
> But i hope the guy who translates 0xFF to -1 for KM200 isn't a employee of bosch anymore.

Please let me explain the ems+ number entity logic again. This is independent from km200 and applies for RC310 as well.

There are 2 types of "number-entities":
1. pure numeric values defined if writable with min/max - e.g.:
 { "id": "/heatingCircuits/hc1/temperatureLevels/comfort2", "type": "floatValue",
      "writeable": 1, "recordable": 0, "value": 24,"unitOfMeasure": "C", "minValue": 20.5, "maxValue": 30, "valIs": "value" } 
 2. a combination of (1) with enum values:
 {"id": "/heatingCircuits/hc1/temporaryRoomSetpoint","type": "floatValue",
      "writeable": 1,"recordable": 0,"value": -1,"unitOfMeasure": "C","minValue": 5, "maxValue": 30,
      "state": [{"off": 0},{"na": -1}],"valIs": "value"}
So enum (state) translates 0xFF to "na: -1" and 0x00 to "off: 0"

(2) is used for other entities as well. E.g. where value comes from an temp-sensor (resistor) to indicate short and/or open circuits:
{"id": "/heatSources/actualSupplyTemperature",type": "floatValue",
     "writeable": 0,"recordable": 0,"value": 52,"unitOfMeasure": "C",
     "state": [{"open": -3276.8},{"short": 3276.7 }],"valIs": "value"}

A missing temp-sonsor or a short circuit will be indicated within RC310 and km200 in the same way.

tp1de, 2022-01-25:
> I don't want to add this. As Michael described it's a silly temporary value, probably a last minute hack by Buderus that I don't want to repeat in EMS-ESP. It's not writable and serves no purpose.

1. The entity /heatingCircuits/hc1/temporaryRoomSetpoint (tempautotemp) is writable (see above).
2. The entity persists and indicates with value 0 that the hc is switched off (pump stops / if mixer-hc mixer closed)
3. If value 0 or within allowed range entered it shows on RC310 and within app's (via km200) a message:
    "your desired temperature has been changed to off/xx°C until hh.mm" (next switchpoint)
4. A value of -1 resets the hc to automatic mode. na: -1 stays persistent to indicate automatic mode

> I keep asking you this - what problem are you trying to solve?

I am using since 3 years this temporary setpoints per hc to switch the respective hc off if no heat demand is calculated.
This reduces / avoids unnecessary takting of the boiler. If all hc's are off the boiler stays off as well. 
I stll have a node-red script running which calculates every 5 minutes if there is a heating demand and then sets the value either to 0 or -1.

I will now test ems-esp .... please wait

MichaelDvP, 2022-01-25:
That some numbers have a special meaning is known and NorbertS has documented this. Sensors 0x8000 for na/open, 0x7FFF for shortened, emsesp disables these sensors. Also for single byte values 0xFF for `na` is common in telegrams and emsesp does not show these values. That happens now to tempautotemp, the value is 0xFF and `na`.
The values you mentioned for sensor open/short fits the scale 3276.7 is 0x7FFF with scale of 1/10 °C. But 0xFF is NOT -1 in scale of 0.5°C.  It means `na` andnot -1. Also `na` means NOT AVAILABLE, so it is not shown and this is right. 

What you seems not to understand: writing to `seltemp` in auto mode writes to the field you know as tempautotemp. This was also in v1.9 (as `temp` command).

tp1de, 2022-01-25:
> In emsesp the writing of `seltemp` makes an automatic selection of the right setpoint, in manual mode it writes to manualtemp, in auto mode it writes to tempautotemp

When being started in automatic mode without changing seltemp then tempautotemp is not shown within dashboard or api/thermostat. When seltemp value is changed then tempautotemp suddenly appears and disappears again when automatic mode is selected on RC310 / km200 or next switchpoint has arrived. 

1. So the existence of tempautotemp does not persist. This should be changed. (A -0.5 is better than nothing).
2. tempautotemp is not writable but it should be:
{
  "name": "tempautotemp",
  "fullname": "hc1 temporary set temperature automode",
  "circuit": "hc1",
  "value": 0,
  "type": "number",
  "min": 5,
  "max": 30,
  "uom": "°C",
  "writeable": true,
  "visible": true
}
3. tempautotemp min should be changed to -1
4. entering -1 within seltemp does work on api call but not within dashboard and is changing seltemp back 
to automatic mode temperature - tempautotemp disappears

Two have to fields - seltemp for entering and test on existence of tempautotemp - makes visualisation unnecessary difficult and differs from km200 behavior.
[image]

**I hope that you can follow my argumentation that tempautotemp should persist and should be writable.
(if -1 is not possible since all values are divided by 2 then -0.5 is better then nothing)**

MichaelDvP, 2022-01-25:
> tempautotemp is not writable but it should be:
> entering -1 within seltemp does work on api call but not within dashboard

Works here. 
[image]
```
000+04:29:27.757 I 13: [command] selected room temperature: -1
000+04:29:27.927 N 14: [emsesp] Me(0x0B) -> Thermostat(0x10), RC35Set(0x47), data: FE (offset 37)
000+04:29:38.188 N 15: [emsesp] Thermostat(0x10) -> All(0x00), RC35Set(0x47), data: 01 22 2C 1E 00 28 FE 02 00 05 05 2D 01 01 04 55 0F 42 01 00 3C 00 12 05 07 01 02
000+04:29:38.529 N 16: [emsesp] Thermostat(0x10) -> All(0x00), RC35Set(0x47), data: 00 01 28 19 0A 01 00 FF 32 2D 00 EC 05 05 02 (offset 27)
000+04:40:59.412 I 17: [command] temporary set temperature automode: -1
000+04:40:59.560 N 18: [emsesp] Me(0x0B) -> Thermostat(0x10), RC35Set(0x47), data: FE (offset 37)
000+04
```
> Two have to fields - seltemp for entering and test

seltemp always works and shows the right setpoint, i agree that two fields for the same value is not nessesary, so imo we should remove tempautotemp completly to avoid confusion.

 > differs from km200 behavior.

Yes, emsesp is not a km200 clone, it's a different project. 

I've made a pr with the change to -1, but it's proddy decision if he want's it in his software.

closed 2023-02-19 by proddy