emsesp/EMS-ESP32 #322: Please ad "ENTITIES" like on dashboard within api call ems/api/system per device

tp1de, 2022-01-24:
Please ad the "ENTITIES" info (similar to dashboard) within system JSON within Devices array per type (boiler, thermostat etc): 

[image]

proddy, 2022-01-24:
- what problem are you trying to solve? Why is this useful?
- there is an entities endpoint (e.g. http://ems-esp.local/api/boiler/entities)
- do you want the description of the shortnames (like seltemp)?
- it will be a lot of data, more than 250 extra words which may be too large for the JSON. It's better to include this

tp1de, 2022-01-24:
I think I was not clear enough:

"Devices": [
    {
      "type": "Boiler",
      "name": "Buderus Logano GB125/KB195i/Logamatic MC110 (DeviceID:0x08, ProductID:133, Version:01.44)",
      **"entities": 56,**
      "handlers_received": "0xC2 0x15 0x1C 0xD1 0xE3 0xE4 0xE5 0xE9",
      "handlers_fetched": "0x14 0xE6 0xEA",
      "handlers_pending": "0x10 0x11 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A"
    },
    {
      "type": "Thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:18.03)",
      **"entities": 58,**
      "handlers_received": "0x06 0x02A5 0x02A6  ....",
      "handlers_fetched": "0x02B9 0x02AF 0x029B ......",
      "handlers_pending": "0xA3 0xA2 0x12 0x0471 ...."
    },
    {
      "type": "Controller",
      "name": "BC Base (DeviceID:0x09, ProductID:230, Version:37.01)" ,
      **"entities":  0**

proddy, 2022-01-24:
got it, some more questions. Is the # entities
- including commands? (e.g. reset is included which is not a data value) ?
- excluding those that are excluded (e.g. only the ones you see from the API/Console/MQTT) ?

tp1de, 2022-01-24:
[image]

proddy, 2022-01-24:
ok. I'll add

one observation: the Controller never has any entities so I exclude that from the WebUI (as shown above). Should we do the same with the Gateway? 

2nd observation: 58 entities for an RC300 thermostat! holy cow!

tp1de, 2022-01-24:
> Should we do the same with the Gateway?

I would leave it in to show that a gateway is available

> 2nd observation: 58 entities for an RC300 thermostat! holy cow

Still some enteties missing which are available within km200 (switchTimes, holidayModes etc) and some more within RC310. Still some job to do ....

closed 2022-01-25 by proddy

MichaelDvP, 2022-01-25:
> 2nd observation: 58 entities for an RC300 thermostat! holy cow

look at [this](https://github.com/emsesp/EMS-ESP32/issues/294#issuecomment-1013938531), with some more hcs there are 150 entities.