emsesp/EMS-ESP32 #325: Update to Build v3.4.0b1 not possible !

zibous, 2022-01-25:
**Bug description**
 - Build v3.4.0b1 downloaded
 - Select EMS-ESP-3_4_0b1-ESP32.bin
 - Start Upload
 - Nothing happens - it looks like that the upload do not work
 - EMS-ESP is restarting, please wait…
 - System Status shows EMS-ESP Version v3.4.0b0

Furthermore, I no longer have a **HC1 thermostat** available in the Homeassistant.

[image]

I can't update any version at the moment.

**Steps to reproduce**
Used different Browsers (Safari, Chrome, Firefox..), Clear cache, start Browser clean...

**Device information**
```
{
  "System": {
    "version": "3.4.0b0",
    "uptime": "000+00:29:51.665",
    "uptime (seconds)": 1791,
    "freemem": 162,
    "reset reason": "Software reset CPU / Software reset CPU",
    "Temperature sensors": 0,
    "Analog sensors": 0
  },
  "Network": {
    "connection": "WiFi",
    "hostname": "*********",
    "SSID": "***********",
    "BSSID": "**************",
    "RSSI": -46,
    "MAC": "********",
    "IPv4 address": "10.1.1.147/255.255.255.0",
    "IPv4 gateway": "10.1.1.254",
    "IPv4 nameserver": "10.1.1.217"
  },
  "Status": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 1793,
    "bus reads (tx)": 396,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 0,
    "bus reads failed": 3,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100,
    "MQTT status": "connected",
    "MQTT publishes": 814,
    "MQTT publish fails": 0,
    "Temperature sensors": 0,
    "Temperature sensor reads": 0,
    "Temperature sensor fails": 0,
    "Analog sensors": 0,
    "Analog sensor reads": 0,
    "Analog sensor fails": 0,
    "API calls": 3,
    "API fails": 0,
    "syslog_started": true,
    "syslog_level": "info",
    "syslog_ip": "10.1.1.96",
    "syslog_queue": 0
  },
  "Devices": [
    {
      "type": "Boiler",
      "name": "Buderus GBx72/Trendline/Cerapur/Greenstar Si/27i (DeviceID:0x08, ProductID:123, Version:07.00)",
      "handlers_received": "0x10 0x11 0x15 0x1C 0x18 0x19 0x1A 0x35 0x34 0x2A",
      "handlers_fetched": "0x14 0x16 0x33 0x26",
      "handlers_pending": "0xC2 0xD1 0xE3 0xE4 0xE5 0xE6 0xE9 0xEA"
    },
    {
      "type": "Thermostat",
      "name": "RC300/RC310/Moduline 3000/1010H/CW400/Sense II (DeviceID:0x10, ProductID:158, Version:74.02)",
      "handlers_received": "0x06 0xA2 0x02A5 0x02BA 0x02BB 0x02BC 0x02BD 0x02BE 0x02BF 0x02C0 0x031D 0x0267",
      "handlers_fetched": "0x02B9 0x02AF 0x029B 0x02F5 0x031B 0x023A 0x0240",
      "handlers_pending": "0xA3 0x12 0x0471 0x02A6 0x02B0 0x029C 0x0472 0x02A7 0x02B1 0x029D 0x0473 0x02A8 0x02B2 0x029E 0x0474 0x02A9 0x02B3 0x029F 0x0475 0x02AA 0x02B4 0x02A0 0x0476 0x02AB 0x02B5 0x02A1 0x0477 0x02AC 0x02B6 "
    },
    {
      "type": "Controller",
      "name": "BC25 (DeviceID:0x09, ProductID:125, Version:03.05)"
    }
  ]
}
```
**Additional context**
[image]

```
{
  "datetime": "19:36:38 25.01.2022",
  "floordry": "off",
  "dampedoutdoortemp": -0.5,
  "floordrytemp": 0,
  "building": "light",
  "minexttemp": -10,
  "wwsettemp": 60,
  "wwmode": "own_prog",
  "wwsettemplow": 45,
  "wwcircmode": "own_prog",
  "wwchargeduration": 60,
  "wwcharge": "off",
  "wwextra1": 0,
  "wwdisinfecting": "on",
  "wwdisinfectday": "we",
  "wwdisinfecttime": 180,
  "hc1": {
    "seltemp": 24,
    "mode": "auto",
    "modetype": "comfort",
    "ecotemp": 20,
    "manualtemp": 24,
    "comforttemp": 24,
    "summertemp": 15,
    "designtemp": 45,
    "offsettemp": 0,
    "minflowtemp": 25,
    "maxflowtemp": 60,
    "roominfluence": 0,
    "curroominfl": 0,
    "nofrosttemp": 0,
    "targetflowtemp": 43,
    "heatingtype": "radiator",
    "summersetmode": "auto",
    "summermode": "off",
    "controlmode": "optimized",
    "program": "prog_1",
    "fastheatup": 0
  }
}
```
**What option do I have to deploy a working version again?**

Is there any other way than removing the ESP32 and flashing it via USB?
Couldn't I use OTA?

proddy, 2022-01-25:
using the web UI to go from 3.4.0b0 to 3.4.0b1 failed?

You can try going via USB or OTA. USB is the easiest. See https://emsesp.github.io/docs/#/Uploading-firmware

zibous, 2022-01-26:
@proddy 
Thanks, but can't do OTA:

```
iMac:devlab ws$ mkdir ESP32 && cd ESP32
iMac:ESP32 ws$ python3 -m venv esp32-env
iMac:ESP32 ws$ . esp32-env/bin/activate
(esp32-env) iMac:ESP32 ws$ pip install --upgrade pip
(esp32-env) iMac:ESP32 ws$ pip install --upgrade esptool

(esp32-env) iMac:ESP32 ws$ esptool.py --debug --progress --port 8266 --auth ems-esp-neo -i 10.1.1.147 -f /Users/iMac/Downloads/EMS-ESP-3_4_0b1-ESP32.bin

### ERROR MESSAGE ####
esptool.py v3.2 : error: argument operation: invalid choice: 'ems-esp-neo' 
         (choose from 'load_ram', 'dump_mem', 'read_mem', 'write_mem', 'write_flash', 'run', 
		              'image_info', 'make_image', 'elf2image', 'read_mac', 
					  'chip_id', 'flash_id', 'read_flash_status', 
					  'write_flash_status', 'read_flash', 
					  'verify_flash', 'erase_flash', 
					  'erase_region', 'merge_bin', 
					  'version', 'get_security_info')
```
I can't find any information about how to use the correct parameters in esptool.py?

My other consideration is to buy a new ESP32 and flash it via USB and then replace it in the device.
That way I would always have one in reserve and it would be easier

**Can you give me a tip which one should I buy?**

proddy, 2022-01-26:
have you tried via USB? Have you disconnected it from the gateway circuit? The upload procedure I think is clearly documented in the EMS-ESP documentation. I just uploaded v3.4 via USB using the ESP-Flasher tool and that worked too. Perhaps your ESP32 is damaged? As for the ESP32 to buy depends on the footprint. Are you using a custom designed board? If not contact BBQKees for advice.

MichaelDvP, 2022-01-26:
@zibous for uploading via OTA you have to use `espota.py`.
The `esptool.py` is for flashing via USB.

zibous, 2022-01-26:
@MichaelDvP 

Many Thanks. I overlooked this in the description.
OTA worked perfectly, but it doesn't work via the GUI.

Unfortunately with this version:

- SYSLOG not working any more
- HC1 thermostat available in the home assistant.

Can I go back to EMS-ESP-3_3_1-ESP32 ?

MichaelDvP, 2022-01-26:
You can flash any build, upgrading and downgrading. Remember to refresh page in browser after changing the software (only 3.4 do this automatically). 

Have you checked the settings after upgrade? Or tried a factory reset and reconfigure from AP?

zibous, 2022-01-26:
@MichaelDvP 

> Have you checked the settings after upgrade?
Have testet on different syslog server, nothing will be logged there.

```
"Status": {
    "bus status": "connected",
    "bus protocol": "Buderus",
    "bus telegrams received (rx)": 1587,
    "bus reads (tx)": 366,
    "bus writes (tx)": 0,
    "bus incomplete telegrams": 1,
    "bus reads failed": 3,
    "bus writes failed": 0,
    "bus rx line quality": 100,
    "bus tx line quality": 100,
    "MQTT status": "connected",
    "MQTT publishes": 734,
    "MQTT publish fails": 0,
    "temperature sensors": 0,
    "temperature sensor reads": 0,
    "temperature sensor fails": 0,
    "analog sensors": 0,
    "analog sensor reads": 0,
    "analog sensor fails": 0,
    "API calls": 0,
    "API fails": 0,
    "syslog started": true,
    "syslog level": "debug",
    "syslog ip": "10.1.1.219",
    "syslog queue": 1
  },
  ```

zibous, 2022-01-26:
@proddy, @MichaelDvP 

> You can flash any build, upgrading and downgrading. Remember to refresh page in browser after changing the software (only 3.4 do this automatically).

Yes, with OTA works.
Now i am back to EMS-ESP Version v3.3.1 and can see the `climate.thermostat_hc1`.

[image]

proddy, 2022-01-26:
@zibous :
> Are you using a custom designed board?

Note., at some point we're going to discontinue support for v3.4 so I would try and figure out why it isn't working.

zibous, 2022-01-26:
@proddy 
Negativ, device from BBQKees.
`climate.thermostat_hc1` with EMS-ESP Version v3.3.1  is working -  with v3.4.0b0 not. 

**Upload Firmeware with GUI 3.4.0b0**

1. Select firmware

[image]

2. Drag firmware to GUI
[image]

OTA with `esptool.py` is working.

proddy, 2022-01-26:
I've never tested on OSX. I'll try it on my MacBook.

zibous, 2022-01-26:
@proddy 

**Syslog** is now working. Mea culpa was my fault I had not enabled the port for syslog.

`sudo nano /etc/rsyslog.conf`

```
module(load="imudp")
input(type="imudp" port="514")
module(load="imtcp")
input(type="imtcp" port="514")
```
`sudo systemctl restart rsyslog`

Now it is working.

**Upload firmware**
Works in Windows10 - can select file and/or drag file. Upload works.

MichaelDvP, 2022-01-26:
Regarding the climate control and thermostat_hc1 topic: There are a lot of changes and renamings in the HA autodiscovery. 
There is [this short troubleshooting article](https://emsesp.github.io/docs/#/Troubleshooting?id=ha-is-showing-errors-like-quotreceived-message-on-illegal-discovery-topicquot-or-quottemplate-variable-warning-39dict-object39-has-no-attributequot), but it's quite often that HA-users have issues with missing/wrong entities after upgrade.

zibous, 2022-01-26:
@MichaelDvP 
Yes is see, but on v3.3.1 i can see  the topic `homeassistant/climate/ems-heizung/thermostat_hc1/config`
```
{
   "mode_cmd_t":"~/thermostat_hc1",
   "temp_cmd_t":"~/thermostat_hc1",
   "name":"Thermostat hc1",
   "uniq_id":"thermostat_hc1",
   "~":"ems-heizung",
   "mode_stat_tpl":"{{value_json.hc1.hamode}}",
   "temp_stat_tpl":"{{value_json.hc1.seltemp}}",
   "curr_temp_tpl":"{{value_json.hc1.hatemp}}",
   "mode_stat_t":"~/thermostat_data",
   "temp_stat_t":"~/thermostat_data",
   "curr_temp_t":"~/thermostat_data",
   "min_temp":"5",
   "max_temp":"30",
   "temp_step":"0.5",
   "modes":[
      "auto",
      "heat",
      "off"
   ],
   "dev":{
      "name":"EMS-ESP Thermostat",
      "sw":"3.3.1",
      "mf":"",
      "mdl":"RC300/RC310/Moduline 3000/1010H/CW400/Sense II",
      "ids":[
         "ems-esp-thermostat"
      ]
   }
}
```

When I install a new version, I delete all topic homeassistant/* and then start the device. 
This rebuilds all discovery topics.

In version v3.4 the entry `/thermostat_hc1/config` is missing and therefore no 
thermostat is displayed in the home assistant either.

proddy, 2022-01-26:
it's all changed, there is no more thermostat_hc1 topic anymore. I think if you use MQTTExplorer and search for climate you'll find it.

Also if you go to HA->Configuration->Devices & Services->MQTT broker->EMS-ESP Thermostat you'll see it under the Controls too. 

[image]

And to add the EMS-ESP to HA all you need to do is click on the button "Add to Lovelace". That's all. No editing yaml files.

MichaelDvP, 2022-01-26:
That's because you dont have a roomtemperature sensor. V3.3.x had an option to replace the current roomtemp by setpoint, this option is missing in v3.4, the climate is only created if currtemp, seltemp and mode have values.

marc1de, 2022-01-26:
@MichaelDvP: ah, thats good to know. I was also wondering why I couldn't find the thermostat any more.

I decided to use the "slider" in HA instead (eg. on Thermostat "_hc manual temperature_", but then I realized, that it was somehow difficult to change the values (temperature range for that value was unfortunaly from 0 to 127 degrees C). So I thought, it might be easier to change the temperature values in ems-ESP. But then I discovered the next surprise: in ems-esp you can only change the values in 1 degree steps..... and all entities that have a value that ends with ".5"were not displayed. That field's were completly empty. And setting values in 0.5-Steps was impossible. Macbe you can have a look at that too,  if you are looking for the Thermostat-issue.

[image]

zibous, 2022-01-26:
@MichaelDvP 

ok - thanks for the explanation. But does this mean that in versions v3.4.x only one thermostat is available in Homeassistant when using a room thermostat, or will this be changed again?

proddy, 2022-01-26:
> I decided to use the "slider" in HA instead (eg. on Thermostat "_hc manual temperature_", but then I realized, that it was somehow difficult to change the values (temperature range for that value [...]

can you create a new GH issue for this? We're convoluting this issue (which is about supporting Mac OSX with the firmware upload!)

proddy, 2022-01-26:
> @MichaelDvP
> 
> ok - thanks for the explanation. But does this mean that in versions v3.4.x only one thermostat is available in Homeassistant when using a room thermostat, or will this be changed again?

yes, the HA Climat HVAC control needs current and set temperatures to work. Before we hacked it, but its not the right way. You still have the thermostat control and can use the sliders and the pulldown to change the temperature and mode. You just don't have the fancy circle. Or you can manually add it and fake the room temp yourself or alternatively download one of the many other alternatives on HA. EMS-ESP's focus is about capturing the data, not making pretty dashboards ;)

proddy, 2022-01-26:
actually, I just did a check and looks like in the [latest](https://www.home-assistant.io/integrations/climate.mqtt/) HA the current temp is optional. So we can add it back.

marc1de, 2022-01-26:
@proddy: ok, I will open a new issue report for the 0.5-Steps.

proddy, 2022-01-29:
I can confirm uploading firmware on MacOS/OSX doesn't work. It jump straight to 100% progress and restarts the EMS-ESP. Need to dig into this one.

proddy, 2022-02-05:
And reported by eltoro "I noticed Safari did not give me the possibility to select a .bin file". So multiple OSX issues that need addressing.

proddy, 2022-02-06:
made a quick fix for this. OSX doesn't recognize MIME types as other operating systems. See https://[react-dropzone](https://react-dropzone.js.org/#section-basic-example).js.org/#section-basic-example

closed 2022-02-06 by proddy