emsesp/EMS-ESP32 #326: Variable boiler dhw active  (wwactive) does not work

marc1de, 2022-01-25:
I think there is a bug in v3.4.0b0: **Variable boiler dhw active  (wwactive) does not work any more.** 
Status is always "off" (in ems-esp-Webinterface and also in Homeassistant)

my system: ESP32 / v3.3.5-1 
Boiler: Buderus GB172-20 T150 with Bosch Logamatic BC25
and Thermostat RC300
selected type of Transmission is EMS+

Actually I don't know if the Problem existed in V3.3 because I started with ems-esp just recently and did'nt have a closer look to that value.  But I could see, that the names of the Variables had changed, because I had to change my templates in HA (as far is I can remember, it was something with : "ww....." before and now they start with "dhw".

Note: originaly I posted this in the discord-channel but following the suggestion of Proddy, I opened this bug-report here again.

proddy, 2022-01-25:
I checked on my system and it's always false. Perhaps on some boilers its not used. One description is "warmwasserbetrieb". If you want to know if the hot tap water is running you can use `binary_sensor.boiler_tapwater_active`

marc1de, 2022-01-26:
thats the solution:
using "_**binary_sensor.boiler_tapwater_active**_" instead of "_binary_sensor.boiler_dhw_active_" finaly did the job!

(still shaking my head, why Bosch/Buderus sends out such stupid values that never change their status, but:
@proddy: thanks for that very useful hint!

proddy, 2022-01-26:
I think the Junkers use that field, and Buderus don't. It's hard to tell what is used from the telegrams so we include everything. 

The tapwater_active is a dynamically generated entity created by EMS-ESP using the current flow temp and flow type. There isn't anything we could find in the Boiler that says 'hot water is flowing'.

closed 2023-02-19 by proddy