emsesp/EMS-ESP32 #327: Status of GPIO Inputs and Counters not correctly shown in ems-esp-website

marc1de, 2022-01-25:
I think, this might be another bug: Status-changes on the GPIO's are not reported in the webview of ems-esp v3.4.0b0 (in HA I can see them changing however at least every 10 sek [thats the reporting-intervall, that i used]. Also the "counter-funktion" does not work as expected (short value changes (a few seconds) were ignored. 

At first I did my tests with GPIO 34 (with external pullup of 4k7) but after some hints from MichaelDVP in the discord-channel I understood, that some Pins are not very well suited for this purpose (eg. Pins 34,35,36,39 have no internal Pullups - thanks Michael for that hint)
He also gave me the information, that the pins use debounce-time of 15 ms, so that a minimum impuls lenght is required. 

After some research I found a table of usable Pins (https://randomnerdtutorials.com/esp32-pinout-reference-gpios) 

But that leaves the question for me: why my test with GPIO34 didn't show any results on the counter. External Pullup was connected and I waited minimum 3 seconds betweeen toggling the switch.
But nevertheless: I did an additional test on GPIO32 as counter (which should be perfectly fine, according to the mentioned table) (no ext. pullup / time between toggling from minimum 3sec to 20sec) and still no results in ems-esp-website (same with HA).

Same with inputs (used gpio33 now): It doesn't matter whatever I do. The status in the ems-esp-Website does not update. I waited and least 5 min and pressed "refresh" repeatedly).

_Note: originaly I started this question in the discord-channel but there I got the suggestion to open the issue here._

Edit: about 20 minutes later when I checked the website again, the missing input-counts could be seen...

MichaelDvP, 2022-01-26:
Yes, this is already fixed in the b1 build. Auto-update of the web values is 60 sec, HA updates with the value you have set in mqtt settings.

marc1de, 2022-01-26:
@MichaelDvP
thanks for that info. Meanwhile I updated to the b1 build and did some more testing. I can confirm, that the counter-updates now have improved.
But still there is an issue: I was not able to display the counter values in Homeassistant:

when I checked with MQTT-Explorer I found, that the counter-data format was "**count**"...

`{"32":{"name":"GPIO32","count":5},"33":{"name":"GPIO33","value":0}}`

but when I looked at the definition in MQTT-Explorer under "Sensor/ems-esp/analogsensor_32" it was "**value**" 

`{"stat_t":"ems-esp/analogsensor_data","val_tpl":"{{value_json['32'].value}}","name":"Analog Sensor GPIO32","uniq_id":"analogsensor_32","dev":{"ids":["ems-esp"]}}`
 
I'm not an expert but I suspected, that this could be a wrong definition?

MichaelDvP, 2022-01-26:
>  I suspected, that this could be a wrong definition?

Seems so, I'll make a pr to change the `count` to `value`. I'll also add a command to set/reset the counter value.

I'm thinking about adding factor and UOM also to counters. If you use a water meter with 4 pulses per liter a factor 0,25 counts liters. Same for gas or electrical energymeters counting in kWh.

marc1de, 2022-01-31:
Ok, setting a startvalue for the variables didn't help. The error in the log continued (but well, I never expected that - still wanted to check it out to be shure)

**But that brougt me to another interesting point:**

The ems-esp sends the GPIO-Values 10 seconds. Highly interesting is: **It sends the GPIO32-Value double!** One time only GPIO32 .... and the second time together with the GPIO33 !
Just thinking, if that could be the reason for my `"has no attribute '33'` Message

[image]

_Btw: just thinking:Is it possible, that HA expects a "full-telegramm with all the values" erverytime that it receives somthing from ems-esp? Or is it maybe related to the "retain-flag"? (disabled in my settings)_ 

[image]

[image]

And maybe that could be an explanation for the other errors that I have found:
[image]

@proddy @MichaelDVP: I would be highly interested if you also have this "double-send-effect" and if you can see similar errors in your HA-Logs.

marc1de, 2022-01-31:
@MichaelDVP thanks for looking but what do you mean with:
> .... should be outside the loop (one line below).

In fact I am now quite sure, that this is what causes the Problems: 
**HA dislikes it, if it gets an incomplete update of an topic:**

eg: if ems-esp sends an update for "analogsensor_data", it has to contain **ALL** the GPIO's 
 in analogsensor_data.

same happens to be with other topics like:
`boiler_data` or `boiler_data_ww`

because it always claims an error if there a single value missing.

MichaelDvP, 2022-01-31:
> In fact I am now quite sure, that this is what causes the Problems:

Yes, the bug is that emsesp sends out a message after adding each sensor and not only after all sensors are added. I'll change and make a PR.

MichaelDvP, 2022-01-31:
I think it takes a while for proddy to check all the PRs. You can test my appoach to make the counters scalable. The fix is included (and selburnpow to max. 130%). 
[EMS-ESP-3_4_0b2-ESP32.zip](https://github.com/emsesp/EMS-ESP32/files/7971761/EMS-ESP-3_4_0b2-ESP32.zip)

marc1de, 2022-01-31:
@MichaelDvP  thanks, - that perfectly solved the issue with the GPIO's.

_I still have found some other error messages, but because they are not directly related to the GPIO's I will report them in #337._

proddy, 2022-02-05:
merged,

commit 233d8280 (2022-01-26): add analog counter command, fix #327 "value"
commit 915418ec (2022-01-26): Merge pull request #330 from MichaelDvP/dev_

fixes #327 #329
commit 01c75ad5 (2022-02-05): Merge pull request #348 from MichaelDvP/dev_327

fix #327, analogsensor sending multiple mqtt messages
commit acc24127 (2022-02-05): fix #327, analogsensor sending multiple mqtt messages
closed 2022-02-05 by proddy