emsesp/EMS-ESP32 #329: unable to set values in 0.5-Steps (temperature) in ems-esp-webinterface

marc1de, 2022-01-26:
In ems-esp 3.3 I used a thermostate-card in Home-Assistant to change temperatures. 

[image]

After upgrading to ems v3.4.0b1 I realized, that this card was no longer available and so I decided to use the "slider" in HA instead (eg. on Thermostat "hc manual temperature"), but then I realized, that it was somehow difficult to change the values (temperature range for that value was unfortunaly from 0 to 127 degrees C).

[image]

So I thought, it might be easier to change the temperature values in ems-ESP. But then I discovered the next surprise: in ems-esp you can only change the values in 1 degree steps..... and all entities that have a value that ends with ".5"were not displayed in the "change-Window". That field's were completly empty. And setting values in 0.5-Steps was impossible. 

[image]

**Request:** to make it possible to enter xx.5-Values in ems-esp

_Note: originaly I mentioned this issue in #325, but was asked to open a new issue for that._

MichaelDvP, 2022-01-26:
The steps by the up/down arrows are always 1, but you can enter decimal places via keyboard.

marc1de, 2022-01-26:
@MichaelDvP 
Nope - unfortunaly decimal values were not accepted in this field. I tried it with decimal-point and komma. None of them were working.  If you enter a "." or "," the cursor just jumps back and clears the entry.

MichaelDvP, 2022-01-26:
I can not reproduce this. 
[image]

But entering the value is a bit nasty, you have always start with a number, `.5` is not allowed, it has to be `0.5` and for `-1` you have to enter `1`, go back with cursor and enter `-`. 
I don't know how to change this to have number check but easier input. But i can adjust the up/down buttons to the right steps.

marc1de, 2022-01-26:
Oh, thats funny! I think it's maybe browser dependant. I tried it with the actual Edge 97.0.1072.69 and Chrome  97.0.4692.99. With both of them I was not able to enter the decimal values (tried it with eg. 20.5 and 20.5 -  not with ,5 or .5)

After that I tried an older FirefoxPortable and there I was able to enter the values. :-)

Ok - I know, that it is really a pain to get a website working with different browsers but hopefully there is a way to get them working too.

proddy, 2022-01-26:
Tested with Edge 97.0.1072.69 and Chrome 97.0.4692.99 and it works for me. It uses what ever the decimal separator is on your browser's local (e.g. , for NL and . for UK).

marc1de, 2022-01-26:
Sorry, but now I am getting a little bit confused. Did you already implement the changes to v3.4.0b2 ? 
Because I just uggraded to this version, restarted, checked the version .... and could not see any changes in the behavior of the value fields. Still my Edge and Chrome show the same issue.
_(and regarding issue 327 I also couldn't see any "reset-counter" abilities)_ 

So I suppose that I was mistaken and the changes will apply in one of the future versions?

proddy, 2022-01-27:
The step still needs some tweaking. And the Edge/Chrome decimal. Is that what you're referring too?

The separator issue I couldn't reproduce so need more info

MichaelDvP, 2022-01-27:
Strange, steps working for me with Firefox and Edge (same build as you). Since edge is chromium based it think Chrome works like Edge in this case.
But entering decimals in Edge does not work if Intl.NumberFormat is set. Input of dot or comma clears the input field, also if the value was set to non-integer the field is empty on dialog open. If the value is integer with step 0.5, the input field is cleared when clicking arrow. The other inputs of dallas/analog uses numberValue() and working in Edge, but shows decimal dot.
My region setting is german with decimal comma.

marc1de, 2022-01-27:
uhh, thats really strange: as far as I can see you used the same browser-version than me:

**What I did:** After upgrading to v3.4.0b2, 
- I restarted and did a hardware reset (power off) on my ems-esp
- checked again, that version v3.4.0b2 was active 
- cleared my browsers cache and restartet the browser
 Then I opened ems-esp again and tried to change "Thermostat/hc1 manual temperature" via  keybord from 20,5 to 21,5.
[image]
_(Pic1: in the beginning the field is empty when I open it by clicking on the edit-Symbol - Note: its only empty when the old value was a decimal. When it was an integer the value is displayed at this step)_ 

[image]
_Pic 2: .... and that's as far as I could get: In that Moment, when I try to enter the "5" after the "21," the cursor jumps back and the filed is empty again._

[image]
_Note: if I try to use the "." instead of"," the behavior is a little different: then the cursor jumps already back, when I press the "." (But that seems logical to me, because in my settings the "." would be the wrong number format)_

**My System:**
Betriebsystemname	Microsoft Windows 10 Home
Version	10.0.19043 Build 19043
Betriebsystemhersteller	Microsoft Corporation
Systemtyp	x64-basierter PC
Prozessor	Intel(R) Core(TM) i5-8400 CPU @ 2.80GHz, 2808 MHz, 6 Kern(e), 6 

[image]

[image]

if there is any additional information, that I can provide - please let me know.

_**P.S.:** double checked with my Notebook and older Edge 96.0.1054.34 x64: same Results_

MichaelDvP, 2022-01-27:
Exactly the same here with same version of Edge (on Win10). I've tried:
`value={deviceValue.u ? Intl.NumberFormat().format(deviceValue.v) : deviceValue.v}`
This is as described, clears on dot, accepts comma, but clears on next number.
`value={deviceValue.u ? numberValue(deviceValue.v) : deviceValue.v}`
works, accepts only dot and shows only dot, same as in dallas/analogsettings dialog, comma key is ignored. 
`value={deviceValue.v}`
works, same as above, always dot.

I've also tested `type="text" inputmode="numeric" pattern="[0-9]*([,.][0-9])?"`
But this does not show the arrows.

MichaelDvP, 2022-01-27:
@marc1de This is the bin from my PR#332 (merged with b2). For proddy it makes no difference, for me it works on Edge, including the steps.
Please check on your system.
[EMS-ESP-3_4_0b2-ESP32.zip](https://github.com/emsesp/EMS-ESP32/files/7950747/EMS-ESP-3_4_0b2-ESP32.zip)

marc1de, 2022-01-27:
ok, her are my results:

**Chrome**: works , values per keyboard with comma, Arrows with full- and halfstep, -> everything ok
_(Note: "." does not work -> Field is cleared as expected)_ 

**Edge:** works , values per keyboard with comma, Arrows with full- and halfstep, -> everything ok
_(Note: "." does not work -> Field is cleared as expected)_ 

_but now it is getting really interesting:_
**my old FirefoxPortable**: Values per Keyboard with comma ->ok; but here comes the Clue: values with "." are also accepted (after "send" they will be displayed as decimal with comma, but the new values were accepted and afterwards they were displayed with comma
but on the other side: **the arrow keys do not work here** plus they are displayed much smaller (see Pic)
[image]

but now follows a problem:
**Android-tablet** (Android-Version 11 / UI-Core 3.1): integer values per keyboard ok
but decimal values are not accepted (with"." the field gets cleared and comma is not accepted (can not be entered .... I can see the comma on the tastatur-field but when I press it, it does not appear in the value field)
and : no arrow keys visible and therefore no chance to enter a decimal value.

proddy, 2022-01-28:
well, we need to decide whether we're going to use system locales or not. I prefer to as it's in line for the i8n changes I'm working on. For example there's no point (no pun intended!) showing the temperature values in the dashboard with a comma (,) separator like the 20,5 in your screenshot and then having to enter a new value using a point (.) separator. It should be consistent. 

That's why it needs some thinking and a bit more work.

MichaelDvP, 2022-01-28:
I totally agree that is shouldbe consitent and regional, but it should also work with the leading, actual browsers.
Now the dallas/analog input uses non-regional `numberValue()` and only valueInput the regional setting.
I've found some articles that inputs with type="number" are critical in a lot of implementations and recommended is  type="text" and set inputProps with numeric inputtype and pattern. But this does not show the arrows.

marc1de, 2022-01-28:
just to avoid confusion with my test-results:

**Chrome:** showed values with comma and accepted values with comma => **consistent**
**Edge:** showed values with comma and accepted values with comma => **consistent**

**FirefoxPtb:** showed values with comma and accepted values with comma => consistent
                  plus accepted also "." in the value field. 
_so in my opinion, there is no need to change something here (the normal display is perfectly consistent plus it allows you to enter the values with dot **or** comma)._

The only problem that I see, is that someone can not change the values on Android because there were no arrows and there was no option to enter the decimals with the keyboard.

MichaelDvP, 2022-01-28:
I think android depends on the used kexboard, not on browser. I tested with lightning and chrome on android 12 with google-keyboard, both working with dot and comma, but do not show arrows.
The issues of type="number" are also [mentioned here](https://mui.com/components/text-fields/#type-quot-number-quot) and there is a linked article.

proddy, 2022-01-28:
I will test again tonight, switching my Windows11 regional settings between NL and UK to see if the . and , work as well as the step

proddy, 2022-02-05:
merged Michaels changes to b3. works fine. thanks all

commit 2db053ab (2022-02-05): fix #329, numberformat in Edge, Chrome
commit 5143040b (2022-02-05): Merge pull request #332 from MichaelDvP/dev_

fix #329, numberformat in Edge, Chrome
closed 2022-02-05 by proddy