emsesp/EMS-ESP32 #333: Please explain the 3 different handlers within api/system

tp1de, 2022-01-27:
There are now 3 different handlers. 
I assume that handlers_received are those telegram types which are processed for entities.
What is the difference between fetched and pending?

"Devices": [
    {
      "type": "Boiler",
      "name": "Buderus Logano GB125/KB195i/Logamatic MC110 (DeviceID:0x08, ProductID:133, Version:01.44)",
      "entities": 56,
      "handlers received": "0xC2 0x15 0x1C 0xD1 0xE3 0xE4 0xE5 0xE9",
      "handlers fetched": "0x14 0xE6 0xEA",
      "handlers pending": "0x10 0x11 0x18 0x19 0x1A 0x35 0x16 0x33 0x34 0x26 0x2A"
    },

MichaelDvP, 2022-01-27:
"handlers received": Telegrams that are periodically sent by device to all Boiler(0x08)->All(0x00)...
"handlers fetched": Telegrams that are not broadcasted and requested by emsesp once a minute: Me(0x0B)<-Boiler(0x08) .. Boiler(0x08)->Me(0x0B)...
"handlers pending": Telegrams not received yet or empty on fetch. (Example: 0x16 and 0x33 were fetched but your boiler replys empty. 0x18 is a broadcasted Monitor for ems boilers but not yet received because your boiler does not use it.)

tp1de, 2022-01-27:
Thanks - understood

proddy, 2022-01-28:
we should add this explanation to the doc too

MichaelDvP, 2022-01-28:
To "How the busworks" as subchapter "Fetching telegrams" after "Reading and Writing"? 
Or somewhere in troubleshooting?

proddy, 2022-01-29:
thanks Michael for updating the docs.

closed 2023-02-19 by proddy